facebookresearch/habitat-lab #384: Inappropriate Parallelization in RL Evaluation for PointNav

Xiaoming-Zhao, 2020-05-01:
## ❓ Questions and Help

Hi, I find that the current code of RL models for PointNav does not support `VectorEnv` appropriately. Because we only have a single `val.json.gz` file in the `val` folder of PointNav dataset, the dataset will load ALL episodes from [this line](https://github.com/facebookresearch/habitat-api/blob/v0.1.4/habitat/datasets/pointnav/pointnav_dataset.py#L83). 

During the evaluation procedure, each rollout process will go through all episodes in a random order, which will result in an undefined evaluation behavior.

There are two ways to modify:
- Restrict the `NUM_PROCESSES` as 1 for evaluation
- Another easy workaround is to make the `val` folder the same structure as the `train` folder. Namely, put episodes from different scenes in separate files. Then we can safely utilize parallelization for accelerating the evaluation. We do not need to change any code.

erikwijmans, 2020-05-01:
You are correct, this is currently fixed on master however: https://github.com/facebookresearch/habitat-api/blob/master/habitat/datasets/pointnav/pointnav_dataset.py

Xiaoming-Zhao, 2020-05-01:
@erikwijmans Got you. The updated version is great. Thanks!

closed 2020-05-01 by Xiaoming-Zhao