facebookresearch/habitat-lab #385: Doc build crashes

dhruvbatra, 2020-05-03:
## 🐛 Bug

Doc build dies. Habitat-sim docs have already been built, m.css is present and accessible in that folder. 

## Command

```bash
(habitat-conda)
dbatra at dbatra-mbp in ~/projects/habitat-api/docs on master
$ ./build-public.sh
Warning: libgs not found
Traceback (most recent call last):
  File "../../habitat-sim/docs/m.css/documentation/python.py", line 2645, in <module>
    run(os.path.dirname(os.path.abspath(args.conf)), config, templates=os.path.abspath(args.templates))
  File "../../habitat-sim/docs/m.css/documentation/python.py", line 2445, in run
    modules_to_crawl += crawl_module(state, path, object)
  File "../../habitat-sim/docs/m.css/documentation/python.py", line 670, in crawl_module
    crawl_class(state, subpath, object)
  File "../../habitat-sim/docs/m.css/documentation/python.py", line 408, in crawl_class
    crawl_class(state, subpath, object)
  File "../../habitat-sim/docs/m.css/documentation/python.py", line 383, in crawl_class
    assert id(class_) not in state.crawled
AssertionError
```

mosra, 2020-05-07:
Opened https://github.com/facebookresearch/habitat-sim/pull/624 for this. Unfortunately I wasn't able to reproduce locally (maybe it's something specific to a particular Python version, I'm on 3.8 here), so if this happens again it should print a descriptive warning and continue (but without any guarantee on output sanity). In any case I'm interested in the warning, because it most probably points to a deeper problem somewhere else. It should look like this in the output:

> ERROR:root:Class thing.Foo previously found in other_thing.Bar, only one occurence will be chosen. Ensure each class is exposed only in a single module for generating correct documentation.

Thank you!

dhruvbatra, 2020-05-09:
Indeed, docs build fine with python 3.8: 

```bash
(py38)
dbatra at dbatra-mbp in ~/projects/habitat-api/docs on master
$ ./build-public.sh
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.config.get_config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.env.Env.__init__), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.env.Env.reconfigure), possibly hidden/undocumented?
WARNING:root:habitat.core.env.Env.step(self, action: typing.Union[int, str, typing.Dict[str, typing.Any]], kwargs) parameter kwargs is not documented
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.env.RLEnv.__init__), possibly hidden/undocumented?
WARNING:root:habitat.core.embodied_task.Action.step(self, args: typing.Any, kwargs: typing.Any) parameter args is not documented
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.embodied_task.EmbodiedTask.__init__), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.embodied_task.EmbodiedTask.overwrite_sim_config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.embodied_task.EmbodiedTask.overwrite_sim_config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.embodied_task.SimulatorTaskAction.__init__), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.dataset.Dataset.get_scenes_to_load), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.ActionSpaceConfiguration.__init__), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.ActionSpaceConfiguration.config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.BumpSensor.config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.DepthSensor.config), possibly hidden/undocumented?
WARNING:root:habitat.core.simulator.Observations.__init__(self, sensors: typing.Dict[str, habitat.core.simulator.Sensor], args: typing.Any, kwargs: typing.Any) parameter args is not documented
WARNING:root:habitat.core.simulator.Observations.__init__(self, sensors: typing.Dict[str, habitat.core.simulator.Sensor], args: typing.Any, kwargs: typing.Any) parameter kwargs is not documented
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.RGBSensor.config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.SemanticSensor.config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.Sensor.config), possibly hidden/undocumented?
<string>:1: (ERROR/3) Error in "py:function" directive:
invalid option block.

.. py:function:: habitat.core.simulator.Simulator.geodesic_distance
    :summary: Calculates geodesic distance between two points.
    :param position_a: coordinates of first point.
    :param position_b: coordinates of second point or list of goal points
    coordinates.
    :param episode: The episode with these ends points.  This is used for shortest path computation caching
    :return:
        the geodesic distance in the cartesian space between points
        :p:`position_a` and :p:`position_b`, if no path is found between
        the points then `math.inf` is returned.

ERROR:root:Failed to process a docstring for habitat.core.simulator.Simulator.geodesic_distance, ignoring:
 | Calculates geodesic distance between two points.
 |
 | :param position_a: coordinates of first point.
 | :param position_b: coordinates of second point or list of goal points
 | coordinates.
 | :param episode: The episode with these ends points.  This is used for shortest path computation caching
 | :return:
 |     the geodesic distance in the cartesian space between points
 |     :p:`position_a` and :p:`position_b`, if no path is found between
 |     the points then `math.inf` is returned.

WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.Simulator.reconfigure), possibly hidden/undocumented?
WARNING:root:habitat.core.simulator.Simulator.step(self, action, args, kwargs) parameter args is not documented
[... cut ...]
WARNING:root:habitat.core.registry.Registry.register_task_action(to_register, name: typing.Optional[str]) documents parameter action_space, which isn't in the signature
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.vector_env._make_env_fn), possibly hidden/undocumented?
------------------------------------------------------------------------
Public docs were successfully generated to the following location. Note
that the search functionality requires a web server in this case.
```

dhruvbatra, 2020-05-09:
Even with a clean python 3.6 environment with the same doxygen etc packages as the CI config (https://github.com/facebookresearch/habitat-sim/blob/master/.circleci/config.yml#L384), I am unable to build documentation. 

Here's your error: 

```bash
(py36)
dbatra at dbatra-mbp in ~/projects/habitat-api/docs on master
$ ./build-public.sh
ERROR:root:Class habitat.core.dataset.EpisodeIterator.__next_in_mro__ previously found in habitat.core.dataset.Dataset.__next_in_mro__, only one occurence will be chosen. Ensure each class is exposed only in a single module for generating correct documentation.
Traceback (most recent call last):
  File "../../habitat-sim/docs/m.css/documentation/python.py", line 2655, in <module>
    run(os.path.dirname(os.path.abspath(args.conf)), config, templates=os.path.abspath(args.templates))
  File "../../habitat-sim/docs/m.css/documentation/python.py", line 2455, in run
    modules_to_crawl += crawl_module(state, path, object)
  File "../../habitat-sim/docs/m.css/documentation/python.py", line 678, in crawl_module
    crawl_class(state, subpath, object)
  File "../../habitat-sim/docs/m.css/documentation/python.py", line 416, in crawl_class
    crawl_class(state, subpath, object)
  File "../../habitat-sim/docs/m.css/documentation/python.py", line 387, in crawl_class
    else: assert False
AssertionError
```

mosra, 2020-05-11:
Thanks, this helped very much, was able to localize and fix the assertion. Opened (and immediately merged) https://github.com/facebookresearch/habitat-sim/pull/627, now the docs should build on 3.6 again.

dhruvbatra, 2020-05-11:
[Making my way through the various PRs/issues on this]

Yes, docs for h-api (master) with h-sim (master) now build. There's still an error in there, but it appears unrelated, so I'll close this issue and move onto others. Thanks @mosra. 

```bash
dbatra at dbatra-mbp in ~/projects/habitat-api/docs on master
$ ./build-public.sh 
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.config.get_config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.env.Env.__init__), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.env.Env.reconfigure), possibly hidden/undocumented?
WARNING:root:habitat.core.env.Env.step(self, action: typing.Union[int, str, typing.Dict[str, typing.Any]], kwargs) parameter kwargs is not documented
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.env.RLEnv.__init__), possibly hidden/undocumented?
WARNING:root:habitat.core.embodied_task.Action.step(self, args: typing.Any, kwargs: typing.Any) parameter args is not documented
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.embodied_task.EmbodiedTask.__init__), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.embodied_task.EmbodiedTask.overwrite_sim_config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.embodied_task.EmbodiedTask.overwrite_sim_config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.embodied_task.SimulatorTaskAction.__init__), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.dataset.Dataset.get_scenes_to_load), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.ActionSpaceConfiguration.__init__), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.ActionSpaceConfiguration.config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.BumpSensor.config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.DepthSensor.config), possibly hidden/undocumented?
WARNING:root:habitat.core.simulator.Observations.__init__(self, sensors: typing.Dict[str, habitat.core.simulator.Sensor], args: typing.Any, kwargs: typing.Any) parameter args is not documented
WARNING:root:habitat.core.simulator.Observations.__init__(self, sensors: typing.Dict[str, habitat.core.simulator.Sensor], args: typing.Any, kwargs: typing.Any) parameter kwargs is not documented
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.RGBSensor.config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.SemanticSensor.config), possibly hidden/undocumented?
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.simulator.Sensor.config), possibly hidden/undocumented?
<string>:1: (ERROR/3) Error in "py:function" directive:
invalid option block.

.. py:function:: habitat.core.simulator.Simulator.geodesic_distance
    :summary: Calculates geodesic distance between two points.
    :param position_a: coordinates of first point.
    :param position_b: coordinates of second point or list of goal points
    coordinates.
    :param episode: The episode with these ends points.  This is used for shortest path computation caching
    :return:
        the geodesic distance in the cartesian space between points
        :p:`position_a` and :p:`position_b`, if no path is found between
        the points then `math.inf` is returned.

ERROR:root:Failed to process a docstring for habitat.core.simulator.Simulator.geodesic_distance, ignoring:
 | Calculates geodesic distance between two points.
 | 
 | :param position_a: coordinates of first point.
 | :param position_b: coordinates of second point or list of goal points
 | coordinates.
 | :param episode: The episode with these ends points.  This is used for shortest path computation caching
 | :return:
 |     the geodesic distance in the cartesian space between points
 |     :p:`position_a` and :p:`position_b`, if no path is found between
 |     the points then `math.inf` is returned.

[... cut ...]
WARNING:root:habitat.core.registry.Registry.register_task_action(to_register, name: typing.Optional[str]) documents parameter action_space, which isn't in the signature
WARNING:root:could not resolve a link to habitat.config.default.Config which is among INPUT_MODULES (referred from habitat.core.vector_env._make_env_fn), possibly hidden/undocumented?
------------------------------------------------------------------------
Public docs were successfully generated to the following location. Note
that the search functionality requires a web server in this case.
```

closed 2020-05-11 by dhruvbatra