facebookresearch/habitat-lab #386: how to specify agent rotation in pointnav task ?

ronjamoller, 2020-05-06:
## ❓ Questions and Help
HI, i am working on a modified pointnav task and i would like my agent to not only navigate to the correct position in the environment but also turn into a certain direction/orientation. I have successfully trained a normal pointnav task and looked into the jsons for object goal task so now i see 2 possible options:
1. modify the json to include a goal rotation (in addition to position and radius) and try to modify the code so it is used for the reward check (i guess there will have to be a range requirement like the radius for the position)
2. add a measure for the correct rotation and call it when the reward is computed
If anyone is working on something similar or has suggestions please let me know.
thank you

ronjamoller, 2020-05-07:
In case anyone else stumbles upon this and wants to do this too - i registered an additional rotational measure that is called in habitat_baselines.common.environments.NavRLEnv.get_reward and it seems to work out. Im pretty sure its not the clean way to do it though.

mathfac, 2020-05-08:
Similar logic was implemented for ImageNav task implementation and you can check it [here](https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/nav/nav.py#L253-L259). To avoid editing the PointNav dataset goal rotation was set as random rotation using random seed equal to `episode_id`. That guarantees consistent goal orientation across episodes.

Then you will need to implement reward changes anyway.

closed 2020-11-10 by ronjamoller