facebookresearch/habitat-lab #388: action_space_shortest_path function doesn't work

phongnhhn92, 2020-05-07:
Hello, I have been following the example of dataloader using the vectorized Habitat-env from the [Synsin project](https://github.com/facebookresearch/synsin). As mentioned in [this issue](https://github.com/facebookresearch/synsin/issues/2), the author have made some internal changes to the code so that we can randomly sample inputs. In my case, I would like to use the function 
[action_space_shortest_path](https://github.com/facebookresearch/habitat-api/blob/6091d0aedd41d348824252d74119d5d0d3355b8e/habitat/sims/habitat_simulator/habitat_simulator.py#L328) to get all the point between two agent's positions. Therefore, as extended from the suggestion from the author of Synsin, I have made another function to call the action_space_shortest_path but somehow it doens't work. Other functions of the Habitat simulator such as is_navigable,get_geodesic_distance, get_straight_shortest_path works well. Here is the error I have faced when I tried to run the action_space_shortest_path function:

`File "/home/phong/data/Work/Paper3/Libraries/habitat-api/habitat/sims/habitat_simulator/habitat_simulator.py", line 333, in action_space_shortest_path
    "This function is no longer implemented. Please use the greedy "
NotImplementedError: This function is no longer implemented. Please use the greedy follower instead`

This is my custom vector_env.py
```
import multiprocessing as mp
from multiprocessing.connection import Connection
from queue import Queue
from threading import Thread
from typing import Any, Callable, Iterable, List, Optional, Set, Tuple, Union

import gym
import numpy as np
from gym.spaces.dict_space import Dict as SpaceDict

import habitat
from habitat.config import Config
from habitat.core.env import Env, Observations
from habitat.core.logging import logger
from habitat.core.utils import tile_images

from habitat.core.simulator import AgentState

STEP_COMMAND = "step"
RESET_COMMAND = "reset"
RENDER_COMMAND = "render"
CLOSE_COMMAND = "close"
OBSERVATION_SPACE_COMMAND = "observation_space"
ACTION_SPACE_COMMAND = "action_space"
CALL_COMMAND = "call"
NAVIGABLE_COMMAND = 'navigate'
OBSERVATIONS = 'observations'
AGENT_STATE_COMMAND = 'get_agent_state'
GEODESIC_DISTANCE = 'get_geodesic_distance'
IS_NAVIGABLE = 'is_navigable'
ACTION_SPACE_SHORTEST_PATH = 'action_space_shortest_path'
GET_STRAIGHT_SHORTEST_PATH = 'get_straight_shortest_path_points'

def _make_env_fn(
    config: Config, dataset: Optional[habitat.Dataset] = None, rank: int = 0
) -> Env:
    r"""Constructor for default habitat Env.

    Args:
        config: configuration for environment.
        dataset: dataset for environment.
        rank: rank for setting seed of environment

    Returns:
        ``Env``/``RLEnv`` object
    """
    habitat_env = Env(config=config, dataset=dataset)
    habitat_env.seed(config.SEED + rank)
    return habitat_env

class VectorEnv:
    r"""Vectorized environment which creates multiple processes where each
    process runs its own environment. All the environments are synchronized
    on step and reset methods.

    Args:
        make_env_fn: function which creates a single environment. An
            environment can be of type Env or RLEnv
        env_fn_args: tuple of tuple of args to pass to the make_env_fn.
        auto_reset_done: automatically reset the environment when
            done. This functionality is provided for seamless training
[... 486 lines cut ...]
        self._workers = []
        for parent_read_queue, parent_write_queue, env_args in zip(
            parent_read_queues, parent_write_queues, env_fn_args
        ):
            thread = Thread(
                target=self._worker_env,
                args=(
                    parent_write_queue.get,
                    parent_read_queue.put,
                    make_env_fn,
                    env_args,
                    self._auto_reset_done,
                ),
            )
            self._workers.append(thread)
            thread.daemon = True
            thread.start()
        return (
            [q.get for q in parent_read_queues],
            [q.put for q in parent_write_queues],)

```

I also notice that this function action_space_shortest_path is currently not implemented and being replaced by the Greedy Follower but how to use it in the vector_env as I mentioned above ?
https://github.com/facebookresearch/habitat-api/blob/74795f2e8be32d9670fce378a887a26e53aee964/test/test_habitat_env.py#L385

erikwijmans, 2020-05-07:
This example shows how the shortest path follower works: https://github.com/facebookresearch/habitat-api/blob/master/examples/shortest_path_follower_example.py

phongnhhn92, 2020-05-08:
Thanks I manage to make it work using the RLEnv but my aim is to make this function work in the Vector_Env so that I can have multi-thread during training. I will update here if I have any success.

closed 2022-03-16 by phongnhhn92