facebookresearch/habitat-lab #389: About MAX_EPISODE_STEPS in configs

qweas120, 2020-05-08:
## ❓ Questions and Help
The MAX_EPISODE_STEPS in [objectnav_mp3d.yaml](https://github.com/facebookresearch/habitat-api/blob/master/configs/tasks/objectnav_mp3d.yaml) is 500, which also count turning actions(turn left and turn right). It greatly limits the range the agent can explore.

Can I modify this figure to 1000 when I submit to the evaluation system?

erikwijmans, 2020-05-08:
No, you unable not able to change the config in any way for evaluation (the simulator runs in a completely separate process and selects the config itself).

qweas120, 2020-05-08:
@erikwijmans Thanks for your reply!

I think 500 is very limited since the turning actions are counted. If an agent needs to explore the surrounding, it would turn many times before moving forward and It seems that the shortest geodesic path to the target needs 0~150 forward steps. A little exploration would exceed this limit.

dhruvbatra, 2020-05-09:
I agree with your intuitions -- that a higher action limit would allow for more interesting exploration behavior. 

You are welcome to modify this max-episode-steps variable at your end for your research (and associated papers). 

However, for the 2020 challenge we will not change this parameter. Multiple reasons: challenge is already underway, higher max-step limit means significantly longer evaluation times at our end, etc. We are happy to revisit this for 2021 and beyond.

closed 2020-05-09 by dhruvbatra