facebookresearch/habitat-lab #390: How many types of coordinates are there in Habitat?

Xiaoming-Zhao, 2020-05-08:
## ❓ Questions and Help
Hi, can I know how many types of coordinates are there in the Habitat-API/Habita-Sim settings? What is the origin of each coordinate? For origin, I mean the following:

- When we consider cartesian, it is `x=0`, `y=0` (position) and `theta=0` (orientation)
- When we consider polar, it is `r=0` (position) and `rho=0` (orientation). 

Currently, I have found the following types of coordinates in the Point Goal Navigation scenario:

- Global coordinate
  - Not sure where the origin is
  - For orientation, we have [`HeadingSensor`](https://github.com/facebookresearch/habitat-api/blob/59228a5e246dccbab4809ed0bcac3218760a365a/habitat/tasks/nav/nav.py#L321)
  - For position, it seems like we do not have a sensor. **If I want to have access to the global position, do I have to write my own sensor for this?**
  - My understanding is the `agent_state` generated from [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/blob/3943cf3bc7d9af24196dded001221632a7d21091/habitat_sim/agent/agent.py#L176), which is required by almost all sensors, gives us the global coordinate information. **Is this correct?**
- Episode coordinate
  - The origin position and heading is defined by the starting point of each episode
  - For orientation, we have [`EpisodicCompassSensor`](https://github.com/facebookresearch/habitat-api/blob/59228a5e246dccbab4809ed0bcac3218760a365a/habitat/tasks/nav/nav.py#L364)
  - For position, we have [`EpisodicGPSSensor`](https://github.com/facebookresearch/habitat-api/blob/59228a5e246dccbab4809ed0bcac3218760a365a/habitat/tasks/nav/nav.py#L386)
- Top-down map coordinate
  - From my understanding based on [this](https://github.com/facebookresearch/habitat-api/blob/59228a5e246dccbab4809ed0bcac3218760a365a/habitat/utils/visualizations/maps.py#L187), the origin is the top-left corner of the map
  - Can I know the difference between `_ind_x_min`, `_ind_x_max`, and `_coordinate_min`, `_coordinate_max` defined in [`TopDownMap`](https://github.com/facebookresearch/habitat-api/blob/59228a5e246dccbab4809ed0bcac3218760a365a/habitat/tasks/nav/nav.py#L664)?

Thanks!

erikwijmans, 2020-05-09:
For polar vs. cartesian, you have 2D cartesian correct, but 2D polar is wrong. `(r, rho)` defines the position and `theta` still defines heading/orientation.

For the global coordinate system, the origin is arbitrary as the global coordinate system is defined by the mesh itself and those typically have arbitrary origins.  Yes, if you want global position you have to add something for that.  Global heading and global position really shouldn't be inputs to the agent and thus should be sensors (we should really move the HeadingSensor CC @mathfac @abhiskk).  You are correct about agent state.

For TopDownMap, `_ind_x_min` and co, you can see how they are computed here: https://github.com/facebookresearch/habitat-api/blob/59228a5e246dccbab4809ed0bcac3218760a365a/habitat/tasks/nav/nav.py#L714-L717 
It is simply what part of the map actually has navigable space in it.

`COORDINATE_MIN`, `COORDINATE_MAX` are really related to only the grid like coordinate system used for the top-down map as we have to do some shift/scaling to get into a grid-like coordinate system. The global coordinate system is fixed in the sense that every scene has a fixed coordinate system, but the coordinate system in each scene is arbitrary (i.e. the point (0, 0, 0) has no physical or special meaning in each scene).

The episodic coordinate frame specifics both a rotation and translation wrt to the coordinate frame of the scene.  The rotation means that `p1 - p2` will change.  `||p1 - p2||` won't change, nor will the angle between them change, etc, but `p1 - p2` will.

Xiaoming-Zhao, 2020-05-09:
@erikwijmans Hi, thanks a lot! Yes, I find that I make a mistake that `p1 - p2` will indeed change. What will remain the same are `|| p1 - p2 ||` and the angel.

A follow-up question to verify that I understand the sensor part correctly. Let's define the **local coordinate**, whose origin is the `agent_state.position` and the negative-z axis is the direction the agent faces, namely `agent_state.rotation`.

When the agent takes an action, if I want to know the position change in the local coordinate, will the following code give me the result?
```python
# agent_state1 is the state before taking action
pos1 = quaternion_rotate_vector(agent_state1.rotation.inverse(), agent_state1.position)

# agent_state2 is the state after taking action
pos2 = quaternion_rotate_vector(agent_state2.rotation.inverse(), agent_state2.position)

local_pos_change = (pos2[0] - pos1[0], - (pos2[-1] - pos1[-1]))
```
`quaternion_rotate_vector` is defined as following:
https://github.com/facebookresearch/habitat-api/blob/59228a5e246dccbab4809ed0bcac3218760a365a/habitat/tasks/utils.py#L39

erikwijmans, 2020-05-09:
Whoops, apologies, I misread your statement about the agent state.  The agent state does not "gives us the global coordinate information", the state itself is in the global coordinate system for a given scene.  What it does is you you access to information in the global coordinate system.

I don't understand your example. In your example, the difference in local pos would be 0 as, by definition, `agent_state1.position == [0, 0, 0]` in the coordinate system defined by `agent_state1` and similarly for `agent_state2`.  If you want to compare positions/orientations, it only makes to do so in a common coordinate frame.

Xiaoming-Zhao, 2020-05-10:
Hi, I think there are some misunderstandings between us.
- Let us assume we have an `env = habitat.Env`. I could have `agent_state = env._sim.get_agent_state()`. As you clarified, **global coordinate system** has some arbitrary origin `(x0_global, y0_global, z0_global)` and origin orientation `theta0_global`. My understanding is `agent_state` will give us position `(x, y, z)` and orientation `theta` information w.r.t `(x0_global, y0_global, z0_global)` and `theta0_global`.
- The aim of defining **local coordinate system** is that I want to know the distance the agent moves along each axis when enabling actuation noises. For example, when disabling noise, `MOVE_FORWARD` will give us 0.25m along the negative-z axis and 0m along the x axis. I think these x and z axes are defined locally.
- When taking an action, I think the only information we could have from the simulator is the position and orientation changes w.r.t the **global coordinate systems**. However, the actions, namely `MOVE_FORWARD`, are defined locally since the concept of `FORWARD` is local.  What my code wants to do is to compute the changes in **local coordinate system** from the information of the **global coordinate system**.

Does this make sense? Thanks

erikwijmans, 2020-05-10:
Ah, makes sense.  In order to do what you want, here is how I think about it: The state of the agent at time `t` can be interpreted as the transformation from a its local coordinate frame to world coordinates.  The delta in position/orientation between two subsequent time steps is the position/orientation of the agent at time `t+1` in the local coordinate frame defined by the state at time `t`.  This can be computed by applying in inverse of the transformation defined by the state at time `t` to the state at time `t+1`.

So,
```python

agent_state1 = ... # Agent state at time t
agent_state2 = ... # Agent state at time t+1

state_delta = ... # This is position/orientation of the state at time t+1 in t's coordinates,
# AKA the delta between t and t+1

state_delta.rotation = agent_state1.rotation.inverse() * agent_state2.rotatiton
state_delta.position = (
    quaternion_rotate_vector(
        agent_state1.rotation.inverse(), 
        agent_state2.position,
    ) 
    - quaternion_rotate_vector(
        agent_state1.rotation.inverse(), 
        agent_state1.position,
    )
)
```

Skylion007, 2020-05-10:
We may want to consider writing a utility function for this.

Xiaoming-Zhao, 2020-05-10:
> Ah, makes sense. In order to do what you want, here is how I think about it: The state of the agent at time `t` can be interpreted as the transformation from its local coordinate frame to world coordinates. The delta in position/orientation between two subsequent time steps is the position/orientation of the agent at time `t+1` in the local coordinate frame defined by the state at time `t`. This can be computed by applying in inverse of the transformation defined by the state at time `t` to the state at time `t+1`.
> 
> So,
> 
> ```python
> agent_state1 = ... # Agent state at time t
> agent_state2 = ... # Agent state at time t+1
> 
> state_delta = ... # This is position/orientation of the state at time t+1 in t's coordinates,
> # AKA the delta between t and t+1
> 
> state_delta.rotation = agent_state1.rotation.inverse() * agent_state2.rotatiton
> state_delta.position = (
>     quaternion_rotate_vector(
>         agent_state1.rotation.inverse(), 
>         agent_state2.position,
>     ) 
>     - quaternion_rotate_vector(
>         agent_state1.rotation.inverse(), 
>         agent_state1.position,
>     )
> )
> ```

@erikwijmans Hi, I totally agree with your logic. I think there is a typo and it should be:
```python
quaternion_rotate_vector(
        agent_state2.rotation.inverse(), 
        agent_state2.position,
    ) 
```

Another thing is since we the `FORWARD` is along negative-z, we may need to reverse the sign of delta along z axis. The delta on x axis does not have this problem.

Xiaoming-Zhao, 2020-05-10:
> We may want to consider writing a utility function for this.
> […](#)

@Skylion007 Hi, do I need to send a PR?

dhruvbatra, 2020-05-10:
That would be very welcome!

erikwijmans, 2020-05-10:
No,
```
quaternion_rotate_vector(
        agent_state1.rotation.inverse(), 
        agent_state2.position,
    )
```
is not a typo.  That is the way to calculate the position of `agent_state2` with respect to the coordinate frame defined by `agent_state1`.

erikwijmans, 2020-05-10:
Looks like it is correct in the PR.

closed 2020-05-10 by erikwijmans