facebookresearch/habitat-lab #408: Collision mechanics for flying agents

smorad, 2020-05-21:
## ❓ Questions and Help

Collision detection using the navmesh works quite well for ground agents. How can we do collision detection for flying agents? From what I can tell, a large change in terrain height forms  a navmesh boundary. Going past this boundary results in a collision. When flying, we should be able to cross these boundaries without collision.  E.g. flying over (or maybe under) tables and chairs.

Is there a way to see if the agent is touching the rendered mesh? Is there something more advanced I could be doing?

erikwijmans, 2020-05-22:
You may find the discussion here helpful: https://github.com/facebookresearch/habitat-sim/issues/345#issuecomment-554472393

aclegg3, 2022-03-01:
For more detailed discussion of Bullet physics functionality in Habitat, check out [the ECCV 202 tutorial series](https://aihabitat.org/tutorial/2020/), specifically the `Habitat-Sim for Interaction` section.

closed 2022-03-01 by aclegg3