facebookresearch/habitat-lab #411: Segmentation fault (core dumped) - Ubuntu 20.04

ulysse-bouchet, 2020-05-29:
## Configuration
Ubuntu 20.04
Conda 4.8.2
Python 3.6.10

## Problem
I'm trying to install **habitat api** following [these instructions](https://github.com/facebookresearch/habitat-api#installation).

Everything works until step **4**, the installation and testing of **habitat sim** works well, no issues.

But when I try to run `python examples/example.py`, `python examples/benchmark.py`, or the code [here](http://aihabitat.org/docs/habitat-api/quickstart.html), it just prints `Segmentation fault (core dumped)`.

The output for `python setup.py test` is a bit different, but ends the same way :
`
running pytest
Searching for pytest-mock
Best match: pytest-mock 3.1.0
Processing pytest_mock-3.1.0-py3.6.egg

Using /home/juungko/habitat/habitat-api/.eggs/pytest_mock-3.1.0-py3.6.egg
running egg_info
writing habitat.egg-info/PKG-INFO
writing dependency_links to habitat.egg-info/dependency_links.txt
writing requirements to habitat.egg-info/requires.txt
writing top-level names to habitat.egg-info/top_level.txt
reading manifest file 'habitat.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat.egg-info/SOURCES.txt'
running build_ext
======================================== test session starts =========================================
platform linux -- Python 3.6.10, pytest-5.4.2, py-1.8.1, pluggy-0.13.1
rootdir: /home/juungko/habitat/habitat-api, inifile: setup.cfg, testpaths: test
plugins: mock-3.1.0, cov-2.9.0
collecting ... Fatal Python error: Segmentation fault

Current thread 0x00007fade7658740 (most recent call first):
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/llvmlite/binding/ffi.py", line 113 in __call__
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/llvmlite/binding/targets.py", line 60 in get_host_cpu_features
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/codegen.py", line 881 in get_host_cpu_features
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/codegen.py", line 782 in _get_host_cpu_features
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/codegen.py", line 846 in _customize_tm_features
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/codegen.py", line 652 in _init
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/codegen.py", line 645 in __init__
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/cpu.py", line 47 in init
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/compiler_lock.py", line 32 in _acquire_compile_lock
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/base.py", line 259 in __init__
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/registry.py", line 31 in _toplevel_target_context
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/utils.py", line 332 in __get__
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/registry.py", line 47 in target_context
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/dispatcher.py", line 670 in __init__
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/decorators.py", line 189 in wrapper
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/decorators.py", line 173 in jit
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/core/decorators.py", line 236 in njit
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/typed/typeddict.py", line 22 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/typed/__init__.py", line 1 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/numba/__init__.py", line 298 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
[... 44 lines cut ...]
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "/home/juungko/habitat/habitat-api/test/test_baseline_agents.py", line 11 in <module>
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/assertion/rewrite.py", line 152 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/py/_path/local.py", line 701 in pyimport
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/python.py", line 511 in _importtestmodule
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/python.py", line 443 in _getobj
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/python.py", line 261 in obj
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/python.py", line 459 in _inject_setup_module_fixture
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/python.py", line 446 in collect
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/runner.py", line 264 in <lambda>
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/runner.py", line 244 in from_call
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/runner.py", line 264 in pytest_make_collect_report
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 87 in <lambda>
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 93 in _hookexec
  ...
Segmentation fault (core dumped)
`

erikwijmans, 2020-05-29:
Seems similar to these issues: 
https://github.com/numba/numba/issues/5071
https://github.com/numba/llvmlite/issues/523

ulysse-bouchet, 2020-06-02:
Thanks.

I tried to install LLVM as said in these links, and the situation is slightly better but still not working.
Now, the different outputs end by 'Aborted (core dumped)'.

Output for `python examples/example.py` :
```
2020-06-02 16:22:11,077 Initializing dataset PointNav-v1
2020-06-02 16:22:11,389 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0602 16:22:11.475940  7741 Simulator.cpp:96] Loading navmesh from data/scene_datasets/habitat-test-scenes/van-gogh-room.navmesh
I0602 16:22:11.493600  7741 Simulator.cpp:98] Loaded.
I0602 16:22:11.493722  7741 SceneGraph.h:92] Created DrawableGroup: 
Renderer: GeForce GTX 1050/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 440.64
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
F0602 16:22:12.123987  7741 ResourceManager.cpp:483] Check failed: Cr::Utility::Directory::exists(physicsFilename) 
*** Check failure stack trace: ***
Aborted (core dumped)
```

Output for `python examples/benchmark.py` (pretty much the same):
```
2020-06-02 16:22:43,322 Initializing dataset PointNav-v1
2020-06-02 16:22:43,488 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0602 16:22:43.493085  7769 Simulator.cpp:96] Loading navmesh from data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
I0602 16:22:43.493177  7769 Simulator.cpp:98] Loaded.
I0602 16:22:43.493196  7769 SceneGraph.h:92] Created DrawableGroup: 
Renderer: GeForce GTX 1050/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 440.64
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
[... 57 lines cut ...]
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 87 in <lambda>
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/main.py", line 272 in pytest_runtestloop
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 87 in <lambda>
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/main.py", line 247 in _main
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/main.py", line 191 in wrap_session
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/main.py", line 240 in pytest_cmdline_main
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 87 in <lambda>
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/config/__init__.py", line 125 in main
  File "/home/juungko/habitat/habitat-api/.eggs/pytest_runner-5.2-py3.6.egg/ptr.py", line 220 in run_tests
  File "/home/juungko/habitat/habitat-api/.eggs/pytest_runner-5.2-py3.6.egg/ptr.py", line 209 in run
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974 in run_command
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 955 in run_commands
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/distutils/core.py", line 148 in setup
  File "/home/juungko/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/__init__.py", line 144 in setup
  File "setup.py", line 99 in <module>
Aborted (core dumped)
```

erikwijmans, 2020-06-02:
Seems like `enable_physics` is getting set to True, is that intended?

ulysse-bouchet, 2020-06-03:
It wasn't intended. 
I changed it and it seems to work, even though some tests fail : 

Output for `python examples/example.py` (seems to work) : 
```
2020-06-03 16:23:48,200 Initializing dataset PointNav-v1
2020-06-03 16:23:48,354 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0603 16:23:48.359730 16056 Simulator.cpp:96] Loading navmesh from data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
I0603 16:23:48.359822 16056 Simulator.cpp:98] Loaded.
I0603 16:23:48.359843 16056 SceneGraph.h:92] Created DrawableGroup: 
Renderer: GeForce GTX 1050/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 440.64
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
I0603 16:23:48.523753 16056 ResourceManager.cpp:1391] Importing Basis files as BC7
W0603 16:23:49.591574 16056 Simulator.cpp:199] :
---
 The active scene does not contain semantic annotations. 
---
I0603 16:23:49.591840 16056 simulator.py:146] Loaded navmesh data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
2020-06-03 16:23:49,593 Initializing task Nav-v0
Environment creation successful
Agent stepping around inside environment.
Episode finished after 3 steps.
I0603 16:23:49.604430 16056 SemanticScene.h:40] Deconstructing SemanticScene
I0603 16:23:49.604442 16056 SceneManager.h:24] Deconstructing SceneManager
I0603 16:23:49.604445 16056 SceneGraph.h:25] Deconstructing SceneGraph
I0603 16:23:49.604454 16056 RenderTarget.h:51] Deconstructing RenderTarget
I0603 16:23:49.604482 16056 Sensor.h:80] Deconstructing Sensor
I0603 16:23:49.616307 16056 Renderer.cpp:33] Deconstructing Renderer
I0603 16:23:49.616343 16056 WindowlessContext.h:16] Deconstructing WindowlessContext
I0603 16:23:49.616345 16056 WindowlessContext.cpp:245] Deconstructing GL context
I0603 16:23:49.629195 16056 Simulator.cpp:39] Deconstructing Simulator
```

[Output](https://pastebin.com/TkrLBu1m) for `python setup.py test` (some tests fail) (too long to put it here)

Output for `python examples/benchmark.py` (seems to work, but 'success' is always 0, is that normal ?) :
```
2020-06-03 16:40:02,737 Initializing dataset PointNav-v1
2020-06-03 16:40:02,918 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0603 16:40:03.005802 16832 Simulator.cpp:96] Loading navmesh from data/scene_datasets/habitat-test-scenes/van-gogh-room.navmesh
I0603 16:40:03.005852 16832 Simulator.cpp:98] Loaded.
I0603 16:40:03.005874 16832 SceneGraph.h:92] Created DrawableGroup: 
Renderer: GeForce GTX 1050/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 440.64
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
I0603 16:40:03.650545 16832 ResourceManager.cpp:1391] Importing Basis files as BC7
W0603 16:40:04.512322 16832 Simulator.cpp:199] :
---
 The active scene does not contain semantic annotations. 
---
I0603 16:40:04.512528 16832 simulator.py:146] Loaded navmesh data/scene_datasets/habitat-test-scenes/van-gogh-room.navmesh
2020-06-03 16:40:04,513 Initializing task Nav-v0
distance_to_goal: 2.218
success: 0.000
spl: 0.000
I0603 16:40:07.703877 16832 SemanticScene.h:40] Deconstructing SemanticScene
I0603 16:40:07.703896 16832 SceneManager.h:24] Deconstructing SceneManager
I0603 16:40:07.703898 16832 SceneGraph.h:25] Deconstructing SceneGraph
I0603 16:40:07.703913 16832 RenderTarget.h:51] Deconstructing RenderTarget
I0603 16:40:07.703950 16832 Sensor.h:80] Deconstructing Sensor
I0603 16:40:07.706326 16832 Renderer.cpp:33] Deconstructing Renderer
I0603 16:40:07.706342 16832 WindowlessContext.h:16] Deconstructing WindowlessContext
I0603 16:40:07.706346 16832 WindowlessContext.cpp:245] Deconstructing GL context
I0603 16:40:07.708226 16832 Simulator.cpp:39] Deconstructing Simulator
```

erikwijmans, 2020-06-03:
All looks fine.  There is a pytorch error where it seems like a convolution is not outputting a contiguous tensor (which is odd).  The other error is from running out of GPU memory: ` RuntimeError: CUDA out of memory. Tried to allocate 18.00 MiB (GPU 0; 1.96 GiB total capacity; 604.65 MiB already allocated; 15.44 MiB free; 648.00 MiB reserved in total by PyTorch)` -- 2 GB won't be enough for the ResNet50 model.

ulysse-bouchet, 2020-06-04:
Thanks.
I'll soon get a new laptop with a better GPU, it should be ok with it.

rpartsey, 2020-08-05:
Hi @ulysse-bouchet
While setting up the habitat-sim & habitat-api (see [CI instructions ](https://github.com/facebookresearch/habitat-api/blob/master/.circleci/config.yml)), I faced up with a similar error to what you described in the **Problem** section above.

I run `numba -s` to verify that the _numba_ was installed successfully. 
But `Segmentation fault (core dumped)` message  was printed instead of information about my system capabilities. 

Then I run `pip uninstall numba` to uninstalled _numba_ .
And installed it again but using  _conda_ `conda install numba` instead of _pip_. (I also remember that this command updated one of _numba_'s dependencies, but I don't remember which exactly).

After that, I was able to successfully run all tests and build documentation pages.

closed 2020-08-05 by Skylion007