facebookresearch/habitat-lab #412: access velocity from a moving agent

xiangyuanhang, 2020-06-04:
## ❓ Questions and Help
Hi,
I am trying to use shortest_path_follower.py to collect some data from the navigation trajectories. What I want to collect is the agent's position and velocity within a navigation.
This is the code I used to collect the data, which is modified from shortest_path_follower.py
```
while not env.habitat_env.episode_over:
            best_action = follower.get_next_action(
                env.habitat_env.current_episode.goals[0].position
            )
            if best_action is None:
                break

            observations, reward, done, info = env.step(best_action)
            print('position:',env.habitat_env.sim.get_agent_state().position, 
            'velocity:', env.habitat_env.sim.get_agent_state().velocity, 
            'rotation:', env.habitat_env.sim.get_agent_state().rotation,
            'angularVelocity:', env.habitat_env.sim.get_agent_state().angular_velocity)
```

and the result I get is 
```
position: [-11.957363    0.1970073  -4.0664253] velocity: [0. 0. 0.] rotation: quaternion(0.334518194198608, 0, 0.942389369010925, 0) angularVelocity: [0. 0. 0.]
position: [-11.957363    0.1970073  -4.0664253] velocity: [0. 0. 0.] rotation: quaternion(0.415379881858826, 0, 0.90964812040329, 0) angularVelocity: [0. 0. 0.]
position: [-12.146288    0.1970073  -3.9026954] velocity: [0. 0. 0.] rotation: quaternion(0.415379881858826, 0, 0.90964812040329, 0) angularVelocity: [0. 0. 0.]
position: [-12.335213    0.1747377  -3.7389655] velocity: [0. 0. 0.] rotation: quaternion(0.415379881858826, 0, 0.90964812040329, 0) angularVelocity: [0. 0. 0.]

```
We can see that the velocity and angular_velocity are both [0, 0, 0].
I suspect this is because that the action is taken step by step, thus the velocity between each step is zero.
So how could I get a non-zero velocity from a moving agent?

Thank you so much!

erikwijmans, 2020-06-05:
Those fields were added perhaps a little prematurely.  We currently don't natively support continuous control (it can be done, but it's not natively supported).  Those field really only make sense for continuous control, thus they are all zero currently.

I am not sure what your use-case is, but for discreet actions, `v_t = p_{t+1} - p_{t}` is a sensible definition of velocity and may work.

xiangyuanhang, 2020-06-05:
Thank you for your reply.
Your proposal is helpful.
To estimate velocity from discret action, we need to know the time spent at every step.
from [https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/simulator.py#L246](url)
The time spent is 1.0/60.0 min, which is 1s? （I am not sure of the unit）

erikwijmans, 2020-06-05:
Unless you have physics simulation enabled, that value doesn't mean anything and there is no concept of time.

closed 2021-02-07 by mathfac