facebookresearch/habitat-lab #414: how to set the agent height from the yaml file ?

ronjamoller, 2020-06-16:
## ❓ Questions and Help
Hi, i am trying to train with a simulated agent that is of a certain height - if i add the height parameter in the yaml, like this (height is indented under the agent_0, just like radius): 
ENVIRONMENT:
       MAX_EPISODE_STEPS: 500
SIMULATOR:
       FORWARD_STEP_SIZE: 0.1
       TURN_ANGLE: 5
       AGENT_0:
             SENSORS: ["RGB_SENSOR", "DEPTH_SENSOR"]
             RADIUS: 0.18
             HEIGHT: 0.5
      HABITAT_SIM_V0:
             GPU_DEVICE_ID: 0
and initialize an environment like that then it does not work, the agent is at the default height. If instead i create a separate simulation config and separately reinitialize the env._sim._sim `env._sim._sim.reconfigure(sim_cfg)` then it works. Unfortunately, i would like to avoid this workaround since 1. the double configs will surely lead to inconsistencies and 2. I also want to set defined step sizes and turn angles, this should ideally be done from the yaml and not with another config to avoid complications and the reinit resets those values.
Please let me know if there is anything i am missing (new stable commit was pulled last week by the way).
thank you

erikwijmans, 2020-06-16:
What I believe is happening is you are trying to adjust the location of the sensor off the floor via `HEIGHT`.  `HEIGHT` refers to the height of the cylinder used to model the agent for collisions.  Sensor positions are specified differently, that is changed with the `POSITION` field like here: https://github.com/facebookresearch/habitat-challenge/blob/master/configs/challenge_pointnav2020_v2.local.rgbd.yaml#L18

ronjamoller, 2020-06-17:
Thank you, that works perfectly.

closed 2020-06-17 by ronjamoller