facebookresearch/habitat-lab #415: Issue with object insertion after update

ronjamoller, 2020-06-16:
## ❓ Questions and Help
Hi, i recently (less than 2 weeks ago) cloned the newest stable version and now tried to insert objects as before (i am using the banana object that is provided). In the new version instead of the banana (and other objects) a cone is inserted, in my old backup it works fine. I have not found a similar issue here and i hesitate to report it as a bug since i have not exactly tested it extensively. I have checked the object files in blender and they look fine. Below are image examples of the insertion working in the old version and not working in the new version. Maybe the insertion procedure changed and this is some sort of placeholder object ?
[image]
[image]

aclegg3, 2020-06-16:
Hey @ronjamoller, looks like you may be using hard-coded object template ids to add your object (e.g. `add_object(0)` ). Recent updates to habitat-sim have introduced primitive objects which may be placed into the library automatically **before** file-loaded objects. 

You can instead [add your objects by string key](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#add_object_by_handle) and [query object template keys via substring ](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#get_template_handles) instead for more precise object instancing scripts. 

Note that the API for loading/managing object templates is in active development, so expect more changes and improvements in the near future. We'll also add some tutorials to clarify this process on the [Habitat-sim docs site](https://aihabitat.org/docs/habitat-sim/index.html) in the coming weeks. 😀

ronjamoller, 2020-06-16:
thank you! that makes it a lot easier.

ronjamoller, 2020-06-17:
The insertion now works with this tactic:
`handle = env._sim._sim.get_template_handles('keyword')
        object_id = env._sim._sim.add_object_by_handle(handle[0])` 
Unfortunately the object seems to be rotated by 90 degrees in case someone does not have primitive shapes like spheres it will show, I guess i will keep pulling the newest changes and see how everything develops and keep checking - very curious to see where the development goes :)

closed 2020-06-17 by ronjamoller