facebookresearch/habitat-lab #416: Browser demo doesn't seem to work

guydav, 2020-06-17:
## 🐛 Bug

The browser demo at https://aihabitat.org/demo doesn't seem to work for me (and another person I asked to test). Running latest chrome on OSX 10.15
 
## Command

## To Reproduce

Go to https://aihabitat.org/demo . It gets stuck on Downloading... 0/4. 

## Expected behavior

A working demo. 

## Additional context

OSX 10.15, Chrome 83.0.4103.97.

guydav, 2020-06-17:
On a related note: the installation link from the [quickstart page](https://aihabitat.org/docs/habitat-api/quickstart.html) points to the issues page, rather than anywhere useful.

dhruvbatra, 2020-06-18:
Can't replicate. Works fine for me on Chrome. And another person I asked to test. We're at a stalemate :-). 

We'll fix the quickstart page.

guydav, 2020-06-18:
Fascinating. I asked another person (latest Chrome and Windows 10) to test, same result. We both receive errors with downloading the assets:

```
Access to XMLHttpRequest at 'https://habitat-resources.s3.amazonaws.com/data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh' from origin 'http://aihabitat.org' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.
Uncaught Loading data file "https://habitat-resources.s3.amazonaws.com/data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh" failed.
GET https://habitat-resources.s3.amazonaws.com/data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh net::ERR_FAILED
Access to XMLHttpRequest at 'https://habitat-resources.s3.amazonaws.com/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb' from origin 'http://aihabitat.org' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.
Uncaught Loading data file "https://habitat-resources.s3.amazonaws.com/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb" failed.
GET https://habitat-resources.s3.amazonaws.com/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb net::ERR_FAILED
```

[image]

Could it perhaps be that you have the assets already? If I try to access the files directly, for example `https://habitat-resources.s3.amazonaws.com/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`, it downloads just fine, so it seems it's something with the CORS policy perhaps? It's been forever since I've had to debug this sort of thing...

dhruvbatra, 2020-06-18:
@apsdehal -- do you know what might be causing this?

apsdehal, 2020-06-18:
@guydav can you load the website on https?
@dhruvbatra this needs to be fixed on our side. Http should automatically redirect to https. OSS team can help with that.

guydav, 2020-06-18:
@apsdehal Indeed, it works if I point to it https. Thank you!

mathfac, 2020-06-18:
@apsdehal, if we change links to `https` will it fix the issue?

apsdehal, 2020-06-18:
We need to understand how user reached the http link and what is the source. I am not sure if fixing http to https will fix the issue as I don't know how internal setup for cname and nameservers is setup for aihabitat.org. Maybe @guydav can tell us how he reached the http link.

mathfac, 2020-06-18:
I see.

guydav, 2020-06-18:
@apsdehal aihabitat.org doesn't enforce https (rather than http). I admit I don't remember where the link to the demo is, though.

Skylion007, 2020-08-28:
We changed some settings on our website to enforce strict HTTPS. Reopen if the issue persists.

closed 2020-08-28 by Skylion007