facebookresearch/habitat-lab #417: Objects and actions in environments, Habitat vs. AI2-THOR

guydav, 2020-06-17:
[if there's a more appropriate place for this sort of long-form question, please do point me to it]

I'm trying to understand what the relationship between Habitat and AI2-THOR is, particularly in how the two of them deal with objects and actions.

The documentation in the video mentions Habitat is trying to standardize over several simulators, one of which is THOR. However, the Habitat simulator and THOR seem to take somewhat different approaches to object interactions. [THOR's objects](https://ai2thor.allenai.org/ithor/documentation/objects/object-types/) seem to center around having particular actionable properties, defining what can be done with each object. Habitat has [rigid physics support](https://aihabitat.org/docs/habitat-sim/rigid-object-tutorial.html), but limited support to interacting with objects in the environments. It also seems that the tasks in the Habitat Challenge right now focus on navigation, rather than interaction.

Is there a way in Habitat to interact with objects in a more abstract fashion, or is the only current API to define velocities/forces/torques acting on objects? 

Alternatively, can the Habitat API be used with THOR as the simulator? What would I gain by doing that? 

If there's anything I should read that might shed light on this, I'd love to read it.

dhruvbatra, 2020-06-18:
> The documentation in the video mentions Habitat is trying to standardize over several simulators, one of which is THOR. 

Thanks for pointing this out. Our intent was to convey that we're trying to standardize the embodied AI software stack and learn from the best aspects of existing efforts (such as AI2 Thor, and others). We should be more clear in that communication. 

> Habitat has rigid physics support, but limited support to interacting with objects in the environments. It also seems that the tasks in the Habitat Challenge right now focus on navigation, rather than interaction.

That is true right now, but Habitat is an ongoing multi-year multi-institution effort. Our ICCV19 paper and 2019/20 challenges focused on navigation, but interactive tasks are coming. We've been working on the core capabilities in the simulator (e.g. integration with Bullet, support for kinematic control of LoCoBot, etc). Tasks will show up in H-API once we feel the right set of capabilities exist in H-Sim. Tentative timeline: 2-3 months for the first interactive task (without a stable API or recommendation for others to start using it); 6-10 months for a more stable interactive tasks API. Of course, this is an open source project -- you'll see as we add things and are free to dive in to the degree of your comfort level. 

> Is there a way in Habitat to interact with objects in a more abstract fashion, or is the only current API to define velocities/forces/torques acting on objects?

That's the only option right now. We're happy to collaborate if someone wants to implement an abstract interaction API. 

> an the Habitat API be used with THOR as the simulator? 

No. We did design H-API specifically to be abstracted away from the choice of a simulator. But due to practical considerations, fleshing this out has never risen to high enough priority.

guydav, 2020-06-18:
Thank you @dhruvbatra! That all makes a lot of sense. I'll continue to follow up on the progress you make with interactive tasks -- I think we need them for the project I have in mind right now, and I don't know if I have the bandwidth to develop an abstract API, but I'll be sure to keep track of what Habitat offers.

mathfac, 2020-06-18:
Hi @guydav,
Regarding abstract API it can look as easy as adding new TaskAction like [AnswerQuestion in Embodied Question Answering Task](https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/eqa/eqa.py#L179) that will change virtual object status.
Schematically action can look like:
```
@registry.register_task_action
class ObjectAction(Action):
    name: str = "OBJECT_ACTION"

    def __init__(self, *args: Any, sim, dataset, **kwargs: Any) -> None:
        self._sim = sim
        self._dataset = dataset

    def reset(self, task: InteractiveTask, *args: Any, **kwargs: Any) -> None:
        task.object_states = {}
        return

    def step(
        self, *args: Any, object_id: int, action_id: int task: InteractiveTask, **kwargs: Any
    ) -> Dict[str, Observations]:
        task.object_states[object_id] = action_on_object(object_id, action_id, self._sim)
        return self._sim.get_observations_at()

    @property
    def action_space(self) -> Space:
        """Answer expected to be single token.
        """
        return spaces.Dict(
            {
                "object_id": spaces.Discrete(
                    self._dataset.max_object_id.get_size()
                ),
                "action_id": spaces.Discrete(
                    self._dataset.object_actions.get_size()
                )
            }
        )
```

guydav, 2020-06-19:
@mathfac Thank you! I think what we have in mind is a bit more complicated, as we hope to also change the world the agent observes in response to the action, but I'll take a look at it!-

rpartsey, 2022-09-01:
Stale.
Habitat-lab now support interaction with the objects.

closed 2022-09-01 by rpartsey