facebookresearch/habitat-lab #419: shortest path to the object goal

yimengli46, 2020-06-19:
## ❓ Questions and Help
Hi, I was wondering if there is a direct way to get the shortest distance to the target object.
I tried the shortest path follower with one of the target object view but it doesn't seem to work.

erikwijmans, 2020-06-20:
You can see how the distance to goal measure works here: https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/nav/nav.py#L953. The `"VIEW_POINTS"` mode is used for objectnav.

yimengli46, 2020-06-20:
Thanks for the answer. 
I have a follow-up question, not sure if it's solvable. 
So this goal measure will tell the distance from agent current state to one of the goal view.
Is there a way to get the geodesic distance after taking each action?
Because calling env.step(action) you can compute the distance for only one action, not all the actions. In the DD-PPO baseline, the reward computing is only for the taken action, not all the actions.

erikwijmans, 2020-06-21:
This is not something that is neatly wrapped up into an existing functionality, but is completely doable.  While it is in cpp, it may be useful to look at the primitive selection part of the local planner/shortest path follower -- here: https://github.com/facebookresearch/habitat-sim/blob/master/src/esp/nav/GreedyFollower.cpp#L81 (maybe also be useful to look at the python code that drives it: https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/nav/greedy_geodesic_follower.py) -- as all the necessary parts are bound to python.

yimengli46, 2020-07-05:
Hi Erik, sorry to bother you again. I looked at the code you suggested me carefully. I would like to know if there is a way to call greedy_geodesic_follower for the objectnav task? It seems to me that greedy_geodesic_follower is designed for a single goal location. However, objectnav has multiple viewpoints for one target object.

erikwijmans, 2020-07-05:
No, there isn't.  This is done on purpose as the multi-goal geodesic distance computation is considerably slower than single goal.  You can get the closest point by first running a multi-goal geodesic distance computation/path find, the `.points` field is the list of way points to the closest goal point, with `.points[-1]` being the closest goal.  You can then give that point to the follower.

yimengli46, 2020-07-05:
Thanks, it's very helpful.

closed 2020-07-05 by yimengli46