facebookresearch/habitat-lab #427: example.py is not working: AttributeError: 'Configuration' object has no attribute 'SCENE'

linhdoan8, 2020-07-12:
## ❓ Questions and Help
Hi all, 

I have installed habitat-sim and habitat-api according to the respective instructions inside the same conda environment. Now I have problems when running example.py in habitat-api.
When in habitat-sim running 

`python examples/example.py `

is working fine but when in habitat-api running the same is giving this error:

```
2020-07-13 00:27:51,957 Initializing dataset PointNav-v1
2020-07-13 00:27:52,187 initializing sim Sim-v0
Traceback (most recent call last):
  File "examples/example.py", line 30, in <module>
    example()
  File "examples/example.py", line 14, in example
    config=habitat.get_config("configs/tasks/pointnav.yaml")
  File "/home/linh/habitat/habitat-api/habitat/core/env.py", line 105, in __init__
    id_sim=self._config.SIMULATOR.TYPE, config=self._config.SIMULATOR
  File "/home/linh/habitat/habitat-api/habitat/sims/registration.py", line 19, in make_sim
    return _sim(**kwargs)
  File "/home/linh/habitat/habitat-api/habitat/sims/habitat_simulator/habitat_simulator.py", line 185, in __init__
    super().__init__(self.sim_config)
  File "<attrs generated init habitat_sim.simulator.Simulator>", line 9, in __init__
  File "/home/linh/habitat/habitat-sim/habitat_sim/simulator.py", line 68, in __attrs_post_init__
    self.reconfigure(config)
  File "/home/linh/habitat/habitat-api/habitat/sims/habitat_simulator/habitat_simulator.py", line 291, in reconfigure
    is_same_scene = habitat_config.SCENE == self._current_scene
AttributeError: 'Configuration' object has no attribute 'SCENE'
[1]    21923 segmentation fault (core dumped)  python examples/example.py
```

I have also tried to add habitat-sim to PYTHONPATH and used ./build.sh. Also, I ensured that I installed both sim and api in the same conda environment and executed python while in the same. Please help and thanks!

erikwijmans, 2020-07-12:
The current master version of habitat-api requires a version of habitat-sim after this PR: https://github.com/facebookresearch/habitat-sim/pull/644

linhdoan8, 2020-07-13:
Thanks! I just switched to stable branch and now its working fine.

closed 2020-07-13 by linhdoan8