facebookresearch/habitat-lab #428: Error Training: tag v0.1.4

AAdeleye, 2020-07-14:
## ❓ Questions and Help

Hello, I get the following error below when trying to train using the mp3d dataset. Training with the ppo_pointnav_example worked fine and it seems to fail at different points doing the updates. It makes it to around the 12000s though. 

[image]

Due to some complications, I am using the habitat docker but running a separate conda env with version 0.1.4. I do not think that should be an issue but please let me know if it is. 

The closes issue I have found is [here](https://github.com/facebookresearch/habitat-api/issues/311)

mathfac, 2020-07-15:
Hi
1. What command line did you use for training?
2. Was there any changes to the code?
3. Did you use habitat sim of the same version?

Thank you!

AAdeleye, 2020-07-20:
Hi,

It seems the issue was the same as the issue I tagged above. Running [the script](https://gist.github.com/erikwijmans/e4410f0e12facb87890e919aa264e3fe) fixed the problem. 

I am going to go ahead and close this, thank you.

closed 2020-07-20 by AAdeleye