facebookresearch/habitat-lab #429: Error: 'Configuration' object has no attribute 'SCENE'

mukulkhanna, 2020-07-15:
## 🐛 Bug

I am unable to train the PPO pointnav baseline with the test-scene dataset. Here is the main part of the error message ->

``` 
File "/home/mukul/habitat-api/habitat/sims/habitat_simulator/habitat_simulator.py", line 291, in reconfigure
    is_same_scene = habitat_config.SCENE == self._current_scene
AttributeError: 'Configuration' object has no attribute 'SCENE'
```

## Command

```
python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type train
```

## To Reproduce

Steps to reproduce the behavior:

Same as mentioned in habitat_baselines [README](https://github.com/facebookresearch/habitat-api/tree/master/habitat_baselines/README.md).
1. Download test data for PPO pointnav baseline example.
2. Run trainer.

<details>
<summary>Complete error log</summary>
```
python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type train
2020-07-15 06:29:36,575 config: BASE_TASK_CONFIG_PATH: configs/tasks/pointnav.yaml
CHECKPOINT_FOLDER: data/new_checkpoints
CHECKPOINT_INTERVAL: 50
CMD_TRAILING_OPTS: []
ENV_NAME: NavRLEnv
EVAL:
  SPLIT: val
  USE_CKPT_CONFIG: True
EVAL_CKPT_PATH_DIR: data/new_checkpoints
FORCE_BLIND_POLICY: False
LOG_FILE: train.log
LOG_INTERVAL: 10
NUM_PROCESSES: 1
NUM_UPDATES: 10000
ORBSLAM2:
  ANGLE_TH: 0.2617993877991494
  BETA: 100
  CAMERA_HEIGHT: 1.25
  DEPTH_DENORM: 10.0
  DIST_REACHED_TH: 0.15
  DIST_TO_STOP: 0.05
  D_OBSTACLE_MAX: 4.0
  D_OBSTACLE_MIN: 0.1
  H_OBSTACLE_MAX: 1.25
  H_OBSTACLE_MIN: 0.375
  MAP_CELL_SIZE: 0.1
  MAP_SIZE: 40
  MIN_PTS_IN_OBSTACLE: 320.0
  NEXT_WAYPOINT_TH: 0.5
  NUM_ACTIONS: 3
  PLANNER_MAX_STEPS: 500
  PREPROCESS_MAP: True
  SLAM_SETTINGS_PATH: habitat_baselines/slambased/data/mp3d3_small1k.yaml
  SLAM_VOCAB_PATH: habitat_baselines/slambased/data/ORBvoc.txt
RL:
  DDPPO:
    backbone: resnet50
    distrib_backend: GLOO
    num_recurrent_layers: 2
    pretrained: False
    pretrained_encoder: False
    pretrained_weights: data/ddppo-models/gibson-2plus-resnet50.pth
[... 272 lines cut ...]
  File "/home/mukul/habitat-api/habitat/core/vector_env.py", line 534, in __del__
    self.close()
  File "/home/mukul/habitat-api/habitat/core/vector_env.py", line 416, in close
    write_fn((CLOSE_COMMAND, None))
  File "/home/mukul/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/home/mukul/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/home/mukul/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe
(habitat) mukul@instance-1:~/habitat-api$ python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type train
```
</details>

## Expected behavior

Expected training to start.

## Additional context

-  I noticed that this wasn't an issue and I was able to train before the #418 PR was merged.
- I am also facing this issue when I try to train a custom baseline trainer that I have implemented in #374.

Thank you for your help!

mukulkhanna, 2020-07-15:
Figured it out - I hadn't pulled habitat_sim sim changes. Thank you.

closed 2020-07-15 by mukulkhanna

LORRY137, 2023-11-23:
> Figured it out - I hadn't pulled habitat_sim sim changes. Thank you.

Could you please tell me how was this issue finally resolved?Thank you!

mukulkhanna, 2023-11-23:
As mentioned, I hadn't pulled habitat_sim changes. Your issue could be due to a version mismatch between hab-lab and hab-sim. Some more info in the issue mentioned above.