facebookresearch/habitat-lab #432: How to run Objectnav task using Gibson datasets?

ybgdgh, 2020-07-26:
## ❓ Questions and Help
Hello! I want to run objectnav task in family scene and I find that the mp3d contains many other scene types. But gibson datasets are suitable for my need. So I want to know if I can run objectnav task in gibson? How to generate the objectnav task dataset (train/val) for gibson environment? I have get the semantic information from 3DSceneGraph for gibson and generate the semantic.ply, can the object annotation show in top-down map?
Thanks!

mathfac, 2020-07-26:
Hi,
> ! I want to run objectnav task in family scene
Can you elaborate more? What is family scene?

Correct, first step is to get semantic.ply and bounding boxes for Gibson
dataset.
Second is to generate ObjectNav episodes based on it. The dataset
generation code is not publicly available. But if you confirm that first
step is working we can prioritize and merge the dataset generation code
into master.
Thank you!

ybgdgh, 2020-07-26:
I am sorry for annoying you about family scene and its not important. I follow the [habitat-smi/datasets](https://github.com/facebookresearch/habitat-sim#datasets) to get the semantic.ply ,ids and scn, but i dont kown how to use them. I am looking forward to your publication about the dataset generation code for objectnav task.
Thank you!

closed 2020-08-07 by ybgdgh

yimengli46, 2021-12-06:
Hi, I'm wondering if you can get the semantic observations from Gibson?

dhruvbatra, 2021-12-06:
https://github.com/facebookresearch/habitat-sim/issues/1080

Roberyan, 2025-03-16:
> Second is to generate ObjectNav episodes based on it. The dataset generation code is not publicly available. But if

May I learn how to get the episode generation code for gibson object goal navigation v1.1? I find the provided training data completely empty and have no idea how to use it