facebookresearch/habitat-lab #434: Semantic_annotations from VectorEnv Class

AAdeleye, 2020-08-04:
Hello, I have a question similar to this. 

I want to use the 'env.sim.semantic_annotations()' method listed [here](https://aihabitat.org/docs/habitat-api/habitat-api-demo.html).

I am trying to do something along the lines of `self.envs._connection_write_fns[0](??, none)` to get a single env.  I don't think this is possible anymore so I was wondering how best to go about this? 

I wanted the semantic_annotations after a signal step - i.e to change the following lines:
 `self.envs_connection_write_fns[0](STEP_COMMAND, action)` 
`self.env_connection_read_fns[0]()`
 so that read includes semantic_annotations. 

Thank you.

_Originally posted by @AAdeleye in https://github.com/facebookresearch/habitat-api/issues/292#issuecomment-668405138_

AAdeleye, 2020-08-04:
I found that the `def _worker_env(...)` method in the VectorEnv class uses the individual RLEnv object to perform a step when the STEP_COMMAND is given.
 
In this method, when running `scene = env._env.sim.semantic_annotations()` (where env is and RLEnv object), the scene annotation list is empty. Are there semantic annotations for RLEnvs, or is there a problem with my code?

 I am using version v0.1.4 of habitat, and the mp3d dataset. 

Thank you.

Skylark0924, 2020-12-22:
> I found that the `def _worker_env(...)` method in the VectorEnv class uses the individual RLEnv object to perform a step when the STEP_COMMAND is given.
>
> In this method, when running `scene = [...]

Hi,
Did you find the solution? 
I also have a similar confusion about how to obtain properties from an RLEnv packed in VectorEnv.

fraazor, 2021-03-23:
Hi,

dhruvbatra, 2021-03-23:
CC: @Skylion007

Skylion007, 2021-03-23:
What scene dataset are you using? Not all datasets have semantic data
associated with them.

fraazor, 2021-03-23:
I am using the .glb matterport 3d scene data set. It certainly has the semantic data I look for and I access it for any regular 
env object like this without a problem:
scene = env.sim.semantic_annotations()

I tried applying the method for a number of vector_env objects by accessing the properties but I am afraid it does not work like this:
functionnames = ["habitat_env.sim.semantic_annotations()"]*self.envs.num_envs
scene = self.envs.call(functionnames)

Thanks a lot for your quick response!

fraazor, 2021-04-18:
Has this been resolved elsewhere by any chance?

erikwijmans, 2021-04-18:
No, and it can't be without a considerable amount of work as what is returned from `habitat_env.sim.semantic_annotations()` is not pickle-able (which is required to be sent between processes).  Whatever your use for the semantic annotations is, it's simplest to do that processing in the task, the Env, a sensor, or a measure.  This will also give you parallelization of that processing.

closed 2022-09-01 by rpartsey