facebookresearch/habitat-lab #436: Invalid docs links

rpartsey, 2020-08-04:
## 🐛 Bug

Invalid links (marked as bold text) in the paragraphs:

> For installing Habitat-Sim and Habitat-API follow instructions **here**.

> ... The data for scene should also be downloaded (steps to do this are provided in the **installation instructions** of Habitat-API).

## Command

## To Reproduce

Steps to reproduce the behavior:

1. Go to https://aihabitat.org/docs/habitat-api/quickstart.html
2. Open links **here** and **installation instructions**.

## Expected behavior

Both **here** and **installation instructions** links point to https://github.com/facebookresearch/habitat-api/issues/installation seems like they should point to https://github.com/facebookresearch/habitat-api#installation instead.
## Additional context

The issue is caused by the `m.css`'s `gh` plugin. 
Whenever link contains '#' symbol it is treated as an issue link.
See https://github.com/mosra/m.css/blob/01ea5ad52eb8e85b018027a5d540f5822ebfebba/plugins/m/gh.py#L41 for more details.

mosra, 2020-08-04:
Hahah, oh well, interesting conflict (thanks for the investigation, btw!). 

I think the solution here is to replace them with full URLs, no `:gh:` anything -- attempting to resolve such conflicts in m.css would only complicate the code and won't be 100% robust anyway.

dhruvbatra, 2020-08-05:
Thanks for flagging this @rpartsey. Would you mind sending us a PR with the full link as suggested by @mosra ?

rpartsey, 2020-08-05:
@dhruvbatra sure!
I’ll create a PR today.

closed 2020-08-05 by Skylion007