facebookresearch/habitat-lab #447: Baselines ddppo evaluation fails with RuntimeError: Error(s) in loading state_dict for DDPPO

asolano, 2020-08-11:
Greetings,

I managed to train the baselines ddppo on a 4 GPU machine and now I am trying to run the evaluation mode to check performance. I am using the `stable` tag of the code.

The evaluation command is:

```shell
$ singularity exec $HOME/headless-rendering-ip.simg bash -c '
source $HOME/miniconda3/etc/profile.d/conda.sh
conda activate habitat

export GLOG_minloglevel=2
export MAGNUM_LOG=quiet

python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ddppo_pointnav.yaml --run-type eval'
```

Note Singularity is required in this environment.

The `ddppo_pointnav.yaml` task configuration is (changed to Matterport dataset and RGBD sensors):

```yaml
# BASE_TASK_CONFIG_PATH: "configs/tasks/pointnav_gibson.yaml"
BASE_TASK_CONFIG_PATH: "configs/tasks/pointnav_mp3d.yaml"

TRAINER_NAME: "ddppo"
ENV_NAME: "NavRLEnv"
SIMULATOR_GPU_ID: 0
TORCH_GPU_ID: 0
VIDEO_OPTION: []
TENSORBOARD_DIR: "tb"
VIDEO_DIR: "video_dir"
TEST_EPISODE_COUNT: -1
EVAL_CKPT_PATH_DIR: "data/new_checkpoints"
#EVAL_CKPT_PATH_DIR: "data/pretrained_checkpoints"
NUM_PROCESSES: 4
# NUM_PROCESSES: 6
#SENSORS: ["DEPTH_SENSOR"]
SENSORS: ["RGB_SENSOR", "DEPTH_SENSOR"]
CHECKPOINT_FOLDER: "data/new_checkpoints"
NUM_UPDATES: 10000
LOG_INTERVAL: 10
CHECKPOINT_INTERVAL: 250

RL:
  SUCCESS_REWARD: 2.5
  PPO:
    # ppo params
    clip_param: 0.2
    ppo_epoch: 2
    num_mini_batch: 2
    value_loss_coef: 0.5
    entropy_coef: 0.01
    lr: 2.5e-4
    eps: 1e-5
    max_grad_norm: 0.2
    num_steps: 128
    use_gae: True
    gamma: 0.99
    tau: 0.95
    use_linear_clip_decay: False
    use_linear_lr_decay: False
    reward_window_size: 50

    use_normalized_advantage: False

    hidden_size: 512

  DDPPO:
    sync_frac: 0.6
[... 574 lines cut ...]
    trainer.eval()
  File "/home/acb11899xv/habitat-api/habitat_baselines/common/base_trainer.py", line 109, in eval
    checkpoint_index=prev_ckpt_ind,
  File "/home/acb11899xv/habitat-api/habitat_baselines/rl/ppo/ppo_trainer.py", line 474, in _eval_checkpoint
    self.agent.load_state_dict(ckpt_dict["state_dict"])
  File "/home/acb11899xv/miniconda3/envs/habitat/lib/python3.6/site-packages/torch/nn/modules/module.py", line 777, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for DDPPO:
        Missing key(s) in state_dict: "actor_critic.net.prev_action_embedding.weight", "actor_critic.net.tgt_embeding.weight", "actor_critic.net.tgt_embeding.bias", "actor_critic.net.visual_encoder.running_mean_and_var._mean", "actor_critic.net.visual_encoder.running_mean_and_var._var", "actor_critic.net.visual_encoder.running_mean_and_var._count", "actor_critic.net.visual_encoder.backbone.conv1.0.weight", "actor_critic.net.visual_encoder.backbone.conv1.1.weight", "actor_critic.net.visual_encoder.backbone.conv1.1.bias", "actor_critic.net.visual_encoder.backbone.layer1.0.convs.0.weight", "actor_critic.net.visual_encoder.backbone.layer1.0.convs.1.weight", "actor_critic.net.visual_encoder.backbone.layer1.0.convs.1.bias", "actor_critic.net.visual_encoder.backbone.layer1.0.convs.3.weight", "actor_critic.net.visual_encoder.backbone.layer1.0.convs.4.weight", "actor_critic.net.visual_encoder.backbone.layer1.0.convs.4.bias", "actor_critic.net.visual_encoder.backbone.layer1.0.convs.6.weight", "actor_critic.net.visual_encoder.backbone.layer1.0.convs.7.weight", "actor_critic.net.visual_encoder.backbone.layer1.0.convs.7.bias", "actor_critic.net.visual_encoder.backbone.layer1.0.downsample.0.weight", "actor_critic.net.visual_encoder.backbone.layer1.0.downsample.1.weight", "actor_critic.net.visual_encoder.backbone.layer1.0.downsample.1.bias", "actor_critic.net.visual_encoder.backbone.layer1.1.convs.0.weight", "actor_critic.net.visual_encoder.backbone.layer1.1.convs.1.weight", "actor_critic.net.visual_encoder.backbone.layer1.1.convs.1.bias", "actor_critic.net.visual_encoder.backbone.layer1.1.convs.3.weight", "actor_critic.net.visual_encoder.backbone.layer1.1.convs.4.weight", "actor_critic.net.visual_encoder.backbone.layer1.1.convs.4.bias", "actor_critic.net.visual_encoder.backbone.layer1.1.convs.6.weight", "actor_critic.net.visual_encoder.backbone.layer1.1.convs.7.weight", "actor_critic.net.visual_encoder.backbone.layer1.1.convs.7.bias", "actor_critic.net.visual_encoder.backbone.layer1.2.convs.0.weight", "actor_critic.net.visual_encoder.backbone.layer1.2.convs.1.weight", "actor_critic.net.visual_encoder.backbone.layer1.2.convs.1.bias", "actor_critic.net.visual_encoder.backbone.layer1.2.convs.3.weight", "actor_critic.net.visual_encoder.backbone.layer1.2.convs.4.weight", "actor_critic.net.visual_encoder.backbone.layer1.2.convs.4.bias", "actor_critic.net.visual_encoder.backbone.layer1.2.convs.6.weight", "actor_critic.net.visual_encoder.backbone.layer1.2.convs.7.weight", "actor_critic.net.visual_encoder.backbone.layer1.2.convs.7.bias", "actor_critic.net.visual_encoder.backbone.layer2.0.convs.0.weight", "actor_critic.net.visual_encoder.backbone.layer2.0.convs.1.weight", "actor_critic.net.visual_encoder.backbone.layer2.0.convs.1.bias", "actor_critic.net.visual_encoder.backbone.layer2.0.convs.3.weight", "actor_critic.net.visual_encoder.backbone.layer2.0.convs.4.weight", "actor_critic.net.visual_encoder.backbone.layer2.0.convs.4.bias", "actor_critic.net.visual_encoder.backbone.layer2.0.convs.6.weight", "actor_critic.net.visual_encoder.backbone.layer2.0.convs.7.weight", "actor_critic.net.visual_encoder.backbone.layer2.0.convs.7.bias", "actor_critic.net.visual_encoder.backbone.layer2.0.downsample.0.weight", "actor_critic.net.visual_encoder.backbone.layer2.0.downsample.1.weight", "actor_critic.net.visual_encoder.backbone.layer2.0.downsample.1.bias", "actor_critic.net.visual_encoder.backbone.layer2.1.convs.0.weight", "actor_critic.net.visual_encoder.backbone.layer2.1.convs.1.weight", "actor_critic.net.visual_encoder.backbone.layer2.1.convs.1.bias", "actor_critic.net.visual_encoder.backbone.layer2.1.convs.3.weight", "actor_critic.net.visual_encoder.backbone.layer2.1.convs.4.weight", "actor_critic.net.visual_encoder.backbone.layer2.1.convs.4.bias", "actor_critic.net.visual_encoder.backbone.layer2.1.convs.6.weight", "actor_critic.net.visual_encoder.backbone.layer2.1.convs.7.weight", "actor_critic.net.visual_encoder.backbone.layer2.1.convs.7.bias", "actor_critic.net.visual_encoder.backbone.layer2.2.convs.0.weight", "actor_critic.net.visual_encoder.backbone.layer2.2.convs.1.weight", "actor_critic.net.visual_encoder.backbone.layer2.2.convs.1.bias", "actor_critic.net.visual_encoder.backbone.layer2.2.convs.3.weight", "actor_critic.net.visual_encoder.backbone.layer2.2.convs.4.weight", "actor_critic.net.visual_encoder.backbone.layer2.2.convs.4.bias", "actor_critic.net.visual_encoder.backbone.layer2.2.convs.6.weight", "actor_critic.net.visual_encoder.backbone.layer2.2.convs.7.weight", "actor_critic.net.visual_encoder.backbone.layer2.2.convs.7.bias", "actor_critic.net.visual_encoder.backbone.layer2.3.convs.0.weight", "actor_critic.net.visual_encoder.backbone.layer2.3.convs.1.weight", "actor_critic.net.visual_encoder.backbone.layer2.3.convs.1.bias", "actor_critic.net.visual_encoder.backbone.layer2.3.convs.3.weight", "actor_critic.net.visual_encoder.backbone.layer2.3.convs.4.weight", "actor_critic.net.visual_encoder.backbone.layer2.3.convs.4.bias", "actor_critic.net.visual_encoder.backbone.layer2.3.convs.6.weight", "actor_critic.net.visual_encoder.backbone.layer2.3.convs.7.weight", "actor_critic.net.visual_encoder.backbone.layer2.3.convs.7.bias", "actor_critic.net.visual_encoder.backbone.layer3.0.convs.0.weight", "actor_critic.net.visual_encoder.backbone.layer3.0.convs.1.weight", "actor_critic.net.visual_encoder.backbone.layer3.0.convs.1.bias", "actor_critic.net.visual_encoder.backbone.layer3.0.convs.3.weight", "actor_critic.net.visual_encoder.backbone.layer3.0.convs.4.weight", "actor_critic.net.visual_encoder.backbone.layer3.0.convs.4.bias", "actor_critic.net.visual_encoder.backbone.layer3.0.convs.6.weight", "actor_critic.net.visual_encoder.backbone.layer3.0.convs.7.weight", "actor_critic.net.visual_encoder.backbone.layer3.0.convs.7.bias", "actor_critic.net.visual_encoder.backbone.layer3.0.downsample.0.weight", "actor_critic.net.visual_encoder.backbone.layer3.0.downsample.1.weight", "actor_critic.net.visual_encoder.backbone.layer3.0.downsample.1.bias", "actor_critic.net.visual_encoder.backbone.layer3.1.convs.0.weight", "actor_critic.net.visual_encoder.backbone.layer3.1.convs.1.weight", "actor_critic.net.visual_e [... cut ...]
Traceback (most recent call last):
  File "/home/acb11899xv/habitat-api/habitat/core/vector_env.py", line 518, in __del__
  File "/home/acb11899xv/habitat-api/habitat/core/vector_env.py", line 400, in close
  File "/home/acb11899xv/miniconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 206, in send
AttributeError: 'NoneType' object has no attribute 'dumps'
```

Does the `Missing key(s) in state_dict:` mean the model was not saved properly or am I missing some configuration entry?

Thanks!

erikwijmans, 2020-08-11:
> =======current_ckpt: data/new_checkpoints/ckpt.40.pth=======

Seems like it is picking up checkpoints in that folder from an older training run with a different model.

asolano, 2020-08-17:
Thanks Erik, that was it. Once I deleted the old checkpoints the job ran normally. 👍

closed 2020-08-17 by erikwijmans