facebookresearch/habitat-lab #473: Is it possible to customize the texture of objects, or add new custom objects?

FutureXiang, 2020-09-01:
## ❓ Questions
Hello,

I wonder if I can manipulate the textures in the scene?

For example, **changing the texture mapping image** of an object to add some perturbation on its texture.
Or **adding a new object with custom shape and texture** (e.g. an image patch on the wall) is useful as well.

Any help would be appreciated!

aclegg3, 2020-09-01:
You can absolutely add new custom objects to a scene. You can check out our recent ECCV tutorials (https://aihabitat.org/tutorial/2020/) for more information. Specifically, I would suggest **Habitat-Sim for Interaction**, **Habitat-Sim Advanced Topics**, and **Habitat-Lab for Interaction**.

Regarding texture mapping: we do not support editing textures inside Habitat currently. You'll need to customize your assets outside of Habitat (e.g. in Blender) and import the new customized asset.

FutureXiang, 2020-09-14:
> You can absolutely add new custom objects to a scene. You can check out our recent ECCV tutorials (https://aihabitat.org/tutorial/2020/) for more information. Specifically, I would suggest [...]

@aclegg3 
Thank you for your helpful information!

However, when I followed the _Habitat-Sim for Interaction_ tutorial, I found that some useful functions like `get_object_template_manager()` are NOT introduced in current STABLE version (?), but appear in current MASTER version.

If I want to add new custom objects to scenes now, should I clone the MASTER version instead of the STABLE one, or wait for the incoming update?
Or there is any other ways to insert _File-based object templates_ without the `get_object_template_manager()` function?

aclegg3, 2020-09-15:
We are in the process of releasing a new stable version now. I suggest you use the master branch which is being released and update to stable in the very near future.

rithik25, 2021-06-17:
@aclegg3  Is it possible to import some object/character with a custom motion/animation? For example, would I be able to import a human with a walking animation into my scene?

aclegg3, 2021-06-17:
@rithik25 This question should probably be its own issue/feature request.

Introduced with [this Habitat-sim PR](https://github.com/facebookresearch/habitat-sim/pull/1312), you can import a character which is described by a URDF file and animate its joints. However, mesh skinning is not currently supported, so your character would need to be composed of rigid links.

rpartsey, 2022-09-01:
All the questions were answered in the Issue comments.

closed 2022-09-01 by rpartsey