facebookresearch/habitat-lab #474: Question about Intrinsic Matrix and Camera Settings

Xiaoming-Zhao, 2020-09-05:
## ❓ Questions and Help

Hi, I have several questions about the intrinsic/camera matrix of Habitat. Feel free to let me know if this question fits Habitat-Sim more. Thanks

1. Can I know whether the intrinsic matrix computed in [demo](https://aihabitat.org/docs/habitat-lab/view-transform-warp.html) is **accurate** or just an approximation?
```python
K = np.array([
    [1 / np.tan(hfov / 2.), 0., 0., 0.],
    [0., 1 / np.tan(hfov / 2.), 0., 0.],
    [0., 0.,  1, 0],
    [0., 0., 0, 1]
])
```

2. If `K` is accurate, I deduce that Habitat uses a pinhole camera model with a screen of size `[2 x height / width, 2]`, whose distance to the optical center is focal length `f`. The first element of the size is for the vertical axis while the second element is for the horizontal axis. Is this correct? 

3. My hypothesis also comes from these two images taken from the same position and orientation. The first is taken with `WIDTH` and `HEIGHT` of 640 and 360 while the latter is taken with a square size of 256. I find that their horizontal view is "same" or really similar. The difference comes from the vertical content.
[image]
[image]

erikwijmans, 2020-09-05:
It is accurate for a 1:1 aspect ratio.  If the aspect ratio is different, then `1 / np.tan(hfov / 2.)` at (1, 1) should be `1 / np.tan(hfov * height/width / 2.)`.  Habitat-sim uses a standard pinhole camera model specified by height, width, and horizontal FOV (hfov).

Those world coordinates are an approximation because perfect accuracy requires either an infinite number of pixels or infinite sub-pixel resolution.

Xiaoming-Zhao, 2020-09-05:
Thanks a lot for your reply. I think the (1, 1) element of `K` should be `(1 / np.tan(hfov / 2.)) * (width / height)`, where the aspect ratio is multiplied outside `tan`, right?

erikwijmans, 2020-09-05:
Yep, put 1/aspect_ratio in the tan for some reason.  You are right that it should be outside.

> Otherwise, we need to know the the number of pixels per unit distance in x-axis and y-axis.

Kinda, you just need to know that ratio (the aspect ratio), not their exact values.

This is the exact function we call to get the perspective project matrix if that is helpful https://doc.magnum.graphics/magnum/classMagnum_1_1Math_1_1Matrix4.html#a6475b3ef155c9142b890c8133504ae9b

I have found this blog helpful in the past for understanding this: https://www.scratchapixel.com/lessons/3d-basic-rendering/perspective-and-orthographic-projection-matrix/opengl-perspective-projection-matrix

You don't see the near-far clipping above as that is already undone by using the depth values instead of normalized depth.

Xiaoming-Zhao, 2020-09-05:
Thanks a lot for your explanation. Once you mention the projection matrix, I understand the demo now. I think the confusion comes from the terminology of **intrinsic parameters** which directs me to think of the intrinsic/camera matrix. However, you are following the procedure of OpenGL.

Actually, I think there are some differences in how computer vision literature and OpenGL process the image generation. When we have points/objects in the camera/eye coordinate system, they process them in different ways:
- For most computer vision textbook, there is only one more camera/intrinsic/calibration matrix, which directly maps points in the camera coordinate system to the canvas/screen.
- For OpenGL, there are two more steps:
   - A projection matrix is utilized to map points to Normalized Coordinate System (NDC)
   - Then a viewport transformation is applied to get pixel coordinates of canvas/screen

I think it may be better to change the terminology a little bit to make it more clear. I am happy to push a PR for it if you think it is necessary.

erikwijmans, 2020-09-06:
In OpenGL there is still an intrinsic matrix, it is just part of the perspective projection matrix.  The intrinsic matrix `K` above is a valid intrinsic matrix, it just assumes the camera has a screen whose pixels are specified by continuous values in [-1, -1]^2.  This is common in graphics as you don't want to have to write things that are locked a specific screen size as that will be variable.  That is really the only difference between the two.

closed 2020-09-06 by Xiaoming-Zhao