facebookresearch/habitat-lab #483: scene lighting control though habitat-lab api  ?

youzhenxing, 2020-09-15:
## ❓ Questions and Help
Hi 
   is there any ways to control scene light in habitat lab, I know it can be done in habitat-sim([scene lighting](https://aihabitat.org/docs/habitat-sim/lighting-setups.html))

Skylion007, 2020-09-15:
You can write some CustomTaskActions that access the HabitatSim instance directly to do that.

mathfac, 2020-09-16:
Do you want just to setup it once per episode  or change and make it part
of action space?

youzhenxing, 2020-09-16:
> Do you want just to setup it once per episode or change and make it part of action space?
> […](#)

Generally speaking, my purpose is to get images from different scene brightness. It seems that  setting it once can meet my requirement, but it is better to set it dynamically, by that way, I can simulate  camera with dynamic exposure. So which is easy to implement?
thanks

Skylion007, 2020-09-17:
Set it as an action for the agent and then just call it, just like you would if you were to move the agent.

rpartsey, 2022-09-01:
Closing the issue because it has not had recent activity.

closed 2022-09-01 by rpartsey