facebookresearch/habitat-lab #489: Automated Task Dataset Download Feature

Skylion007, 2020-09-23:
## 🚀 Feature

It would be nice if there was an API to automatically download the task datasets instead of having to extract them ourselves. Should be pretty easy to write, just not implemented at the moment. I envision there being a dictionary of sorts that maps to URLs or a special path prefix that tells you to download a dataset if it doesn't exist. Then it just calls requests and such to unzip the dataset.

## Motivation
* Just makes Habitat more friendly to use. Low PRI but useful if someone wants to contribute.

## Pitch
Makes Habitat-Lam more welcoming to new comers.

## Alternatives

## Additional context

Skylion007, 2021-07-22:
Ping @aravwatwani if you are still interested in working on this.

Yash621, 2021-11-05:
@Skylion007 can I work on this ?

rpartsey, 2022-09-04:
@Skylion007 Looks like such API has been already implemented in [habitat-sim](https://github.com/facebookresearch/habitat-sim/blob/2662204daa6d6324ced92b7e9e0ae0365cef9e41/src_python/habitat_sim/utils/datasets_download.py#L573).

closed 2023-01-26 by rpartsey