facebookresearch/habitat-lab #491: Trained Agent Compatibility Issue?

ericchen321, 2020-10-02:
## ❓ Questions and Help

Hello, I was trying to evaluate the trained depth agent from [here](https://dl.fbaipublicfiles.com/habitat/data/baselines/v1/habitat_baselines_v1.zip) with the following command:

`python habitat_baselines/agents/ppo_agents.py --input-type depth --model-path data/checkpoints/depth.pth --task-config configs/tasks/pointnav.yaml `

I've also changed the sensor used in pointnav.yaml to the depth sensor. Then I got this error:

```Traceback (most recent call last): 

  File "habitat_baselines/agents/ppo_agents.py", line 170, in <module> 

    main() 

  File "habitat_baselines/agents/ppo_agents.py", line 157, in main 

    agent = PPOAgent(agent_config) 

  File "habitat_baselines/agents/ppo_agents.py", line 90, in __init__ 

    for k, v in ckpt["state_dict"].items() 

  File "/home/eric/anaconda3/envs/hab5/lib/python3.6/site-packages/torch-1.5.0-py3.6-linux-x86_64.egg/torch/nn/modules/module.py", line 847, in load_state_dict 

    self.__class__.__name__, "\n\t".join(error_msgs))) 

RuntimeError: Error(s) in loading state_dict for PointNavBaselinePolicy: 

    Missing key(s) in state_dict: "net.visual_encoder.cnn.0.weight", "net.visual_encoder.cnn.0.bias", "net.visual_encoder.cnn.2.weight", "net.visual_encoder.cnn.2.bias", "net.visual_encoder.cnn.4.weight", "net.visual_encoder.cnn.4.bias", "net.visual_encoder.cnn.6.weight", "net.visual_encoder.cnn.6.bias", "net.state_encoder.rnn.weight_ih_l0", "net.state_encoder.rnn.weight_hh_l0", "net.state_encoder.rnn.bias_ih_l0", "net.state_encoder.rnn.bias_hh_l0", "critic.fc.weight", "critic.fc.bias".  

    Unexpected key(s) in state_dict: "net.cnn.0.weight", "net.cnn.0.bias", "net.cnn.2.weight", "net.cnn.2.bias", "net.cnn.4.weight", "net.cnn.4.bias", "net.cnn.6.weight", "net.cnn.6.bias", "net.rnn.weight_ih_l0", "net.rnn.weight_hh_l0", "net.rnn.bias_ih_l0", "net.rnn.bias_hh_l0", "net.critic_linear.weight", "net.critic_linear.bias".
```

Seems to me this is due to the newer Habitat API uses DD-PPO to train agents, whereas the old agents did not employ this policy. So I'm wondering will new agents that are compatible with the latest Habitat API be released? If so is there a projected time for the release?

Or perhaps I've been using the script incorrectly?

erikwijmans, 2020-10-06:
Huh, I thought we updated those.  This script will remap the weights to the new action ordering and names: https://gist.github.com/erikwijmans/94f33461982ac9e3cc354fb26e4b80e1

ericchen321, 2020-10-07:
Thanks Erik, I'll check it out!

closed 2022-09-04 by rpartsey