facebookresearch/habitat-lab #495: Convert Config System to Hydra/OmegaConf

Skylion007, 2020-10-12:
## 🚀 Feature

## Motivation
* Opening an issue to keep track of the task. While the YACS config system has served us well, the overhead of maintaining our own registry / constructor system is not insignficant. We should consider moving to a system that makes linking our data classes / attrs with our config system a bit easier. The currently proposed system would be OmegaConf / Hydra.

We should investigate if there is an easy path to converting the repo to use OmegaConf and seeing if we can gradually then enable OmegaConfg features overtime to improve the user experience.

## Pitch
Doing so would allow automatic instationating of various data classes like noise_models and such. It may also help with ObsTransformers, Policies, and other config systems that are pushing the limits we currently have on YACS.

It may also enable us to better enable features like multi-sensor support without excessive dynamic config generation and merging.

Another issue that this should address is how features like OBS_TRANSFORMER can access important config values from other parts of the system (like SENSORS etc...). We should investigate if there is a good way to support this without assuming SINGLETON.

We also have some pretty bad antipatterns of passing the config object everywhere and modifying it where necessary. Hopefully OmegaConf can help us move away from doing that.

It may also letter our config system be more compositional.

Migrate away from YACS to a more mature / better supported features that uses modern Python features like typehints and dataclasses to help reduce the amount of overhead for hacking on habitat_lab and particularly habitat_baselines.

## Alternatives
ginconfig or better using advanced features of the YAML (like referencing nodes from other nodes and such). 
* Update YACS

## Additional context

erikwijmans, 2020-10-13:
No updates on that.  It didn't seem to get/hasn't gotten really any attention.  I can update it (assuming I can find the time), but it is a good day or two of quite tedious work so I won't do that unless that is something folks actually want to merge.

As far as hydra is concerned, I have been watching it and while a lot has changed, AFAIK, it is still in the same place as it was a year ago for us -- it isn't ready to be the config system for a library.

omry, 2020-10-14:
> As far as hydra is concerned, I have been watching it and while a lot has changed, AFAIK, it is still in the same place as it was a year ago for us -- it isn't ready to be the config system for a library.

I think it's more ready than ever. Hydra a year ago is a completely different project with far more limited capabilities.
Structured Configs is a very important feature for frameworks (new in 1.0).
You can see examples like fairseq and other complex frameworks that have made the switch.

Having said that, it's still under very active development and frameworks are getting specific attention in 1.1:
- recursive instantiation is in master
- recursive defaults implementation is in progress (this is essentially a near rewrite of the config composition logic).
- support for controlling composition order in defaults (`_self_`) is coming as well (actually already working in the branch of the recursive defaults).

I am happy to chat about where you think Hydra is falling short for Habitat (or other libraries).

erikwijmans, 2020-10-14:
Hi @omry this is awesome!  I had missed that there is progress on bullets 2 and 3 as they are really the blocker for us.  I will be following them closely :)

mathfac, 2021-01-14:
Hi @omry,
Wanted to check the progress with bullets 2 and 3. Thank you!

omry, 2021-01-14:
Both in Hydra master and are accessible through a dev release! (see the Hydra README for installation instructions).

You can learn about the new support here:
- [The Defaults List](https://hydra.cc/docs/next/advanced/defaults_list)
- [Packages](https://hydra.cc/docs/next/advanced/overriding_packages)
- [Common Patterns/Extending Configs](https://hydra.cc/docs/next/patterns/extending_configs/)
- [Common Patterns/Configuring Experiments](https://hydra.cc/docs/next/patterns/configuring_experiments/)

Please note that Hydra 1.1 is not officially released yet and APIs may still change.
However, I encourage you to experiment with what is there and provide feedback.