facebookresearch/habitat-lab #499: results of testing accuarcy of depth and pose in 3dof/6dof motion is not good

youzhenxing, 2020-10-15:
## ❓ Questions and Help
Hi, I test the accuarcy of depth and pose is not good enough. To sum up, when there is large 3dof rotation or not small  relative pose, errors is large.
 
Testing methods: 
     1. get two frames with overlapping ,named frame1,and frame2
     2.compute relative pose of the two frame
     3. project depth of frame1 with relative pose to frame2,I can get a project 3d point and its relative pixel coordinates, and then compute depth error of projected 3d point and depth from frame2 with same pixel coordinate.

I make several tests, but found there is large errors.
    1. 3dof motion(x,z,yaw)
    2.6dof motion(x,y,z,roll,pitch,yaw),By adding permutation on y,roll and pitch,and get ovservations by funcion ` observations = env.habitat_env.sim.get_observations_at(pose_p,pose_q,keep_agent_at_new_pose=True)`

Here is the results,for three images in the video, the first is the frame1 gray image,second is the frame2 gray images, third is projected depth image. In the third depth image, the dark part and curves is pixel without depth projection, the colors pixel is where depth erros > 5mm, and colored pixels is colord as BGR(255,0,np.abs(np.clip((np.abs(depth_error) * 100), 0, 255))),By this way green pixel is error nrear 5mm, while purple pixel is large errors near 2.5m.

[image]
realtive full test is recorded as a video at https://drive.google.com/file/d/174FBd37HnoiN-6Q7T6tTFgaBxtjQXEpI/view?usp=sharing

Relative code:
yaml config file
```
  STEREO_RGB_LEFT:
    HFOV: 110
    WIDTH: 640
    HEIGHT: 480
    POSITION: [0,1.5,0.]
  STEREO_RGB_RIGHT:
    HFOV: 110
    WIDTH: 640
    HEIGHT: 480
    POSITION: [0.1,1.5,0.]
  STEREO_DEPTH_LEFT:
    HFOV: 110
    WIDTH: 640
    HEIGHT: 480
    POSITION: [0,1.5,0.]
  STEREO_DEPTH_RIGHT:
    HFOV: 110
    WIDTH: 640
    HEIGHT: 480
    POSITION: [0.1,1.5,0.]
```
some usage funcion:compute camera intrinsics
```
CAM_HFOV = 110.0
def compute_camera_intrinsic():
         hfov= CAM_HFOV * (np.pi) / 180.0
         l = np.sqrt(640.0*640.0+480.0*480.0)
        fx = (1.0 / np.tan(hfov / 2.))*640.0/2.0
        fy = fx
       cx = 320.0
       cy = 240.0
          K = np.array([
        [fx, 0., cx],
        [0., fy, cy],
        [0., 0., 1.],
    ],dtype=np.float64)
     
```

usage funcions: convert replica coordinate to habitat camera coordinate, refer to 
```
def replica_camera_to_habitat_camera(point3d: np.array)->np.array:
    return np.array([point3d[0],-point3d[1],-point3d[2]])

def habitat_camera_to_replica_camera(point3d: np.array)->np.array:
    return np.array([point3d[0],-point3d[1],-point3d[2]])
```
code for computing relative pose of two frames
```
    c2_q_c1 = frame2["pose_q"].inverse() * frame1["pose_q"]
    c2_p_c1 =  quaternion_rotate_vector(frame2["pose_q"].inverse(),frame1["pose_p"]-frame2["pose_p"])
```

uage funciton conversation between 3d point and pixel coordinate:
````
def recover_pixel_to_3d(u, v, fx, fy, cx, cy, Z)->np.array:
    X = ((u-cx)/fx)*Z
    Y = ((v-cy)/fy)*Z
    return np.array([X,Y,Z])
def project_3d_to_pixel(point3d: np.array,fx,fy,cx,cy):
    X = float(point3d[0])
    Y = float(point3d[1])
    Z = float(point3d[2])
    u = fx*X/Z+cx
    v = fy*Y/Z+cy
    return [u,v]
````

depth projection code 
```
   for i in range(0,rows,1):
        for j in range(0,cols,1):
            # compute projection 3d
            replica_c1_left_p1_depth = depth_left1[i][j]
            replica_c1_left_p1_3d = recover_pixel_to_3d(j, i, fx, fy, cx, cy, replica_c1_left_p1_depth)
            habitat_c1_left_p1_3d = replica_camera_to_habitat_camera(replica_c1_left_p1_3d)
            habitat_c2_left_p1_3d_projected =  quaternion_rotate_vector(c2_q_c1,
                                                                                         habitat_c1_left_p1_3d) + c2_p_c1
            # print("habitat_c2_left_p1_3d_projected:", habitat_c2_left_p1_3d_projected)
            replica_c2_left_p1_3d_projected = habitat_camera_to_replica_camera(habitat_c2_left_p1_3d_projected)
            # print(replica_c1_left_p1_3d, "-->", replica_c2_left_p1_3d_projected,"diff:",replica_c2_left_p1_3d_projected - replica_c1_left_p1_3d)
            u,v = project_3d_to_pixel(replica_c2_left_p1_3d_projected, fx, fy, cx, cy)
            u = np.round(u).astype(np.int)
            v = np.round(v).astype(np.int)
            # print("u,v:",u,v)
            if u >=0 and u<cols and v >=0 and v<rows:
                depth_error = depth_left2[v][u]-replica_c2_left_p1_3d_projected[2]
                depth_diff_image[v][u] = depth_error
                depth_errors.append(depth_error)
                projected_depth[v][u] = replica_c2_left_p1_3d_projected[2]

                depth_error_uint8 = np.abs(np.clip((np.abs(depth_error) * 100), 0, 255))
                if depth_error_uint8 > 5:
                    # print(depth_diff_image_uint8[i][j])
                    projected_depth_show_uint8[v][u] =  (255, 0, depth_error_uint8) #BGR
                    image2_show[v][u] =  (255, 0, depth_error_uint8)
                else:
                    projected_depth_show_uint8[v][u] = (int(np.clip(replica_c2_left_p1_3d_projected[2] * 100, 0, 255)))
```

mathfac, 2020-10-15:
@erikwijmans, @aclegg3 maybe you can take a look.

youzhenxing, 2020-10-15:
I will upload more tests soon.

erikwijmans, 2020-10-15:
This may be helpful: https://aihabitat.org/docs/habitat-lab/view-transform-warp.html

> usage funcions: convert replica coordinate to habitat camera coordinate

This is almost certainly unnecessary and most likely the source of error.  There is a difference in coordinate frame between the Replica render and Habitat's Render, or in the issue you linked Blender's render and Habitat'.  However, as far as I can tell, everything is starting and ending in Habitat's coordinate frame, so there is no need to transform to a different one.

youzhenxing, 2020-10-15:
> This may be helpful: https://aihabitat.org/docs/habitat-lab/view-transform-warp.html
> 
> > usage funcions: convert replica coordinate to habitat camera coordinate
> 
> This is likely the source of error. There is a difference in coordinate frame between the Replica render and Habitat's render, or in the issue you linked, Blender's render and Habitat's. However, as far as I can tell, everything is starting and ending in Habitat's coordinate frame, so there is no need to transform to a different one.
> 
> EDIT: Maybe never-mind on that. I am still somewhat weary of it, but it seems like this is a difference in naming, that is part of converting from OpenGL screen space back to clip space.
> 
> Hopefully the demo helps. If it doesn't, I can look into it.

Thanks, seems I misunderstood the coordinate transformation, I am confused the way of computing instrinsic matrix in the demo,and I found discussion in https://github.com/facebookresearch/habitat-lab/issues/474#, but still can't get a clear understanding. what I concerned is:
1. I use image with resolution(W=640,H=480),How can I change the instrinsic matrix?
2. If I use intrinsic by the following way, is there any additional conversation?
```
CAM_HFOV = 110.0
def compute_camera_intrinsic():
         hfov= CAM_HFOV * (np.pi) / 180.0
         l = np.sqrt(640.0*640.0+480.0*480.0)
        fx = (1.0 / np.tan(hfov / 2.))*640.0/2.0
        fy = fx
       cx = 320.0
       cy = 240.0
          K = np.array([
        [fx, 0., cx],
        [0., fy, cy],
        [0., 0., 1.],
    ],dtype=np.float64)
   return K,fx,fy,cx,cy

def recover_pixel_to_3d(u, v, fx, fy, cx, cy, Z)->np.array:
    X = ((u-cx)/fx)*Z
    Y = ((v-cy)/fy)*Z
    return np.array[X,Y,Z]
```

youzhenxing, 2020-10-15:
> usage funcions: convert replica coordinate to habitat camera coordinate

After reading transform code in the demo,I found it seems that I got the right coordinate transformation by accident.  In the usage funcion, I transfrom 3d point expressed in camera coordinate axis(C) to word coordinate axis(W) using function `replica_camera_to_habitat_camera` , and before computing pixel coordinates, the projected 3d points expressed in world coordinate axis was converted to  coordinates in camera coordiante axis using functions `habitat_camera_to_replica_camera`.

[image]

erikwijmans, 2020-10-15:
For the demo, the intrinsic matrix would be
```python
vfov = 2 * np.atan(np.tan(hfov/2)*height/width)
fx = 1.0 / np.tan(hfov / 2.)
fy = 1.0 / np.tan(vfov / 2.)
K = np.array([
    [fx, 0., 0., 0.],
    [0., fy, 0., 0.],
    [0., 0.,  1, 0],
    [0., 0., 0, 1]])
```

erikwijmans, 2020-10-15:
Based on this
```python
fx = (1.0 / np.tan(hfov / 2.))*640.0/2.0
fy = fx
```
that is likely the source of the error.  Since you have a non square aspect ratio, fy won't equal fx.

Skylion007, 2020-10-15:
BTW, @youzhenxing if you want to write an stereo_example.py/collab for Habitat-Lab, we could potentially merge it in as a PR. :)

erikwijmans, 2020-10-15:
Where are `frame1["pose_q"]` and `frame1["pose_p"]` coming from?  Based on your post, it seems like those are the rotation and position of the agent body, not the camera.  If so, that would explain this.

youzhenxing, 2020-10-15:
> Where are `frame1["pose_q"]` and `frame1["pose_p"]` coming from? Based on your post, it seems like those are the rotation and position of the agent body (i.e. some point on the floor), not the camera. If so, that would explain this.

yes, `frame1["pose_q"]` and `frame1["pose_p"]` is pose of agent body, and I found this was what causes unexplainable errors, I had neglected pose between agent pose and camera. After fixing this bug when computing relative pose,I got reasonable results. 

new code to compute relative pose
```
def toSE3(pose_p,pose_q):
    rotation = quaternion.as_rotation_matrix(pose_q)
    T = np.eye(4)
    T[0:3, 0:3] = rotation
    T[0:3, 3] = pose_p
    return T
....

    w_T_r1 = toSE3(frame1["pose_p"],frame1["pose_q"])
    w_T_r2 = toSE3(frame2["pose_p"],frame2["pose_q"])
    r_T_c = toSE3(np.array([0.,1.5,0.]),np.quaternion(1.,0.,0.,0.))

    c2_T_c1 = np.matmul(np.matmul(np.linalg.inv(r_T_c),np.linalg.inv(w_T_r2)),np.matmul(w_T_r1,r_T_c))
    c2_p_c1 = c2_T_c1[0:3,3]
    c2_q_c1 = quaternion.from_rotation_matrix(c2_T_c1[0:3,0:3])
```
- errors which are larger than 1cm only occured where there are index roundings or shieldings, which are reasonable.
[image]

Thanks.

youzhenxing, 2020-10-15:
This is what causes unexplainable errors

> BTW, @youzhenxing if you want to write an stereo_example.py/collab for Habitat-Lab, we could potentially merge it in as a PR. :)

OK,that's great

erikwijmans, 2020-10-15:
You can get the sensor states as
```python
observations = env.habitat_env.sim.get_observations_at(pose_p,pose_q,keep_agent_at_new_pose=True)
sensor_states = env.habitat_env.sim.get_agent_state().sensor_states

....

pose_q = sensor_states["depth_left"].rotation
pose_p = sensor_states["depth_left"].position
```

youzhenxing, 2020-10-16:
> You can get the sensor states as
>
> ```python
> observations = env.habitat_env.sim.get_observations_at(pose_p,pose_q,keep_agent_at_new_pose=True)
> sensor_states = [...]

that will be helpful.
Thanks for figuring out these problems.

closed 2020-10-16 by youzhenxing