facebookresearch/habitat-lab #500: Add Observation Transformers for Equirec2Perspective

Skylion007, 2020-10-16:
## 🚀 Feature

 360 sensor -> perspective view observation transformer

I would propose two observation transformers for this:
* Equirect -> perspective imgs (an observation transfomer that can extract multiple perspective imgs from an equirect input)
* A special case of this: Equirect -> Cubemap (allows policies trained on cubemaps to be evaluate on real robots with equirect sensors).

## Motivation
* So one useful feature that would make sim2real comparisons is easier if sampling a perspective image out of a 360 camera sensor (assuming equirect inputs). This would allow easier transfer learning / comparisons from real robots that only come with 360 sensors.

Implementations to base this off of:
* https://github.com/haruishi43/equilib/tree/master/equilib/equi2cube/equi2cube_torch
* https://github.com/haruishi43/equilib/tree/master/equilib/equi2pers/equi2pers_torch

Bonus:
* https://github.com/haruishi43/equilib/tree/master/equilib/equi2equi/equi2equi_torch

Please look at the recent PRs #485 and #486 for examples about how to add these observation transformers.

## Additional context
@matsuren Does this task interest you at all? 

The improvements /additions you made with cubemap2equirect and cubemap2fisheye were greatly appreciated.

matsuren, 2020-10-19:
Hi @Skylion007, 
It sounds good! I'm a little bit concerned about the spatial resolution when you convert an equirectangular image into a perspective image (You might not be able to see the detailed structures in the perspective image). But if you use high-resolution equirectangular images, it should be okay?

I can try to implement this feature to see how it looks when I have time (I'm busy this week, so hopefully in two or three weeks). Also, I'm interested in the integration of cubemap2equirect, cubemap2fisheye, and equirect2perspective, so I can try this as well if it's okay for you.

Thanks,

Skylion007, 2020-10-19:
@matsuren Yeah, for reference I imagine this being used on output from commodity 360 sensors like a Ricoh Theta (which easily supports 4k) so that should be more than enough for low resolution perspective images. 👍

matsuren, 2020-10-20:
Hi @Skylion007, I got it! I will ask you if I have any problems regarding the implementation.
Thanks,

matsuren, 2020-10-21:
Todo list:
- [ ]  Implementation of equirect2perspective
- [ ]  Bug fix for depthmap generation in cubemap2equirect and cubemap2fisheye (Currently, six depthmaps in cubemap are the distance from each image plane, but it should be the distance from the optical center to create depthmap for equirect and fisheye.)
- [ ]  Integration of cubemap2equirect, cubemap2fisheye, and equirect2perspective

Skylion007, 2020-10-26:
Great point of the depth map generation bug! I hadn't considered that. 

* We will need some special handling for Sensor Observation Types for if its a depth sensor. We can have the user specify it in the mapping (with a boolean) or have some flag or even hardcode it if it's a 1 channel float32 sensor.

Skylion007, 2020-11-17:
Fixed by #517

closed 2020-11-17 by Skylion007