facebookresearch/habitat-lab #501: Question about "get_observations_at()"

jzhzhang, 2020-10-18:
## ❓ Questions and Help
Thanks for your great project. I use the **get_observations_at** to get the specific view.
But I find the new agent state is different from the given rotation.
The code is really samply as below:
```
    print("translation",translation)
    print("oritention,",orientation)
    obs = env._sim.get_observations_at(position=translation, rotation=orientation, keep_agent_at_new_pose=True)
    quaternion_0 =env._sim.get_agent_state().sensor_states['depth'].rotation
    translation_0 = env._sim.get_agent_state().sensor_states['depth'].position
    print("translation state",translation_0)
    print("oritention state,",quaternion_0)  
```
And the output:
```
**translation [ 0.91128135 -1.60025     4.0374794 ]**
oritention, quaternion(0.231165246789361, 0.523560462941795, 0.150128546914304, 0.806169020567267)
**translation state [ 0.6418891 -2.6603103  4.6426244]**
oritention state, quaternion(0.231165245175362, 0.523560464382172, 0.150128543376923, 0.806169033050537)
```

I can't figure out why there is a difference between the translations. Is this a BUG or what?

mathfac, 2020-10-18:
Hi @

Thank you for the question:
Check the docs, the passed orientation and position are agent’s orientation
and position:
https://aihabitat.org/docs/habitat-lab/habitat.core.simulator.Simulator.html#get_observations_at

Sensors positions are set in the config relative to Agent’s bottom’s center.

mathfac, 2020-10-18:
Hi @jzhzhang 

Thank you for the question.
Based on [the docs](https://aihabitat.org/docs/habitat-lab/habitat.core.simulator.Simulator.html#get_observations_at), the passed orientation and position are **agent’s orientation
and position**:
https://github.com/facebookresearch/habitat-lab/blob/7c4286653211bbfaca59d0807c28bfb3a6b962bf/habitat/core/simulator.py#L307-L325

You’re checking sensors positions, not agent’s. Sensors positions are set in the config relative to Agent’s bottom’s center.

https://github.com/facebookresearch/habitat-lab/blob/7c4286653211bbfaca59d0807c28bfb3a6b962bf/habitat/config/default.py#L227-L228

jzhzhang, 2020-10-19:
@mathfac Thanks for your quick response. It solves my problem perfectly!

closed 2020-10-19 by jzhzhang