facebookresearch/habitat-lab #513: How to use pre-trained pytorch models?

Sunnyzhr, 2020-10-26:
## ❓ Questions and Help
According to the tutorials, I download [pre-trained pytorch models](https://dl.fbaipublicfiles.com/habitat/data/baselines/v1/habitat_baselines_v1.zip). But where should I unzip and how to specify it by codes?

First, I unzip it to `/home/u/Desktop/habitat-lab/habitat_baselines/habitat_baselines_v1(1)`.
Second, I change codes `deterministic=False,` into `deterministic="/home/u/Desktop/habitat-lab/habitat_baselines/habitat_baselines_v1(1)/rgbd.pth",` in `/home/u/Desktop/habitat-lab/habitat_baselines/agents/ppo_agents.py`

Then I run command`(habitat) root@c:~/Desktop/habitat-lab# python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type eval
`
I get the error:

> WARNING:root:This caffe2 python run does not have GPU support. Will run in CPU only mode.
2020-10-26 15:53:53,103 config: BASE_TASK_CONFIG_PATH: configs/tasks/pointnav.yaml
CHECKPOINT_FOLDER: data/new_checkpoints
CHECKPOINT_INTERVAL: 50
CMD_TRAILING_OPTS: []
ENV_NAME: NavRLEnv
EVAL:
  SPLIT: val
  USE_CKPT_CONFIG: True
EVAL_CKPT_PATH_DIR: data/new_checkpoints
FORCE_BLIND_POLICY: False
LOG_FILE: train.log
LOG_INTERVAL: 10
NUM_PROCESSES: 1
NUM_UPDATES: 10000
ORBSLAM2:
  ANGLE_TH: 0.2617993877991494
  BETA: 100
  CAMERA_HEIGHT: 1.25
  DEPTH_DENORM: 10.0
  DIST_REACHED_TH: 0.15
  DIST_TO_STOP: 0.05
  D_OBSTACLE_MAX: 4.0
  D_OBSTACLE_MIN: 0.1
  H_OBSTACLE_MAX: 1.25
  H_OBSTACLE_MIN: 0.375
  MAP_CELL_SIZE: 0.1
  MAP_SIZE: 40
  MIN_PTS_IN_OBSTACLE: 320.0
  NEXT_WAYPOINT_TH: 0.5
  NUM_ACTIONS: 3
  PLANNER_MAX_STEPS: 500
  PREPROCESS_MAP: True
  SLAM_SETTINGS_PATH: habitat_baselines/slambased/data/mp3d3_small1k.yaml
  SLAM_VOCAB_PATH: habitat_baselines/slambased/data/ORBvoc.txt
RL:
  DDPPO:
    backbone: resnet50
    distrib_backend: GLOO
    num_recurrent_layers: 2
    pretrained: False
    pretrained_encoder: False
    pretrained_weights: data/ddppo-models/gibson-2plus-resnet50.pth
    reset_critic: True
    rnn_type: LSTM
    sync_frac: 0.6
    train_encoder: True
  POLICY:
    OBS_TRANSFORMS:
      CENTER_CROPPER:
        HEIGHT: 256
        WIDTH: 256
      CUBE2EQ:
        CUBE_LENGTH: 256
        HEIGHT: 256
        SENSOR_UUIDS: []
        WIDTH: 512
      ENABLED_TRANSFORMS: ()
      RESIZE_SHORTEST_EDGE:
        SIZE: 256
[... 215 lines cut ...]
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/tensorflow-1.13.1-py3.6-linux-x86_64.egg/tensorflow/python/framework/dtypes.py:527: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/tensorflow-1.13.1-py3.6-linux-x86_64.egg/tensorflow/python/framework/dtypes.py:528: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/tensorflow-1.13.1-py3.6-linux-x86_64.egg/tensorflow/python/framework/dtypes.py:529: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/tensorflow-1.13.1-py3.6-linux-x86_64.egg/tensorflow/python/framework/dtypes.py:530: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/tensorflow-1.13.1-py3.6-linux-x86_64.egg/tensorflow/python/framework/dtypes.py:535: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
Traceback (most recent call last):
  File "habitat_baselines/run.py", line 79, in <module>
    main()
  File "habitat_baselines/run.py", line 40, in main
    run_exp(**vars(args))
  File "habitat_baselines/run.py", line 75, in run_exp
    execute_exp(config, run_type)
  File "habitat_baselines/run.py", line 60, in execute_exp
    trainer.eval()
  File "/home/u/Desktop/habitat-lab/habitat_baselines/common/base_trainer.py", line 103, in eval
    self.config.EVAL_CKPT_PATH_DIR, prev_ckpt_ind
  File "/home/u/Desktop/habitat-lab/habitat_baselines/utils/common.py", line 123, in poll_checkpoint_folder
    f"invalid checkpoint folder " f"path {checkpoint_folder}"
AssertionError: invalid checkpoint folder path data/new_checkpoints

Skylion007, 2020-11-06:
You still have to change ```EVAL_CKPT_PATH_DIR``` to point to the pretrained checkpoint you want to load.

ronjamoller, 2020-11-16:
HI, i am quite late to this issue but i didnt want to open a new one for a similar problem: I am trying to use the pretrained PPO and DDPPO models and im running into some issues (to make sure that its not a path issue - the checkpoint is loaded : `  2020-11-16 20:09:19,161 =======current_ckpt: data/new_checkpoints/depth.pth=======`) . 
1. the checkpoints dont seems to contain configs - `config = self._setup_eval_config(ckpt_dict["config"])` fails with both the PPO models and the DDPPO models so i have to fall back onto the default config ` config = self.config.clone()` in PPOTrainer for example. I dont have this issues with checkpoints i have generated myself. Is this a new change (i redownloaded and installed habitat to check so my code is current) ? If none of the checkpoint files contain configs, i guess for the PPO and DDPPO models the correct configs would be in the default.yaml files, is this correct? This brings up the problem of the varying DDPPO-models because as i understand it the best-performing depth model uses a 1024 state encoder, unlike the default ddppo yaml - is there a place where the different config files for the different dd-ppo checkpoints are stored ?
This is the error:
`config = self._setup_eval_config(ckpt_dict["config"])
KeyError: 'config'`
2. I consistently run into naming issues with the checkpoint dicts, im assuming this is because of newer more informative names, the error is this:
`RuntimeError: Error(s) in loading state_dict for PPO:
        Missing key(s) in state_dict: "actor_critic.net.visual_encoder.cnn.0.weight", "actor_critic.net.visual_encoder.cnn.0.bias", "actor_critic.net.visual_encoder.cnn.2.weight", "actor_critic.net.visual_encoder.cnn.2.bias", "actor_critic.net.visual_encoder.cnn.4.weight", "actor_critic.net.visual_encoder.cnn.4.bias", "actor_critic.net.visual_encoder.cnn.6.weight", "actor_critic.net.visual_encoder.cnn.6.bias", "actor_critic.net.state_encoder.rnn.weight_ih_l0", "actor_critic.net.state_encoder.rnn.weight_hh_l0", "actor_critic.net.state_encoder.rnn.bias_ih_l0", "actor_critic.net.state_encoder.rnn.bias_hh_l0", "actor_critic.critic.fc.weight", "actor_critic.critic.fc.bias". 
        Unexpected key(s) in state_dict: "actor_critic.net.cnn.0.weight", "actor_critic.net.cnn.0.bias", "actor_critic.net.cnn.2.weight", "actor_critic.net.cnn.2.bias", "actor_critic.net.cnn.4.weight", "actor_critic.net.cnn.4.bias", "actor_critic.net.cnn.6.weight", "actor_critic.net.cnn.6.bias", "actor_critic.net.rnn.weight_ih_l0", "actor_critic.net.rnn.weight_hh_l0", "actor_critic.net.rnn.bias_ih_l0", "actor_critic.net.rnn.bias_hh_l0", "actor_critic.net.critic_linear.weight", "actor_critic.net.critic_linear.bias".`
in short: I cant use the configs from the checkpoints (if they contain configs) and i cant load pretrained weights into my current models because of name mismatches.
thanks !

Sunnyzhr, 2020-12-06:
> You still have to change `EVAL_CKPT_PATH_DIR` to point to the pretrained checkpoint you want to load.

Thanks for your reply! @Skylion007 Sorry to bother you again, but I still don't know how to change the check point.

I don't find .pth documents in [pre-trained pytorch models](https://dl.fbaipublicfiles.com/habitat/data/baselines/v1/habitat_baselines_v1.zip) and find checkpoint related files in original `/habitat-lab/data/test_checkpoints` folder, so I define the checkpoints as below:

> def get_default_config():
    c = Config()
    c.INPUT_TYPE = "blind"
    # c.MODEL_PATH = "data/checkpoints/blind.pth"
    **c.MODEL_PATH = r"/home/u/Desktop/habitat-lab/data/test_checkpoints/ppo/pointnav/ckpt.0.pth"**
    c.RESOLUTION = 256
    c.HIDDEN_SIZE = 512
    c.RANDOM_SEED = 7
    c.PTH_GPU_ID = 0
    c.GOAL_SENSOR_UUID = "pointgoal_with_gps_compass"
    return c

Do I correctly define the checkpoints? Since there are only 4 files `blind.pth`,`rgb.pth`,`depth.pth`,`rgbd.pth` in [pre-trained pytorch models](https://dl.fbaipublicfiles.com/habitat/data/baselines/v1/habitat_baselines_v1.zip)

I really appreciate your help!

Sunnyzhr, 2020-12-06:
Moreover, after deleting all and retrying the tutorials, I do not encounter the checkpoint any more. Instead, I get the error below:

> .......
TENSORBOARD_DIR: tb
TEST_EPISODE_COUNT: 2
TORCH_GPU_ID: 0
TRAINER_NAME: ppo
VIDEO_DIR: video_dir
VIDEO_OPTION: ['disk', 'tensorboard']
2020-12-06 23:20:59,906 Initializing dataset PointNav-v1
2020-12-06 23:21:00,907 Initializing dataset PointNav-v1
2020-12-06 23:21:00,909 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1206 23:21:00.915284  6450 AssetAttributesManager.cpp:122] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I1206 23:21:00.915319  6450 AssetAttributesManager.cpp:122] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I1206 23:21:00.915344  6450 AssetAttributesManager.cpp:122] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I1206 23:21:00.915381  6450 AssetAttributesManager.cpp:122] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I1206 23:21:00.915391  6450 AssetAttributesManager.cpp:122] Asset attributes (cubeSolid : cubeSolid) created and registered.
I1206 23:21:00.915398  6450 AssetAttributesManager.cpp:122] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I1206 23:21:00.915419  6450 AssetAttributesManager.cpp:122] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I1206 23:21:00.915438  6450 AssetAttributesManager.cpp:122] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I1206 23:21:00.915462  6450 AssetAttributesManager.cpp:122] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I1206 23:21:00.915489  6450 AssetAttributesManager.cpp:122] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I1206 23:21:00.915503  6450 AssetAttributesManager.cpp:122] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I1206 23:21:00.915516  6450 AssetAttributesManager.cpp:122] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I1206 23:21:00.915522  6450 AssetAttributesManager.cpp:108] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I1206 23:21:00.915926  6450 ObjectAttributesManager.cpp:336] Parsing object library directory: ./data/objects
I1206 23:21:00.915954  6450 ObjectAttributesManager.cpp:294] Loading file-based object template: ./data/objects/banana.phys_properties.json
I1206 23:21:00.916010  6450 ObjectAttributesManager.cpp:294] Loading file-based object template: ./data/objects/cheezit.phys_properties.json
I1206 23:21:00.916061  6450 ObjectAttributesManager.cpp:294] Loading file-based object template: ./data/objects/chefcan.phys_properties.json
I1206 23:21:00.916110  6450 ObjectAttributesManager.cpp:294] Loading file-based object template: ./data/objects/largeclamp.phys_properties.json
I1206 23:21:00.916186  6450 ObjectAttributesManager.cpp:294] Loading file-based object template: ./data/objects/skillet.phys_properties.json
I1206 23:21:00.916262  6450 ObjectAttributesManager.cpp:306] Loaded file-based object templates: 5
I1206 23:21:00.916271  6450 PhysicsAttributesManager.cpp:39] File (./data/default.phys_scene_config.json) Based physics manager attributes created and registered.
I1206 23:21:00.916357  6450 StageAttributesManager.cpp:79] File (data/scene_datasets/habitat-test-scenes/van-gogh-room.glb) Based stage attributes created and registered.
I1206 23:21:00.916385  6450 Simulator.cpp:145] Loading navmesh from data/scene_datasets/habitat-test-scenes/van-gogh-room.navmesh
I1206 23:21:00.916404  6450 Simulator.cpp:147] Loaded.
I1206 23:21:00.916414  6450 SceneGraph.h:93] Created DrawableGroup: 
Renderer: GeForce GTX 1650/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 450.80.02
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-forward-compatible-core-context
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
I1206 23:21:00.984727  6450 ResourceManager.cpp:934] Importing Basis files as BC7
I1206 23:21:01.581944  6450 ResourceManager.cpp:302] ResourceManager::loadStage : Not loading semantic mesh
W1206 23:21:01.581966  6450 Simulator.cpp:235] :
 The active scene does not contain semantic annotations. 
I1206 23:21:01.582213 6450 simulator.py:181] Loaded navmesh data/scene_datasets/habitat-test-scenes/van-gogh-room.navmesh
2020-12-06 23:21:01,584 Initializing task Nav-v0
[... cut ...]
> Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7f3e2ac62898>>
> Traceback (most recent call last):
>   File "/home/u/Desktop/habitat-lab/habitat/core/vector_env.py", line 529, in __del__
>   File "/home/u/Desktop/habitat-lab/habitat/core/vector_env.py", line 413, in close
>   File "/home/u/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 206, in send
> AttributeError: 'NoneType' object has no attribute 'dumps'
> (habitat) root@c:~/Desktop/habitat-lab# 
>

ronjamoller, 2020-12-06:
Hi @Sunnyzhr  I am not a developer but I have some experience with habitat - I am not sure if you have already solved your issues but i will try to go through them:
1.  You mentioned that you set a deterministic parameter to a path, i think this parameter should only be true or false, no checkpoint. Deterministic in this case refers to the type of policy i think, as in deterministic or stochastic, I would leave it at False. 
2. re: checkpoint files, if your goal is to simply evaluate a pretrained model, you can just put it into  data/new_checkpoints/ and the eval script will automatically evaluate whats in thew new_checkpoints folder - depending on which model you want, just put for example the depth.pth file you downloaded into new_checkpoints. This is something to keep in mind for later - if there are several checkpoints in the folder the script will try to evaluate all of them, even if they dont actually fit your architecture so this folder should contain only the files you actually want to evaluate
3. with the write_gif issue: this seems to be an issue in the tensorboard code - I have never had this issue but i suspect it might be a mismatch in your anaconda env. - if you have installed your anaconda env according to the instructions and only pulled stable releases this might be a bug. As a workaround I would try to modify the ppo_pointnav_example.yaml to remove any option that generated video for tensorboard: change VIDEO_OPTION: ["disk", "tensorboard"] to VIDEO_OPTION: ["disk"]  or even VIDEO_OPTION: [] . This way the evaluation still runs through but no videos for manual inspection are generated.
Hope this helps and sorry if i explained things you already fixed.
best of luck

KenaHemnani, 2020-12-30:
I am also encountering similar problem, when trying to reproduce the results of pretrained models. 
`!python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav.yaml --run-type eval`

I am getting following error:
/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:526: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:527: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:528: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:529: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:530: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:535: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
2020-12-30 15:09:23,572 =======current_ckpt: data/new_checkpoints/depth.pth=======
Traceback (most recent call last):
  File "habitat_baselines/run.py", line 79, in <module>
    main()
  File "habitat_baselines/run.py", line 40, in main
    run_exp(**vars(args))
  File "habitat_baselines/run.py", line 75, in run_exp
    execute_exp(config, run_type)
  File "habitat_baselines/run.py", line 60, in execute_exp
    trainer.eval()
  File "/usr/local/lib/python3.6/site-packages/habitat_baselines/common/base_trainer.py", line 128, in eval
    checkpoint_index=prev_ckpt_ind,
  File "/usr/local/lib/python3.6/site-packages/habitat_baselines/rl/ppo/ppo_trainer.py", line 495, in _eval_checkpoint
    config = self._setup_eval_config(ckpt_dict["config"])
KeyError: 'config'

It will be very helpful if someone can suggest some solution.

KenaHemnani, 2020-12-30:
I have made a folder of new_checkpoints inside /content/habitat-lab/data/new_checkpoints/depth.pth and trying to run above command.

ronjamoller, 2020-12-30:
Hi @KenaHemnani i am not a developer but I have some experience with habitat. I encountered your error and the issue is that newer checkpoint files also contain the config file they were trained with while the older ones do not. The PPO checkpoints are older. As a result you can use a workaround and just take the config specified by your yaml: 
The issue is the code in the ppo trainer that throws your error:
`if self.config.EVAL.USE_CKPT_CONFIG:
config = self._setup_eval_config(ckpt_dict["config"])
else:
config = self.config.clone()`
so it tries to get the dict but doesnt find it - instead you can do this:
`        try:
            config = trainer._setup_eval_config(ckpt_dict["config"])
        except:
            config = trainer.config.clone() `
or just clone the config by default.

KenaHemnani, 2020-12-30:
@ronjamoller Thank you very much for the suggestion. I have tried using both ways as you mentioned above. But, both way getting same error as previous ones. Is there can be any other way?

Thank you for these suggestion. It will really helpful.

KenaHemnani, 2020-12-30:
When I try running:
`!python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type train`
`!python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type eval`

Training ppo_point_nav_example from scratch and then on eval mode. It is working fine. But, for checkpoints of pretrained models it is showing above errors.

ronjamoller, 2020-12-30:
Have you tried replacing `config = self._setup_eval_config(ckpt_dict["config"])` with `config = trainer.config.clone()` in the places where the program gives the error ? That is how I fixed that issue - as for your experiments, that is exactly the problem - newer checkpoint files contain a config entry, older checkpoint files do not. While it might be feasible to retrain the simple PPO model I think it is worth to try fixing this issue to get a good pretrained model to run later (the DDPPO models are way too big to train from scratch).
tldr: if the error is the same, then there is still a part of the code that tries to access `ckpt_dict["config"]`, which does not exist - you have the change them like above - it is uncomfortable to change the code in `habitat_baselines/rl/ppo/ppo_trainer.py` but i dont see another way.
Good luck, dont be deterred !

KenaHemnani, 2020-12-30:
@ronjamoller Yes, I tried that way , even tried keeping only `config = trainer.config.clone()` .But it gives same error.  Then from where self.config.EVAL.USE_CKPT_CONFIG this can be made false? I think it is still True as whatever information it prints before error.

ronjamoller, 2020-12-30:
you will need a config later on - so you can set the parameter to false but you will still need `config = trainer.config.clone()` what is the error if you only have `config = trainer.config.clone()` ?

KenaHemnani, 2020-12-30:
I am getting following error:

`Neither `ifconfig` (`ifconfig -a`) nor `ip` (`ip address show`) commands are available, listing network interfaces is likely to fail
2020-12-30 19:27:52,521 config: BASE_TASK_CONFIG_PATH: configs/tasks/pointnav_gibson.yaml
CHECKPOINT_FOLDER: data/new_checkpoints
CHECKPOINT_INTERVAL: 2000
CMD_TRAILING_OPTS: []
ENV_NAME: NavRLEnv
EVAL:
  SPLIT: val
  USE_CKPT_CONFIG: True
EVAL_CKPT_PATH_DIR: data/new_checkpoints
FORCE_BLIND_POLICY: False
LOG_FILE: train.log
LOG_INTERVAL: 25
NUM_PROCESSES: 4
NUM_UPDATES: 270000
ORBSLAM2:

.
.
.
.
.
.
.
.
.

TRAINER_NAME: ppo
VIDEO_DIR: video_dir
VIDEO_OPTION: []
/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:526: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:527: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:528: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:529: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:530: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:535: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
2020-12-30 19:27:55,489 =======current_ckpt: data/new_checkpoints/ckpt.0.pth=======
Traceback (most recent call last):
  File "habitat_baselines/run.py", line 79, in <module>
    main()
  File "habitat_baselines/run.py", line 40, in main
    run_exp(**vars(args))
  File "habitat_baselines/run.py", line 75, in run_exp
    execute_exp(config, run_type)
  File "habitat_baselines/run.py", line 60, in execute_exp
    trainer.eval()
  File "/usr/local/lib/python3.6/site-packages/habitat_baselines/common/base_trainer.py", line 128, in eval
    checkpoint_index=prev_ckpt_ind,
  File "/usr/local/lib/python3.6/site-packages/habitat_baselines/rl/ppo/ppo_trainer.py", line 495, in _eval_checkpoint
    config = self._setup_eval_config(ckpt_dict["config"])
KeyError: 'config'`

ronjamoller, 2020-12-30:
Thank you, as you can see the code still tries to access ckpt_dict["config"] - you have to change that section of the code in `/usr/local/lib/python3.6/site-packages/habitat_baselines/rl/ppo/ppo_trainer.py`

KenaHemnani, 2020-12-30:
Hii, @ronjamoller  I tried what said. Thank you very much for help!!  KeyError seems did not appear. But I am again getting some other error:

Traceback (most recent call last):
  File "habitat_baselines/run.py", line 79, in <module>
    main()
  File "habitat_baselines/run.py", line 40, in main
    run_exp(**vars(args))
  File "habitat_baselines/run.py", line 75, in run_exp
    execute_exp(config, run_type)
  File "habitat_baselines/run.py", line 60, in execute_exp
    trainer.eval()
  File "/usr/local/lib/python3.6/site-packages/habitat_baselines/common/base_trainer.py", line 128, in eval
    checkpoint_index=prev_ckpt_ind,
  File "/usr/local/lib/python3.6/site-packages/habitat_baselines/rl/ppo/ppo_trainer.py", line 512, in _eval_checkpoint
    self.envs = construct_envs(config, get_env_class(config.ENV_NAME))
  File "/usr/local/lib/python3.6/site-packages/habitat_baselines/utils/env_utils.py", line 59, in construct_envs
    scenes = dataset.get_scenes_to_load(config.TASK_CONFIG.DATASET)
  File "/usr/local/lib/python3.6/site-packages/habitat/datasets/pointnav/pointnav_dataset.py", line 43, in get_scenes_to_load
    assert cls.check_config_paths_exist(config)
AssertionError

ronjamoller, 2020-12-30:
Congrats on taking the first hurdle :D - this looks like your dataset is not in the correct place - there are sections on the habitat frontpage about setting up your dataset - you will need the .json files that contain the train and test episodes (start and end point of the robot basically) and the corresponding files for the rooms (.glb and .navmesh). The .jsons have to be in a specific folder structure and so do the room files. In habitat-lab/habitat/config/default.py you can see that the default path for the jsons is `data/datasets/pointnav/habitat-test-scenes/v1/{split}/{split}.json.gz` - split is replaced by test, train, val so its 3 folders and each contain a json. You can download the jsons from the habitat website. The files for the houses (.glb and.navmesh) go into `data/scene_datasets/habitat-test-scenes/` - to double check you can open the .json files and the entries will contain the path of the file that is needed for each episode.

KenaHemnani, 2020-12-30:
Yes, I loaded the dataset. Thank you very much @ronjamoller.
Sorry to bother u again.
After loading the dataset I am getting followiing:
How shall I resolve this:
Traceback (most recent call last):
  File "habitat_baselines/run.py", line 79, in <module>
    main()
  File "habitat_baselines/run.py", line 40, in main
    run_exp(**vars(args))
  File "habitat_baselines/run.py", line 75, in run_exp
    execute_exp(config, run_type)
  File "habitat_baselines/run.py", line 60, in execute_exp
    trainer.eval()
  File "/usr/local/lib/python3.6/site-packages/habitat_baselines/common/base_trainer.py", line 128, in eval
    checkpoint_index=prev_ckpt_ind,
  File "/usr/local/lib/python3.6/site-packages/habitat_baselines/rl/ppo/ppo_trainer.py", line 512, in _eval_checkpoint
    self.envs = construct_envs(config, get_env_class(config.ENV_NAME))
  File "/usr/local/lib/python3.6/site-packages/habitat_baselines/utils/env_utils.py", line 102, in construct_envs
    workers_ignore_signals=workers_ignore_signals,
  File "/usr/local/lib/python3.6/site-packages/habitat/core/vector_env.py", line 153, in __init__
    read_fn() for read_fn in self._connection_read_fns
  File "/usr/local/lib/python3.6/site-packages/habitat/core/vector_env.py", line 153, in <listcomp>
    read_fn() for read_fn in self._connection_read_fns
  File "/usr/local/lib/python3.6/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/usr/local/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/usr/local/lib/python3.6/multiprocessing/connection.py", line 379, in _recv
    chunk = read(handle, remaining)
ConnectionResetError: [Errno 104] Connection reset by peer
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7f5bb975f898>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/habitat/core/vector_env.py", line 553, in __del__
    self.close()
  File "/usr/local/lib/python3.6/site-packages/habitat/core/vector_env.py", line 419, in close
    write_fn((CLOSE_COMMAND, None))
  File "/usr/local/lib/python3.6/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/usr/local/lib/python3.6/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/usr/local/lib/python3.6/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe

ronjamoller, 2020-12-30:
This seems to be an actual issue in the habitat communication procedure, which is unfortunate. I have found other references to this [here](https://www.gitmemory.com/issue/facebookresearch/habitat-api/311/599099949) and [here on github](https://github.com/facebookresearch/habitat-sim/issues/692) but it seems that it was either not resolved or resolved through checking out different branches. So unfortunately i am not knowledgeable enough for this - i dont know what version of habitat you are using but something that has worked for me so far (after rebooting my computer and or restarting the anaconda env - dont laugh ...) is to install the latest **stable** version of habitat-sim (the conda version works for me) and habitat-api according the the installation guide (i dont use docker). I hope this helps although i cant really tell you the issue in this case.

erikwijmans, 2021-01-03:
>  newer checkpoint files contain a config entry, older checkpoint files do not.

The configs have been removed from released models to not tie those weights to a specific version of the config.  We'd have to update those configs for even minor things like adding new fields or changing existing field names.

> as you have pointed out, you can also set the self.config.EVAL.USE_CKPT_CONFIG to false but i dont think the default .yaml files contain that parameter so you would have to set it manually in run.py which is not really straightforward. It is set to 

You can add that to any of the .yaml files as

```
EVAL:
    USE_CKPT_CONFIG: False
```

@KenaHemnani Seeing communication fail on init likely means the subprocesses are getting killed for some reason.  My guess would be out-of-memory.

ronjamoller, 2021-01-04:
Thank you @erikwijmans ! your solution is much more elegant, thank you. Thank you as well for clearing up the config issue - I had (wrongly) assumed that it was introduced when the models got more and more complex to keep track of the changes, I hadnt thought of the involved maintenance. As for the memory issue - this might happen with a big dataset such as matterport or gibson i guess ? The habitat test scenes dataset should not be too big.

erikwijmans, 2021-01-04:
> As for the memory issue - this might happen with a big dataset such as matterport or gibson i guess ? The habitat test scenes dataset should not be too big.

True.  @KenaHemnani can you type `dmesg` immediately after failure and paste part of that log here?  That may give an idea what is going on.

KenaHemnani, 2021-01-05:
Hello @erikwijmans , I am getting following :(after typing dmesg)
`[    0.000000] Linux version 4.19.112+ (builder@a12462ca91c8) (Chromium OS 10.0_pre377782_p20200113-r10 clang version 10.0.0 (/var/cache/chromeos-cache/distfiles/host/egit-src/llvm-project 4e8231b5cf0f5f62c7a51a857e29f5be5cb55734)) #1 SMP Thu Jul 23 08:00:38 PDT 2020
[    0.000000] Command line: BOOT_IMAGE=/syslinux/vmlinuz.A init=/usr/lib/systemd/systemd boot=local rootwait ro noresume noswap loglevel=7 noinitrd console=ttyS0 security=apparmor virtio_net.napi_tx=1 systemd.unified_cgroup_hierarchy=false systemd.legacy_systemd_cgroup_controller=false csm.disabled=1 dm_verity.error_behavior=3 dm_verity.max_bios=-1 dm_verity.dev_wait=1 i915.modeset=1 cros_efi loadpin.enabled=0 module.sig_enforce=0 root=/dev/dm-0 "dm=1 vroot none ro 1,0 4077568 verity payload=PARTUUID=555BDB75-CBD7-CD4A-B24E-29B13D7AC0DF hashtree=PARTUUID=555BDB75-CBD7-CD4A-B24E-29B13D7AC0DF hashstart=4077568 alg=sha256 root_hexdigest=42104d547ac104fb7061529e78f53e4f3e8c3d3cbb040dc6e0f84aad68491347 salt=9dc7f3acc4e2ce65be16356e960c2b21b51a917fa31d2e891fd295490c991e41" mitigations=off
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x0000000000054fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000055000-0x000000000005ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000060000-0x0000000000097fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000098000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bd956fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bd957000-0x00000000bd959fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000bd95a000-0x00000000bd95afff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bd95b000-0x00000000bdadcfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bdadd000-0x00000000bdae4fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bdae5000-0x00000000bdb1afff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bdb1b000-0x00000000beb9afff] usable
[    0.000000] BIOS-e820: [mem 0x00000000beb9b000-0x00000000bebf2fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bebf3000-0x00000000bebfafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000bebfb000-0x00000000bebfefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bebff000-0x00000000bffdffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bffe0000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000037fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi:  ACPI=0xbebfa000  ACPI 2.0=0xbebfa014  SMBIOS=0xbebcd000  MEMATTR=0xbdea0198  TPMEventLog=0xbd2af018 
[    0.000000] SMBIOS 2.4 present.
[    0.000000] DMI: Google Google Compute Engine/Google Compute Engine, BIOS Google 01/01/2011
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 141600001, primary cpu clock
[    0.000000] kvm-clock: using sched offset of 6181546513 cycles
[    0.000002] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000003] tsc: Initial usec timer 3992678
[    0.000004] tsc: Detected 2200.000 MHz processor
[    0.000080] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000082] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000086] last_pfn = 0x380000 max_arch_pfn = 0x400000000
[    0.000116] MTRR default type: write-back
[    0.000116] MTRR fixed ranges enabled:
[    0.000117]   00000-9FFFF write-back
[    0.000118]   A0000-FFFFF uncachable
[    0.000118] MTRR variable ranges enabled:
[    0.000119]   0 base 0000C0000000 mask 3FFFC0000000 uncachable
[    0.000120]   1 disabled
[    0.000120]   2 disabled
[    0.000120]   3 disabled
[    0.000121]   4 disabled
[    0.000121]   5 disabled
[    0.000121]   6 disabled
[    0.000122]   7 disabled
[    0.000131] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000137] last_pfn = 0xbffe0 max_arch_pfn = 0x400000000
[    0.013012] Scanning 1 areas for low memory corruption
[    0.013038] Kernel/User page tables isolation: disabled on command line.
[    0.013043] Using GB pages for direct mapping
[    0.013045] BRK [0x141801000, 0x141801fff] PGTABLE
[    0.013048] BRK [0x141802000, 0x141802fff] PGTABLE
[    0.013049] BRK [0x141803000, 0x141803fff] PGTABLE
[    0.013064] BRK [0x141804000, 0x141804fff] PGTABLE
[    0.013104] BRK [0x141805000, 0x141805fff] PGTABLE
[    0.013120] BRK [0x141806000, 0x141806fff] PGTABLE
[    0.013130] BRK [0x141807000, 0x141807fff] PGTABLE
[    0.013152] BRK [0x141808000, 0x141808fff] PGTABLE
[    0.013172] BRK [0x141809000, 0x141809fff] PGTABLE
[    0.013186] Secure boot disabled
[... 351 lines cut ...]
[   18.421531] br0: port 2(veth965bf25) entered blocking state
[   18.427357] br0: port 2(veth965bf25) entered disabled state
[... cut ...]
[ 2569.917021] Code: e8 97 fe ff ff 48 39 43 20 74 09 48 8b 40 08 5b c3 0f 1f 00 31 c0 5b c3 0f 1f 40 00 41 57 41 56 41 55 41 54 55 53 48 83 ec 08 <48> 8b 1f 4c 8b 77 08 4c 39 f3 74 52 4c 8b 2e 41 89 d7 45 31 e4 49
[ 2570.040635] python[19702]: segfault at 10 ip 00007f7268976bee sp 00007ffdc109be50 error 4 in libCorradeUtility.so.2[7f726893a000+64000]
[ 2570.052970] Code: e8 97 fe ff ff 48 39 43 20 74 09 48 8b 40 08 5b c3 0f 1f 00 31 c0 5b c3 0f 1f 40 00 41 57 41 56 41 55 41 54 55 53 48 83 ec 08 <48> 8b 1f 4c 8b 77 08 4c 39 f3 74 52 4c 8b 2e 41 89 d7 45 31 e4 49`

KenaHemnani, 2021-01-05:
I am running this on Google Colab

erikwijmans, 2021-01-05:
I am not sure what is going on. Does the RL training part of this colab work: https://colab.research.google.com/github/facebookresearch/habitat-lab/blob/master/examples/tutorials/colabs/Habitat_Lab.ipynb ?  Also make sure you are requesting a GPU instance.

KenaHemnani, 2021-01-05:
Yes, @erikwijmans this works fine. Yes, the GPU runtime is selected.

erikwijmans, 2021-01-05:
Hmm, odd.  Maybe there is something with !python in colab then.  `run.py` is pretty simple and you even grab `run_exp` from it, so maybe things will work if you just call into that function directly?

KenaHemnani, 2021-01-05:
> Hmm, odd. Maybe there is something with !python in colab then. `run.py` is pretty simple and you even grab `run_exp` from it, so maybe things will work if you just call into that function directly?

@erikwijmans , sorry Sir I didn't understand what you are asking.

erikwijmans, 2021-01-05:
It looks like you are calling into habitat-baselines via 
```
!python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type train
```

The example colab doesn't do that, it just invokes the trainers directly and that seems to work.  You can also invoke what `run.py` does directly via the `run_exp` function (https://github.com/facebookresearch/habitat-lab/blob/master/habitat_baselines/run.py#L63).  Can you try doing that and see if that fixes it?

KenaHemnani, 2021-01-06:
@erikwijmans yes, I was doing that.
I tried with changing config=get_baselines_config(
        "./habitat_baselines/config/pointnav/ppo_pointnav.yaml"
    )
Under the RL Training section of Habitat-lab colab:
But, it is giving following error:
`AssertionError                            Traceback (most recent call last)

<ipython-input-33-f6db0338b02a> in <module>()
      2     trainer_init = baseline_registry.get_trainer(config.TRAINER_NAME)
      3     trainer = trainer_init(config)
----> 4     trainer.train()

3 frames

/content/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py in get_scenes_to_load(cls, config)
     41         episodes.
     42         """
---> 43         assert cls.check_config_paths_exist(config)
     44         dataset_dir = os.path.dirname(
     45             config.DATA_PATH.format(split=config.SPLIT)

AssertionError: 

`

erikwijmans, 2021-01-06:
That `habitat_baselines/config/pointnav/ppo_pointnav.yaml` config requires the Gibson scene dataset, did you download those scenes?

davkime, 2021-04-19:
> Hi @Sunnyzhr I am not a developer but I have some experience with habitat - I am not sure if you have already solved your issues but i will try to go through them:
>
> 1. You mentioned that you set a [...]

This [issue](https://github.com/facebookresearch/habitat-lab/issues/617) fixes the `TypeError: write_gif() got an unexpected keyword argument 'verbose'` error for me.

rpartsey, 2022-09-04:
Looks like the Issue was resolved.

@KenaHemnani feel free to reopen if you still have questions.

closed 2022-09-04 by rpartsey