facebookresearch/habitat-lab #515: Habitat-SIM Image Extraction use case can't find navmesh

fraazor, 2020-10-28:
I want to run the Image Extraction example from:
https://aihabitat.org/docs/habitat-sim/image-extractor.html

When using the gibson dataset, it worked fine but since I have not added semantics to the gibson dataset yet, I wanted to try a scan from matterport3d. 
The filepath I used is:
path/to/scan/17DRP5sb8fy/region_segmentations/region0.ply

I replaced the path to my ".ply"-file but receive the following error:

WARNING: Logging before InitGoogleLogging() is written to STDERR
I1028 16:28:43.183029 14892 AssetAttributesManager.cpp:112] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
W1028 16:28:43.183069 14892 Simulator.cpp:100] Navmesh file not found, checked at 
path/to/scan/17DRP5sb8fy/region_segmentations/region0.navmesh
I1028 16:28:43.183084 14892 SceneGraph.h:92] Created DrawableGroup: 
Renderer: GeForce RTX 2080 Ti/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 450.51.05
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-forward-compatible-core-context
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
I1028 16:28:43.238240 14892 ResourceManager.cpp:745] Importing Basis files as BC7
MeshTools::compile(): ignoring unknown/unsupported attribute Trade::MeshAttribute::Custom(0)
MeshTools::compile(): ignoring unknown/unsupported attribute Trade::MeshAttribute::Custom(1)
MeshTools::compile(): ignoring unknown/unsupported attribute Trade::MeshAttribute::Custom(2)
MeshTools::compile(): ignoring unknown/unsupported attribute Trade::MeshAttribute::Custom(3)
MeshTools::compile(): ignoring unknown/unsupported attribute Trade::MeshAttribute::Custom(4)
Trade::AbstractImporter::object3D(): index 0 out of range for 0 entries
Aborted (core dumped)

Since glb and ply files are used in examples and gibson and matterport are supported, I am really confused.

erikwijmans, 2020-10-28:
Please make sure you downloaded the habitat task with the mp3d downloader.  We have our own packaging of it the includes the navmeshes.  When using mp3d, you also should point habitat to the glb in that folder.

fraazor, 2020-10-29:
Thank you so much!

Skylion007, 2020-11-02:
Since there has been no updates, presuming this issue is closed.

closed 2020-11-02 by Skylion007