facebookresearch/habitat-lab #521: Scene_id error when initializing environment

jyunis3, 2020-11-05:
## 🐛 Bug

When running the the tutorial notebooks on colab and trying to initialize an environment, there is an error. 

## Command

## To Reproduce

Steps to reproduce the behavior:

https://colab.research.google.com/github/facebookresearch/habitat-lab/blob/master/examples/tutorials/colabs/Habitat_Lab.ipynb#scrollTo=CxLSY9E_0MoW&line=1&uniqifier=1

When running     env = habitat.Env(config=config), the error  occurs.

`2020-11-05 00:37:43,007 Initializing dataset PointNav-v1
Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7fda29800ef0>>
Traceback (most recent call last):
  File "/content/habitat-lab/habitat/core/vector_env.py", line 538, in __del__
    self.close()
  File "/content/habitat-lab/habitat/core/vector_env.py", line 422, in close
    write_fn((CLOSE_COMMAND, None))
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe
2020-11-05 00:37:43,286 initializing sim Sim-v0
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-10-e43c3483e157> in <module>()
      8     except NameError:
      9         pass
---> 10     env = habitat.Env(config=config)

3 frames
/content/habitat-lab/habitat/core/env.py in __init__(self, config, dataset)
    118 
    119         self._sim = make_sim(
--> 120             id_sim=self._config.SIMULATOR.TYPE, config=self._config.SIMULATOR
    121         )
    122         self._task = make_task(

/content/habitat-lab/habitat/sims/registration.py in make_sim(id_sim, **kwargs)
     17         id_sim
     18     )
---> 19     return _sim(**kwargs)
     20 
     21 

/content/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py in __init__(self, config)
    203 
    204         self._sensor_suite = SensorSuite(sim_sensors)
--> 205         self.sim_config = self.create_sim_config(self._sensor_suite)
    206         self._current_scene = self.sim_config.sim_cfg.scene_id
    207         super().__init__(self.sim_config)

/content/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py in create_sim_config(self, _sensor_suite)
    219             config_to=sim_config,
    220         )
--> 221         sim_config.scene_id = self.habitat_config.SCENE
    222         agent_config = habitat_sim.AgentConfiguration()
    223         overwrite_config(

AttributeError: 'habitat_sim._ext.habitat_sim_bindings.SimulatorConfiguration' object has no attribute 'scene_id'`

## Expected behavior

The environment is loaded without error.

## Additional context

erikwijmans, 2020-11-05:
This will get fixed once the new nightly build is done.

jyunis3, 2020-11-06:
Looks like it's working. Thanks.

closed 2020-11-06 by jyunis3