facebookresearch/habitat-lab #522: ImportError: libBulletDynamics.so.2.88: cannot open shared object file: No such file or directory

jyunis3, 2020-11-05:
## 🐛 Bug

When running the profiling tutorial at https://colab.research.google.com/gist/eundersander/b62bb497519b44cf4ceb10e2079525dc/faster-rl-training-profiling-and-optimization.ipynb, the following error occurs when trying to run the test config.:
ImportError: libBulletDynamics.so.2.88: cannot open shared object file: No such file or directory

## Command

## To Reproduce

After installing the tutorial, run the script
```
%%bash
(echo "python habitat_baselines/run.py --exp-config \
habitat_baselines/config/pointnav/ppo_pointnav_profiling.yaml \
--run-type train" > my_program.sh
chmod +x my_program.sh)
```
## Expected behavior

The program runs without error.

## Additional context

dhruvbatra, 2020-11-05:
CC: @eundersander

eundersander, 2020-11-05:
Apologies for finding this tutorial in a broken state and thanks for bringing this to our attention. I'll try to fix soon, within a week.

At a glance, the problem is due to differences between the profiling tutorial's colab_install_profiling.sh and our "standard" tutorial install script here: https://github.com/facebookresearch/habitat-sim/blob/master/examples/colab_utils/colab_install.sh .

eundersander, 2020-11-05:
Also, the problem here is probably with the tutorial colab install, not the underlying Habitat profiling tech. So this error shouldn't discourage you from diving into profiling for your own Habitat program. I suggest watching through the video and also reading documentation at the top of this file:
https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/utils/profiling_utils.py

jyunis3, 2020-11-09:
We are trying to use colab as our envivornment to run Habitat, but I will look through the resources to see what I can do.

Thanks

eundersander, 2020-11-13:
It looks like there's a problem with Bullet and our Habitat-sim Ubuntu conda package. I've added a workaround to fix the tutorial:
https://github.com/eundersander/habitat-api/commit/481a71091bc100d6fbb089969a66b6fdea47c066

You can use this same workaround wherever you're hitting this Bullet link error.

jyunis3, 2020-11-14:
Thanks for updating this.

closed 2022-09-04 by rpartsey