facebookresearch/habitat-lab #523: when execute python setup.py test , it happens error

Lovemyse1f, 2020-11-05:
## ❓ Questions and Help
========= test session starts =============
platform linux -- Python 3.6.12, pytest-6.1.2, py-1.9.0, pluggy-0.13.1
rootdir: /home/hgm/habitat-sim, configfile: setup.cfg, testpaths: tests
plugins: hypothesis-5.41.0, cov-2.10.1
collected 212 items / 1 error / 211 selected                                                     
====== ERRORS =========================
_________________________ ERROR collecting tests/test_data_extraction.py _________________________
tests/test_data_extraction.py:11: in <module>
    from habitat_sim.utils.data.data_extractor import ImageExtractor
habitat_sim/utils/data/__init__.py:9: in <module>
    from habitat_sim.utils.data import data_extractor, data_structures, pose_extractor
habitat_sim/utils/data/data_extractor.py:10: in <module>
    from habitat_sim.utils.data.pose_extractor import PoseExtractor, TopdownView
habitat_sim/utils/data/pose_extractor.py:113: in <module>
    @registry.register_pose_extractor(name="closest_point_extractor")
E   AttributeError: module 'habitat_sim.registry' has no attribute 'register_pose_extractor'
!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!
====== 1 error in 1.40s ========================================

Skylion007, 2020-11-05:
@Lovemyse1f Are you on the latest habitat-sim nightly? We are currently doing a rather large refactor of the Habitat-Sim API so using an old version of Habitat-Sim with the bleeding edge of Habitat-Lab master may cause breakage.

Lovemyse1f, 2020-11-06:
Thanks for answering it.  I installed it  from source ,and not sure about it being habitat-sim nightly.  There is another issue(will i open new issue?) : 
 ~/habitat-sim/build$ ./viewer /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
Renderer: Mesa DRI Intel(R) HD Graphics 5500 (Broadwell GT2) by Intel Open Source Technology Center
OpenGL version: 4.5 (Core Profile) Mesa 19.2.8
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    mesa-implementation-color-read-format-dsa-explicit-binding
    mesa-dsa-createquery-except-pipeline-stats
    mesa-forward-compatible-line-width-range
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1106 16:47:35.509431 10962 AssetAttributesManager.cpp:122] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I1106 16:47:35.509572 10962 AssetAttributesManager.cpp:122] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I1106 16:47:35.509649 10962 AssetAttributesManager.cpp:122] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I1106 16:47:35.509697 10962 AssetAttributesManager.cpp:122] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I1106 16:47:35.509727 10962 AssetAttributesManager.cpp:122] Asset attributes (cubeSolid : cubeSolid) created and registered.
I1106 16:47:35.509749 10962 AssetAttributesManager.cpp:122] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I1106 16:47:35.509807 10962 AssetAttributesManager.cpp:122] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I1106 16:47:35.509860 10962 AssetAttributesManager.cpp:122] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I1106 16:47:35.509891 10962 AssetAttributesManager.cpp:122] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I1106 16:47:35.509920 10962 AssetAttributesManager.cpp:122] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I1106 16:47:35.509956 10962 AssetAttributesManager.cpp:122] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I1106 16:47:35.509995 10962 AssetAttributesManager.cpp:122] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I1106 16:47:35.510015 10962 AssetAttributesManager.cpp:108] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I1106 16:47:35.510869 10962 PhysicsAttributesManager.cpp:39] File (./data/default.phys_scene_config.json) not found so new, default physics manager attributes created and registered.
E1106 16:47:35.510980 10962 StageAttributesManager.cpp:120] StageAttributesManager::registerAttributesTemplateFinalize : Render asset template handle : /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb specified in stage template with handle : /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb does not correspond to any existing file or primitive render asset.  Aborting. 
Segmentation fault (core dumped)

erikwijmans, 2020-11-06:
You need to download the data and then replace `/path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb` with wherever you downloaded it to.  Please follow along here: https://github.com/facebookresearch/habitat-sim#testing

Lovemyse1f, 2020-11-07:
@erikwijmans tks, all errors have been fixed.  The skipped parts which was listed below won't hinder that  i  normally use habitat?
 python setup.py test  

SKIPPED [8] test/test_baseline_agents.py:24: baseline sub-module not installed
SKIPPED [1] test/test_baseline_agents.py:70: baseline sub-module not installed
SKIPPED [64] test/test_baseline_trainers.py:36: baseline sub-module not installed
SKIPPED [4] test/test_baseline_trainers.py:79: baseline sub-module not installed
SKIPPED [1] test/test_baseline_trainers.py:140: baseline sub-module not installed
SKIPPED [1] test/test_baseline_trainers.py:228: baseline sub-module not installed
SKIPPED [1] test/test_demo_notebook.py:21: Please download the Matterport3D PointNav val dataset and Matterport3D val scenes
SKIPPED [2] test/test_examples.py:40: Requires the habitat-test-scenes
SKIPPED [3] test/test_habitat_env.py:105: GPU-GPU requires CUDA
SKIPPED [1] test/test_habitat_env.py:169: GPU-GPU requires CUDA
SKIPPED [1] test/test_habitat_env.py:229: GPU-GPU requires CUDA
SKIPPED [1] test/test_habitat_env.py:278: GPU-GPU requires CUDA
SKIPPED [1] test/test_habitat_env.py:384: unconditional skip
SKIPPED [1] test/test_habitat_task.py:83: Please download dataset to data folder data/datasets/eqa/mp3d/v1/{split}/{split}.json.gz.
SKIPPED [1] test/test_mp3d_eqa.py:100: Please download Matterport3D EQA dataset to data folder.
SKIPPED [2] test/test_mp3d_eqa.py:119: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] test/test_mp3d_eqa.py:166: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] test/test_mp3d_eqa.py:195: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] test/test_mp3d_eqa.py:285: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] test/test_object_nav_task.py:48: Please download Matterport3D ObjectNav Dataset to data folder.
SKIPPED [2] test/test_object_nav_task.py:71: Test skipped as dataset files are missing.
SKIPPED [1] test/test_object_nav_task.py:117: Please download Matterport3D scene and ObjectNav Datasets to data folder.
SKIPPED [1] test/test_pointnav_dataset.py:70: Test skipped as dataset files are missing.
SKIPPED [1] test/test_pointnav_dataset.py:98: Test skipped as dataset files are missing.
SKIPPED [2] test/test_pointnav_dataset.py:122: Test skipped as dataset files are missing.
SKIPPED [1] test/test_r2r_vln.py:44: Please download Matterport3D R2R dataset to data folder.
SKIPPED [3] test/test_r2r_vln.py:66: Please download Matterport3D R2R dataset to data folder.
SKIPPED [1] test/test_r2r_vln.py:114: Please download Matterport3D R2R VLN dataset to data folder.
============================================== 63 passed, 109 skipped, 7 warnings in 240.98s (0:04:00)

closed 2020-11-09 by Lovemyse1f