facebookresearch/habitat-lab #526: How to reliably get the distance to an object via the depth map

ronjamoller, 2020-11-10:
## ❓ Questions and Help
Hi, i am working on a problem where i would like to verify if a small object is visible or not - my plan was to just compare the entry in the depth image to the known distance in 3d space from the agent to the small object. If the depthmap contains a smaller value, the object should be occluded. I am running into an in interesting issue: if i face the object head on i can just multiply the raw values of the depthmap `depth_obs = observations.get("depth", None)` by 10 and i get pretty close - however, if i rotate my camera and do not face the object straight on, the distance reported by the depthmap decreases significantly until the corner of the image is reached. The difference is significant enough that i can't reliably use it to differentiate between occluded and visible objects anymore (for example - when the groundtruth distance is 1.8m the depth map will report 0.18 to 0.13). My guess is that either the turn movement of the agent moves the camera and therefore changes the information of the depth map or there is some distortion issue. I am using the standard depth sensor with 90 degrees hfov.
thanks !
I am happy for all ideas - if there is an alternative way to check for visibility then im happy to hear about that as well.

aclegg3, 2020-11-10:
Hey @ronjamoller, if you are not using a SEMANTIC sensor for anything else, you could setup a custom semantic_id for the object and then just check the semantic sensor image for that id. An example of this is shown in [this tutorial section](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Advanced_Features.ipynb#scrollTo=CePbNAwoljfn).

ronjamoller, 2020-11-10:
Thanks! I will try that

closed 2020-11-16 by Skylion007