facebookresearch/habitat-lab #528: Confused about prev_action_embedding in dd-ppo

StOnEGiggity, 2020-11-12:
## ❓ Questions and Help
I notice that input_dim of prev_action_embedding is 5 in dd-ppo.
https://github.com/facebookresearch/habitat-lab/blob/7c4286653211bbfaca59d0807c28bfb3a6b962bf/habitat_baselines/rl/ddppo/policy/resnet_policy.py#L204

In my view, the previous action may be a one-dimension vector or a four-dimension one-hot vector. why add one during forward.

erikwijmans, 2020-11-12:
0 is given when there is no previous actions, i.e. at the start of the episode.

StOnEGiggity, 2020-11-12:
Thank you for the fast reply!

So prev_action contains "no previous action", "stop", "move_forward", "turn left" and "turn right".
But I am still confused about the code as below:
https://github.com/facebookresearch/habitat-lab/blob/7c4286653211bbfaca59d0807c28bfb3a6b962bf/habitat_baselines/rl/ddppo/policy/resnet_policy.py#L413

Why do you add "1" in this line?
Thanks

erikwijmans, 2020-11-12:
prev_actions just contains the previous action, so  "stop", "move_forward", "turn left" and "turn right", which map to [0, 1, 2, 3].  In the case of the start of the episode, the value of that is undefined however.  The `+1` shifts all the defined values up 1 and then leaves zero for when it is undefined -- the `* masks` will make it zero when a given step is the start of an episode.

closed 2020-11-13 by StOnEGiggity