facebookresearch/habitat-lab #531: Sensor rotation

aveen-d, 2020-11-17:
## ❓ Questions and Help
I would like to know if it is possible to change the rotation of sensors just like we change its position. I would like to use 2 depth sensors and each facing in the opposite direction. I tried doing it but it throws an error that 'rotation' argument is not valid for sensors. Attaching below the code
[image]

aclegg3, 2020-11-17:
You'll need to use `sensor_spec.orientation` instead of `rotation`. Here are [the docs](https://aihabitat.org/docs/habitat-sim/habitat_sim.sensor.SensorSpec.html). Note that orientation is in Euler angles, not quaternion.

aveen-d, 2020-11-17:
> You'll need to use `sensor_spec.orientation` instead of `rotation`. Here are [the docs](https://aihabitat.org/docs/habitat-sim/habitat_sim.sensor.SensorSpec.html). Note that orientation is in Euler angles, not quaternion.

I still get error, it says 'orientation' is not defined. I am using the official colab implementation
[image]

aclegg3, 2020-11-17:
Looks like you may not define the `"orientation"` key in all of your `sensor_param` dicts. From your first post, the `"semantic_sensor"` dict does not have an `"orientation"` entry. You can set this to `[0,0,0]` for default.

aveen-d, 2020-11-17:
> Looks like you may not define the `"orientation"` key in all of your `sensor_param` dicts. From your first post, the `"semantic_sensor"` dict does not have an `"orientation"` entry. You can set this to `[0,0,0]` for default.

Thanks alot for your help and quick reply. Just another small query, can you please let me know the horizontal Field of view of the camera sensor?

aclegg3, 2020-11-17:
Sure, `'hfov'` is configurable in the [`SensorSpec.parameters`](https://aihabitat.org/docs/habitat-sim/habitat_sim.sensor.SensorSpec.html#parameters) dict. Default is 90 degrees.

closed 2020-11-17 by aclegg3

aveen-d, 2020-11-17:
Got it! Thanks for your help :)

AgentEXPL, 2021-07-29:
> You'll need to use `sensor_spec.orientation` instead of `rotation`. Here are [the docs](https://aihabitat.org/docs/habitat-sim/habitat_sim.sensor.SensorSpec.html). Note that orientation is in Euler angles, not quaternion.

@aclegg3 I am not sure about the following things, could you help me to figure out?
1. I have noticed that “orientation: numpy.ndarray[numpy.float32[3, 1]]”. If I set the value of the orientation as "[x,y,z]", does it mean that the pitch, yaw, and roll of the sensor are x, y, z, respectively, as shown in the following figure? And it means to look more toward the ceiling if x is [positive?]
[image]

2. Is it OK to set the value of VFOV or DFOV in the similar way with HFOV?

AgentEXPL, 2021-07-29:
Based on coding and testing, I think the answer to the above problems are as follows.
1. The answer to question 1 is yes. In addition, the value of x needs to be set as radian (i.e., 30/180*pi) rather than degree (30 degrees).
2. The answer to question 2 is yes.
3. The answer to question 3 is no. It seems that only HFOV can be specified in the .yaml config file.