facebookresearch/habitat-lab #534: Tensorflow 2.0 support

aveen-d, 2020-11-22:
## ❓ Questions and Help
Hi, I was following the video tutorial for navigation. When I installed the initial setup files on colab, it installed tensorflow 1.13 version. I wanted to ask, is there support for tensorflow 2.0 version as I have my Actor critic model in tensorflow 2.0. 

Thanks in advance

erikwijmans, 2020-11-22:
The tensorflow install is just for tensorboard, habitat-baselines uses pytorch as its DL framework.

aveen-d, 2020-11-22:
> The tensorflow install is just for tensorboard, habitat-baselines uses pytorch as its DL framework.

I just have another query, how can I integrate my own A2C model written in tensorflow with the environment. Is there any class I can import to make all the necessary connections?

erikwijmans, 2020-11-23:
It's pretty deeply integrated with pytorch (all the model inputs/ouputs are PyTorch tensors, the loss calculations are necessarily done using pytorch tensors, etc) and there isn't a feasible way to use tensorflow code.

The habitat VectorEnv and RLEnv APIs are compatible with the OpenAI gym APIs however, so you should be able to use those in tensorflow RL frameworks like stable baselines.

aveen-d, 2020-11-23:
Hi I tried using the stable baselines framework but the problem is the observation space of habitat lab is space.Dict and stable baselines doesnt support space.Dict it supports space.Box format. Is there a way to convert this ?

Skylion007, 2020-12-03:
@aveen-d You can write a wrapper that converts spaces.Dict to spaces.Box. I think the issue with that API is that it doesn't support multiple sensor types so you need to work around that limitation somehow.

Skylion007, 2020-12-03:
See: https://github.com/hill-a/stable-baselines/issues/133

rothn, 2022-01-07:
Perhaps this refers to removing the 1.x version lock for tensorflow. It seems like this might not be needed in this case, since tensorflow is only needed for tensorboard and not used directly.

erikwijmans, 2022-01-07:
This should have been closed with #740 since that PR removed the tensor flow dependency.

closed 2022-01-07 by erikwijmans