facebookresearch/habitat-lab #540: AttributeError on runing examples/example.py

jsdd25, 2020-12-05:
## ❓ Questions and Help

Hi guys!

I have installed `habitat-sim`.

Then, I have installed all the requirements with the following command:

`pip install -r requirements.txt`,

and then I have built habitat-lab within habitat conda environment with the following command:

'python setup.py develop --all',

and then I executed the command `python examples/example.py` 

Here is the error I got:

`(habitat3) jin@jin-MS-7B94:~/habitat-lab$ python examples/example.py 
2020-12-05 14:29:48,921 Initializing dataset PointNav-v1
2020-12-05 14:31:23,108 initializing sim Sim-v0
Traceback (most recent call last):
  File "examples/example.py", line 28, in <module>
    example()
  File "examples/example.py", line 14, in example
    config=habitat.get_config("configs/tasks/pointnav.yaml")
  File "/home/jin/habitat-lab/habitat/core/env.py", line 105, in __init__
    id_sim=self._config.SIMULATOR.TYPE, config=self._config.SIMULATOR
  File "/home/jin/habitat-lab/habitat/sims/registration.py", line 19, in make_sim
    return _sim(**kwargs)
  File "/home/jin/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 182, in __init__
    self.sim_config = self.create_sim_config(self._sensor_suite)
  File "/home/jin/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 198, in create_sim_config
    sim_config.scene.id = self.habitat_config.SCENE
AttributeError: 'habitat_sim._ext.habitat_sim_bindings.SimulatorCon' object has no attribute 'scene'
Exception ignored in: <bound method Simulator.__del__ of HabitatSim(config=NOTHING, agents=NOTHING, _num_total_frames=NOTHING, _default_agent=NOTHING, _sensors=NOTHING, _initialized=NOTHING, _previous_step_time=NOTHING)>
Traceback (most recent call last):
  File "/home/jin/anaconda3/envs/habitat3/lib/python3.6/site-packages/habitat_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/simulator.py", line 321, in __del__
    self.close()
  File "/home/jin/anaconda3/envs/habitat3/lib/python3.6/site-packages/habitat_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/simulator.py", line 109, in close
    for sensor in self._sensors.values():
AttributeError: 'HabitatSim' object has no attribute '_sensors'`.

Could anyone help me with this? Thank you very much!

mathfac, 2020-12-05:
Hi Jin,
Something wrong with habitat-sim commit/version you installed. How
habitat-sim was installed?

jsdd25, 2020-12-05:
> Hi Jin, Something wrong with habitat-sim commit/version you installed. How habitat-sim was installed?
> […](#)

Thanks a lot for your help and quick reply. 

First, I downloaded the latest release of habitat-sim with the following command:

`git clone https://github.com/facebookresearch/habitat-sim.git --recursive`

Then, I installed some dependencies with the following commands:

`conda create -n habitat3 python=3.6 cmake=3.14.0
conda activate habitat3
pip install -r requirements.txt`.

Then, I have built Habitat-Sim within habitat conda environment with the following command:

`python setup.py install`,

and then I can successfully run the `examples/example.py` of Habitat-Sim within habitat conda environment with the following command:

`python examples/example.py --scene /home/jin/habitat-test-scenes/data/scene_datasets/habitat-test-scenes/skokloster- castle.glb`.

However, I got an attributeError on runing `examples/example.py` of habitat-lab as mentioned above.

erikwijmans, 2020-12-05:
You seem to be on an older habitat-lab version.  You'll need to change `scene.id` to `scene_id`

jsdd25, 2020-12-06:
Thank you very much! I have successfully run `examples/example.py`.

closed 2020-12-06 by jsdd25

2201957, 2023-09-06:
this is a good question