facebookresearch/habitat-lab #543: Step physics seems to not work properly

gianscarpe, 2020-12-14:
## ❓ Questions and Help
Hi, I'm following the tutorial at https://aihabitat.org/docs/habitat-sim/rigid-object-tutorial.html, but the ball don't fall to the ground even if "step_physics" is properly called. It seems that `sim.step_pyhsics` is not actually simulating physics, and therefore the result is a ball still in mid-air in all the observations. Am I forgetting something?
My environment:
python 3.8
habitat_sim 0.1.6 installed with conda

aclegg3, 2020-12-14:
Sounds like the version of Habitat-sim you installed does not have bullet physics enabled. You should see a message in the terminal output that indicates this.

gianscarpe, 2020-12-14:
Hi @aclegg3, thanks for answering! Actually, the only physics-related log on the terminal is
```
PhysicsAttributesManager.cpp:27] File (./data/default.physics_config.json) not found, so new default physics manager attributes created and registered.
```

aclegg3, 2020-12-14:
Can you provide the full log here? Which conda install did you use?

gianscarpe, 2020-12-14:
Yes sure :+1:
I installed it from nightly channel: `conda install habitat-sim withbullet  -c conda-forge -c aihabitat-nightly`. I tried with `conda install habitat-sim  -c conda-forge -c aihabitat-nightly` as well.
The full log is here:
```
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1214 17:18:13.351907 304462 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : default changed to proposed JSON configuration filename : default.scene_dataset_config.json
I1214 17:18:13.351999 304462 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I1214 17:18:13.352063 304462 AssetAttributesManager.cpp:117] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I1214 17:18:13.352141 304462 AssetAttributesManager.cpp:117] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I1214 17:18:13.352214 304462 AssetAttributesManager.cpp:117] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I1214 17:18:13.352238 304462 AssetAttributesManager.cpp:117] Asset attributes (cubeSolid : cubeSolid) created and registered.
I1214 17:18:13.352267 304462 AssetAttributesManager.cpp:117] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I1214 17:18:13.352295 304462 AssetAttributesManager.cpp:117] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I1214 17:18:13.352350 304462 AssetAttributesManager.cpp:117] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I1214 17:18:13.352363 304462 AssetAttributesManager.cpp:117] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I1214 17:18:13.352375 304462 AssetAttributesManager.cpp:117] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I1214 17:18:13.352393 304462 AssetAttributesManager.cpp:117] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I1214 17:18:13.352411 304462 AssetAttributesManager.cpp:117] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I1214 17:18:13.352416 304462 AssetAttributesManager.cpp:105] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I1214 17:18:13.352939 304462 SceneDatasetAttributesManager.cpp:23] File (default) not found, so new default dataset attributes created and registered.
I1214 17:18:13.352946 304462 MetadataMediator.cpp:46] MetadataMediator::createDataset : Dataset default successfully created.
I1214 17:18:13.356528 304462 PhysicsAttributesManager.cpp:27] JSON Configuration File (./data/default.physics_config.json) based physics manager attributes created and registered.
I1214 17:18:13.356539 304462 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : /home/gianlucascarpellini/dev/habitat-sim/data/scene_datasets/habitat-test-scenes/apartment_1.glb changed to proposed JSON configuration filename : /home/gianlucascarpellini/dev/habitat-sim/data/scene_datasets/habitat-test-scenes/apartment_1.stage_config.json
I1214 17:18:13.356631 304462 AbstractObjectAttributesManagerBase.h:180] File (/home/gianlucascarpellini/dev/habitat-sim/data/scene_datasets/habitat-test-scenes/apartment_1.glb) exists but is not a recognized config filename extension, so new default Stage attributes created and registered.
I1214 17:18:13.356662 304462 Simulator.cpp:153] Loading navmesh from /home/gianlucascarpellini/dev/habitat-sim/data/scene_datasets/habitat-test-scenes/apartment_1.navmesh
I1214 17:18:13.356755 304462 Simulator.cpp:155] Loaded.
I1214 17:18:13.356765 304462 SceneGraph.h:93] Created DrawableGroup: 
Renderer: GeForce GTX 1650/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 450.80.02
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-forward-compatible-core-context
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
I1214 17:18:13.485472 304462 ResourceManager.cpp:247] ResourceManager::loadStage : Not loading semantic mesh
I1214 17:18:13.485492 304462 ResourceManager.cpp:1126] Importing Basis files as BC7
W1214 17:18:14.356468 304462 Simulator.cpp:243] :
---
 The active scene does not contain semantic annotations. 
---
I1214 17:18:14.356747 304462 simulator.py:213] Loaded navmesh /home/gianlucascarpellini/dev/habitat-sim/data/scene_datasets/habitat-test-scenes/apartment_1.navmesh
[... cut ...]
I1214 17:18:15.746168 304462 SceneManager.h:25] Deconstructing SceneManager
I1214 17:18:15.746171 304462 SceneGraph.h:26] Deconstructing SceneGraph
I1214 17:18:15.746603 304462 Sensor.h:87] Deconstructing Sensor
I1214 17:18:15.747010 304462 Sensor.h:87] Deconstructing Sensor
I1214 17:18:15.747270 304462 Sensor.h:87] Deconstructing Sensor
I1214 17:18:15.752295 304462 Renderer.cpp:34] Deconstructing Renderer
I1214 17:18:15.752308 304462 WindowlessContext.h:17] Deconstructing WindowlessContext
I1214 17:18:15.755002 304462 Simulator.cpp:47] Deconstructing `Simulator`
```

gianscarpe, 2020-12-14:
[image]

aclegg3, 2020-12-14:
I see. The "withbullet" build for habitat-nightly does not appear to be up-dating correctly at this time. You should in this case build from source or use the previous version tag (e.g. use `-c aihabitat` instead).

gianscarpe, 2020-12-15:
Hi @aclegg3, Thanks! I solved by build with this (the following code will build two wheels, both with and without display)

```
# Checkout the latest stable release                                                                                                                                                              
cd dev                                                                                                                                                                                            
git clone --branch stable https://github.com/facebookresearch/habitat-sim.git                                                                                                                     
cd habitat-sim                                                                                                                                                                                    
                                                                                                                                                                                                        
# We require python>=3.6 and cmake>=3.10                                                                                                                                                          
conda create -n habitat python=3.8 cmake=3.14.0                                                                                                                                                   
conda activate habitat                                                                                                                                                                            
pip install -r requirements.txt                                                                                                                                                                   
                                                                                                                                                                                                        
python setup.py bdist_wheel --with-cuda --bullet                                                                                                                                                  
python setup.py bdist_wheel --with-cuda --bullet --headless # Headless, for cluster     
```

closed 2020-12-15 by gianscarpe