facebookresearch/habitat-lab #544: how to get a continuous path from pointnav

mmahdavian, 2020-12-16:
Hi

I am trying to move an agent from any point to another point using pointnav path generation method and get its path points. I need the points to be closed to each other to get an almost continuous path. Can anyone help me with this? In pointnav_generator.py code I am using get_action_shortest_path function and printing state.position. But it's not continuous. How should I modify it to make a path with small steps?
(I want to record this path and get rgb images from an agent moving in this path and use it for orb-slam. So it needs to be continuous)

Thank You

Skylion007, 2020-12-28:
You just need to decrease the agent step_size to be a really small value in its action config: https://github.com/facebookresearch/habitat-lab/blob/b7a93bc493f7fb89e5bf30b40be204ff7b5570d7/habitat/datasets/utils.py#L179

mmahdavian, 2021-01-06:
Thanks.It helped

closed 2021-01-06 by mmahdavian