facebookresearch/habitat-lab #545: Standardized Gym Task and Action Wrapper

Skylion007, 2020-12-16:
## 🚀 Feature

Standardized Habitat-Lab gym wrapper
## Motivation

* Currently a lot of Gym interfaces like Stable Baselines can work with Habitat-Lab, but usually need a bit of work to make it out of the box compatible with code. The gym interface is nice, but has several limitations that make it unusable for Habitat's longterm design (like lack of multi-agent support, lack of privileged sensors, etc...). This issue proposes a standardized wrapper around Habitat-Lab to make it fully compatible with these mode.
* We are most of the way there, but we need a wrapper for Gym style actions, observation spaces, and flattening out of observation dicts.
## Pitch
Doing so would make running Tensorflow code and Stable Baselines code relatively easy as part of Habitat-Lab.

## Additional context
* Would allow habitat-lab to work out of the box on more tasks.
* This issue would fix #534

Skylion007, 2020-12-16:
@aveen-d Did you make any progress on this front?

aveen-d, 2020-12-16:
> @aveen-d Did you make any progress on this front?

Hey, I tried to create a wrapper but I couldn't make it work. Instead I developed my own RL agent from  scratch using tensorflow 2.0 and was able to run it the Habitat Lab.