facebookresearch/habitat-lab #547: VTK/GLEW  Error when use both habitat-lab and Mayavi

Skylark0924, 2020-12-19:
## 🐛 Bug
**I wrote a script about using RGB & depth images to reconstruct a 3D map and display it using `Mayavi`. However, I met an extremely weird bug.  I found that I can not show the Mayavi 3D image and its screenshot after creating a habitat environment.**

## Command

## To Reproduce

1. Here is the `reconstruction.py` script:
```
import quaternion
import numpy as np
import random

from PIL import Image
from matplotlib.colors import ListedColormap
import matplotlib.pyplot as plt
from matplotlib.axes._axes import _log as matplotlib_axes_logger

from Env.habitat_dataset.NavRL_env import NavRLEnv
from habitat.config import Config

matplotlib_axes_logger.setLevel('ERROR')
np.warnings.filterwarnings('ignore')

def example_config():
    # task_config = habitat.get_config(
    #     config_paths='/home/skylark/datasets/habitat/configs/tasks/objectnav_mp3d.yaml')
    task_config = habitat.get_config(
        config_paths='/home/skylark/datasets/habitat/configs/tasks/objectnav_mp3d.yaml')
    task_config.defrost()
    task_config.DATASET.DATA_PATH = '/home/skylark/datasets/habitat/data/datasets/objectnav/mp3d/v1/val/val.json.gz'
    task_config.DATASET.DATA_PATH_DIR = '/home/skylark/datasets/habitat/data/datasets/objectnav/mp3d/v1/val/content'
    task_config.DATASET.SCENES_DIR = '/home/skylark/datasets/habitat/data/scene_datasets'
    task_config.SIMULATOR.AGENT_0.SENSORS = ['RGB_SENSOR', 'DEPTH_SENSOR', 'SEMANTIC_SENSOR']
    task_config.SIMULATOR.TURN_ANGLE = 30
    task_config.glove_file = '/home/skylark/datasets/habitat/data/mp3d_glove.json'
    task_config.freeze()

    env_config = {
        'TASK_CONFIG': task_config,
        'RL': {
            'SUCCESS_REWARD': 2.5,
            'SLACK_REWARD': -1e-3,
            'REWARD_MEASURE': "distance_to_goal",
            'SUCCESS_MEASURE': "spl",
            'EPISODE_LENGTH': 300
        }
    }
    return env_config

class HabitatEnv:
    def __init__(self, config):
        self.config = Config(config)
        self.config.freeze()
        self.scene_path = '/home/skylark/datasets/habitat/data/scene_datasets/mp3d'

        # self.glove = Glove(self.config['TASK_CONFIG']['glove_file'])
        self.eng = NavRLEnv(config=self.config)
        self.eng.reset()

def from_depth_to_xyz(depth, cam2world, hfov=90):
    """[summary]

===>input coordinate (2D)
         
          width(e.g. 640)
         o--------------> y  (second axis)
         |
 height  |
(e.g.480)↓
[... 275 lines cut ...]
I1219 17:14:13.042873 19251 Sensor.h:81] Deconstructing Sensor
I1219 17:14:13.043201 19251 Sensor.h:81] Deconstructing Sensor
I1219 17:14:13.043211 19251 SceneGraph.h:26] Deconstructing SceneGraph
I1219 17:14:13.055938 19251 Renderer.cpp:34] Deconstructing Renderer
I1219 17:14:13.055974 19251 WindowlessContext.h:17] Deconstructing WindowlessContext
I1219 17:14:14.646723 19251 Simulator.cpp:46] Deconstructing Simulator

Process finished with exit code 0
```

Here is the wrong display:

[image]

## Expected behavior

If I delete this line in the main function in `reconstruction.py`:
 ```
env = HabitatEnv(env_config)
``` 
Then I can get the mayavi image and its screenshot correctly

[image]

## Additional context

erikwijmans, 2020-12-19:
We'd need to add something to habitat to allow it to piggy back off Mayavi's context to keep both in the same process (assuming mayavi is written correctly, of course).  You can work around this by using `habitat.VectorEnv` (https://github.com/facebookresearch/habitat-lab/blob/master/habitat/core/vector_env.py#L78) with 1 env as that'll run the habitat instance in a separate process

Skylark0924, 2020-12-20:
> We'd need to add something to habitat to allow it to piggy back off Mayavi's context to keep both in the same process (assuming mayavi is written correctly, of course). You can work around this by [...]

Awesome!! That's what I need! Here is my solution.

1. I use `construct_envs` function in [`env_utils.py`](https://github.com/facebookresearch/habitat-lab/blob/master/habitat_baselines/utils/env_utils.py)
2. Change the `HabitatEnv` class as :
```
    class HabitatEnv:
        def __init__(self, config):
            self.config = Config(config)
            self.config.freeze()
            self.scene_path = '/home/skylark/datasets/habitat/data/scene_datasets/mp3d'

            # self.eng = NavRLEnv(config=self.config)
            self.eng = construct_envs(config=self.config, env_class=NavRLEnv)
            self.eng.reset()
```
3.  Add some config
```
        env_config = {
            'TASK_CONFIG': task_config,
            'RL': {
                'SUCCESS_REWARD': 2.5,
                'SLACK_REWARD': -1e-3,
                'REWARD_MEASURE': "distance_to_goal",
                'SUCCESS_MEASURE': "spl",
                'EPISODE_LENGTH': 300
            },
            'NUM_PROCESSES': 1,
            'SIMULATOR_GPU_ID': 0
        }
```
4. **Then I got the right mayavi images and created HabitatEnv at the same time!!**

Skylark0924, 2020-12-20:
While it seems a little slow.

Skylark0924, 2020-12-20:
I got another problem.

**How can I get the property of my Env class if it is packed in a `VertorEnv` thread?**

For example, I used to get the episode number by calling `env.episodes`. But now it is unallowed in a VectorEnv. I just find a method called `call_at`, but it's just for the function.

mathfac, 2021-01-14:
Hi @Skylark0924,
Try [new call_at function](https://github.com/facebookresearch/habitat-lab/blob/c19e5666aac3d17ec55f09122ce4acff0873d4b4/habitat/core/vector_env.py#L458-L478) developed by @erikwijmans. Let us know if the issue was resolved.

Skylark0924, 2021-01-16:
Sorry for replying late, I finally use the plot3d function instead of mayavi. 
The solution you provided looks valuable for getting the property of one of the Env class in a VectorEnv thread. Thanks a lot and let's close this issue!

closed 2022-09-04 by rpartsey