facebookresearch/habitat-lab #549: Error occurs when running "python setup.py test"

PatZhuang, 2020-12-25:
## 🐛 Bug

So I followed the instruction in readme and successfully built habitat-sim and habitat-lab, and I ran `python examples/example.py` and `python examples/benchmark.py` with expected results.

But when I run `python setup.py test`, some error occurs like:

```
test/test_baseline_agents.py .........                                                                                                                                                               [  5%]
test/test_baseline_trainers.py FFFFFFFFFFFFFFFFFFFF....FFFException ignored in: <function VectorEnv.__del__ at 0x7fdf1cc84b90>
Traceback (most recent call last):
  File "/remote-home/my/habitat-lab/habitat/core/vector_env.py", line 529, in __del__
    self.close()
  File "/remote-home/my/habitat-lab/habitat/core/vector_env.py", line 413, in close
    write_fn((CLOSE_COMMAND, None))
  File "/root/anaconda3/envs/vln/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/root/anaconda3/envs/vln/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/root/anaconda3/envs/vln/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
OSError: [Errno 9] Bad file descriptor
Exception ignored in: <function VectorEnv.__del__ at 0x7fdf1cc84b90>
Traceback (most recent call last):
  File "/remote-home/my/habitat-lab/habitat/core/vector_env.py", line 529, in __del__
    self.close()
  File "/remote-home/my/habitat-lab/habitat/core/vector_env.py", line 413, in close
    write_fn((CLOSE_COMMAND, None))
  File "/root/anaconda3/envs/vln/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/root/anaconda3/envs/vln/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/root/anaconda3/envs/vln/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
OSError: [Errno 9] Bad file descriptor
Exception ignored in: <function VectorEnv.__del__ at 0x7fdf1cc84b90>
Traceback (most recent call last):
  File "/remote-home/my/habitat-lab/habitat/core/vector_env.py", line 529, in __del__
    self.close()
  File "/remote-home/my/habitat-lab/habitat/core/vector_env.py", line 413, in close
    write_fn((CLOSE_COMMAND, None))
  File "/root/anaconda3/envs/vln/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/root/anaconda3/envs/vln/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/root/anaconda3/envs/vln/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
OSError: [Errno 9] Bad file descriptor
```

How could I fix this?

Skylion007, 2020-12-28:
The subprocesses are having trouble communicating, it's likely the simulator subprocesses are dying (segfaulting) or such. Ensure that everything is properly installed and all test asset files are available and in the proper location.

@erikwijmans might have a better idea of what is going on.

mathfac, 2020-12-28:
@PatZhuang  can you try older commit before https://github.com/facebookresearch/habitat-lab/commit/7537389aaf5bbd3c3a8524c492c4d520311b9c9d: `git checkout b7a93bc493f7fb89e5bf30b40be204ff7b5570d7 ` and run `python setup.py test` there. Thank you!

mathfac, 2020-12-28:
Otherwise try clean conda environment and install again.

PatZhuang, 2020-12-29:
> @PatZhuang can you try older commit before [7537389](https://github.com/facebookresearch/habitat-lab/commit/7537389aaf5bbd3c3a8524c492c4d520311b9c9d): `git checkout b7a93bc493f7fb89e5bf30b40be204ff7b5570d7 ` and run `python setup.py test` there. Thank you!

Actually I switched to branch v0.1.6 and this problem no longer occurred. The test process finished with 11 tasks failed and 25 tasks skipped( I didn't download the full matterport3D dataset). But I don't see any explanation to why those tasks had failed, except for some "skipped" notice in the last lines of the output which are caused for I did not download the full matterport3D dataset.

mathfac, 2021-01-14:
Is that still `test/test_baseline_trainers.py FFFFFFFFFFFFFFFFFFFF....FFFE` failing?

Maybe, you can provide additional System Info to address the issue:

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/master/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux):
 - How you installed PyTorch (`conda`, `pip`, source):
 - Build command you used (if compiling from source):
 - Python version:
 - GPU models and configuration:
 - CUDA version:
 - PIP/conda dependencies packages versions
 - Any other relevant information:

rpartsey, 2022-09-04:
Closing the issue because it has not had recent activity.

closed 2022-09-04 by rpartsey