facebookresearch/habitat-lab #550: Loading config file into NavRLEnv

aveen-d, 2020-12-26:
## ❓ Questions and Help
I loaded "ppo_pointnav_example.yaml" file using get_baselines_config method. The output of the congif file is the following :

BASE_TASK_CONFIG_PATH: configs/tasks/pointnav.yaml
CHECKPOINT_FOLDER: data/new_checkpoints
CHECKPOINT_INTERVAL: 50
DATASET:
  CONTENT_SCENES: ['*']
  DATA_PATH: data/datasets/pointnav/habitat-test-scenes/v1/{split}/{split}.json.gz
  SCENES_DIR: data/scene_datasets
  SPLIT: train
  TYPE: PointNav-v1
ENVIRONMENT:
  ITERATOR_OPTIONS:
    CYCLE: True
    GROUP_BY_SCENE: True
    MAX_SCENE_REPEAT_EPISODES: -1
    MAX_SCENE_REPEAT_STEPS: 10000
    NUM_EPISODE_SAMPLE: -1
    SHUFFLE: True
    STEP_REPETITION_RANGE: 0.2
  MAX_EPISODE_SECONDS: 10000000
  MAX_EPISODE_STEPS: 1000
ENV_NAME: NavRLEnv
EVAL_CKPT_PATH_DIR: data/new_checkpoints
LOG_INTERVAL: 10
NUM_PROCESSES: 1
NUM_UPDATES: 10000
PYROBOT:
  BASE_CONTROLLER: proportional
  BASE_PLANNER: none
  BUMP_SENSOR:
    TYPE: PyRobotBumpSensor
  DEPTH_SENSOR:
    CENTER_CROP: False
    HEIGHT: 480
    MAX_DEPTH: 5.0
    MIN_DEPTH: 0.0
    NORMALIZE_DEPTH: True
    TYPE: PyRobotDepthSensor
    WIDTH: 640
  LOCOBOT:
    ACTIONS: ['BASE_ACTIONS', 'CAMERA_ACTIONS']
    BASE_ACTIONS: ['go_to_relative', 'go_to_absolute']
    CAMERA_ACTIONS: ['set_pan', 'set_tilt', 'set_pan_tilt']
  RGB_SENSOR:
    CENTER_CROP: False
    HEIGHT: 480
    TYPE: PyRobotRGBSensor
    WIDTH: 640
  ROBOT: locobot
  ROBOTS: ['locobot']
  SENSORS: ['RGB_SENSOR', 'DEPTH_SENSOR', 'BUMP_SENSOR']
RL:
  PPO:
    clip_param: 0.1
    entropy_coef: 0.01
    eps: 1e-5
    gamma: 0.99
    hidden_size: 512
    lr: 0.00025
    max_grad_norm: 0.5
    num_mini_batch: 1
    num_steps: 128
    ppo_epoch: 4
    reward_window_size: 50
    tau: 0.95
    use_gae: True
    use_linear_clip_decay: True
    use_linear_lr_decay: True
    value_loss_coef: 0.5
SEED: 100
SENSORS: ['RGB_SENSOR', 'DEPTH_SENSOR']
SIMULATOR:
  ACTION_SPACE_CONFIG: v0
  AGENTS: ['AGENT_0']
  AGENT_0:
    ANGULAR_ACCELERATION: 12.56
    ANGULAR_FRICTION: 1.0
    COEFFICIENT_OF_RESTITUTION: 0.0
    HEIGHT: 1.5
    IS_SET_START_STATE: False
    LINEAR_ACCELERATION: 20.0
    LINEAR_FRICTION: 0.5
    MASS: 32.0
    RADIUS: 0.1
    SENSORS: ['RGB_SENSOR']
    START_POSITION: [0, 0, 0]
    START_ROTATION: [0, 0, 0, 1]
  DEFAULT_AGENT_ID: 0
  DEPTH_SENSOR:
    HEIGHT: 480
    HFOV: 90
    MAX_DEPTH: 10.0
    MIN_DEPTH: 0.0
    NORMALIZE_DEPTH: True
    ORIENTATION: [0.0, 0.0, 0.0]
    POSITION: [0, 1.25, 0]
    TYPE: HabitatSimDepthSensor
    WIDTH: 640
  FORWARD_STEP_SIZE: 0.25
  HABITAT_SIM_V0:
    ALLOW_SLIDING: True
    ENABLE_PHYSICS: False
    GPU_DEVICE_ID: 0
    GPU_GPU: False
    PHYSICS_CONFIG_FILE: ./data/default.physics_config.json
  RGB_SENSOR:
    HEIGHT: 480
    HFOV: 90
    ORIENTATION: [0.0, 0.0, 0.0]
    POSITION: [0, 1.25, 0]
    TYPE: HabitatSimRGBSensor
    WIDTH: 640
  SCENE: data/scene_datasets/habitat-test-scenes/van-gogh-room.glb
  SEED: 100
  SEMANTIC_SENSOR:
    HEIGHT: 480
    HFOV: 90
    ORIENTATION: [0.0, 0.0, 0.0]
    POSITION: [0, 1.25, 0]
    TYPE: HabitatSimSemanticSensor
    WIDTH: 640
  TILT_ANGLE: 15
  TURN_ANGLE: 10
  TYPE: Sim-v0
SIMULATOR_GPU_ID: 0
TASK:
  ACTIONS:
    ANSWER:
      TYPE: AnswerAction
    LOOK_DOWN:
      TYPE: LookDownAction
    LOOK_UP:
      TYPE: LookUpAction
    MOVE_FORWARD:
      TYPE: MoveForwardAction
    STOP:
      TYPE: StopAction
    TELEPORT:
      TYPE: TeleportAction
    TURN_LEFT:
      TYPE: TurnLeftAction
    TURN_RIGHT:
      TYPE: TurnRightAction
  ANSWER_ACCURACY:
    TYPE: AnswerAccuracy
  COLLISIONS:
    TYPE: Collisions
  COMPASS_SENSOR:
    TYPE: CompassSensor
  CORRECT_ANSWER:
    TYPE: CorrectAnswer
  DISTANCE_TO_GOAL:
    DISTANCE_TO: POINT
    TYPE: DistanceToGoal
  EPISODE_INFO:
    TYPE: EpisodeInfo
  GOAL_SENSOR_UUID: pointgoal
  GPS_SENSOR:
    DIMENSIONALITY: 2
    TYPE: GPSSensor
  HEADING_SENSOR:
    TYPE: HeadingSensor
  IMAGEGOAL_SENSOR:
    TYPE: ImageGoalSensor
  INSTRUCTION_SENSOR:
    TYPE: InstructionSensor
  INSTRUCTION_SENSOR_UUID: instruction
  MEASUREMENTS: []
  OBJECTGOAL_SENSOR:
    GOAL_SPEC: TASK_CATEGORY_ID
    GOAL_SPEC_MAX_VAL: 50
    TYPE: ObjectGoalSensor
  POINTGOAL_SENSOR:
    DIMENSIONALITY: 2
    GOAL_FORMAT: POLAR
    TYPE: PointGoalSensor
  POINTGOAL_WITH_GPS_COMPASS_SENSOR:
    DIMENSIONALITY: 2
    GOAL_FORMAT: POLAR
    TYPE: PointGoalWithGPSCompassSensor
  POSSIBLE_ACTIONS: ['STOP', 'MOVE_FORWARD', 'TURN_LEFT', 'TURN_RIGHT']
  PROXIMITY_SENSOR:
    MAX_DETECTION_RADIUS: 2.0
    TYPE: ProximitySensor
  QUESTION_SENSOR:
    TYPE: QuestionSensor
  SENSORS: []
  SOFT_SPL:
    TYPE: SoftSPL
  SPL:
    TYPE: SPL
  SUCCESS:
    SUCCESS_DISTANCE: 0.2
    TYPE: Success
  SUCCESS_DISTANCE: 0.2
  TOP_DOWN_MAP:
    DRAW_BORDER: True
    DRAW_GOAL_AABBS: True
    DRAW_GOAL_POSITIONS: True
    DRAW_SHORTEST_PATH: True
    DRAW_SOURCE: True
    DRAW_VIEW_POINTS: True
    FOG_OF_WAR:
      DRAW: True
      FOV: 90
      VISIBILITY_DIST: 5.0
    MAP_PADDING: 3
    MAP_RESOLUTION: 1024
    MAX_EPISODE_STEPS: 1000
    TYPE: TopDownMap
  TYPE: Nav-v0
TENSORBOARD_DIR: tb
TEST_EPISODE_COUNT: 2
TORCH_GPU_ID: 0
TRAINER_NAME: ppo
VIDEO_DIR: video_dir
VIDEO_OPTION: ['disk', 'tensorboard']

I now loaded this config file into NavRLEnv using the following command
env = NavRLEnv(config=config)
The problem I face is after this, when I use the command env.reset(). This command should return the observation from sensors. In the config file it says two sensor, 'RGB sensor' and 'DEPTH sensor' is present but when I print the values of env.reset() it shows only rgb sensor values. The output is shown below. 

{'rgb': array([[[109, 103, 105],
        [110, 104, 106],
        [110, 104, 107],
        ...,
        [204, 183, 153],
        [205, 184, 154],
        [207, 186, 157]],

       [[112, 105, 107],
        [113, 106, 108],
        [115, 108, 110],
        ...,
        [214, 193, 164],
        [214, 193, 164],
        [215, 194, 165]],

       [[123, 114, 114],
        [127, 119, 117],
        [118, 111, 112],
        ...,
        [214, 194, 167],
        [212, 192, 165],
        [209, 189, 162]],

       ...,

       [[183, 163, 152],
        [184, 164, 153],
        [185, 165, 154],
        ...,
        [148, 152, 164],
        [150, 153, 165],
        [152, 154, 167]],

       [[181, 161, 150],
        [183, 163, 152],
        [184, 164, 153],
        ...,
        [148, 151, 163],
        [152, 154, 166],
        [154, 156, 169]],

       [[180, 160, 149],
        [181, 161, 150],
        [182, 162, 151],
        ...,
        [142, 144, 155],
        [149, 150, 162],
        [153, 155, 166]]], dtype=uint8), 'pointgoal_with_gps_compass': array([10.900574 ,  1.5645034], dtype=float32)}

Can you please let me know how to access the depth sensor values?

mathfac, 2020-12-26:
Hi @aveen-d,
You use a baseline config, that uses `/configs/tasks/pointnav.yaml` as Environment config and that config has only RGB sensor specified:
https://github.com/facebookresearch/habitat-lab/blob/7537389aaf5bbd3c3a8524c492c4d520311b9c9d/configs/tasks/pointnav.yaml#L5

Sensors specified in the baseline config are used only when you run Trainer.

mathfac, 2020-12-26:
Also, if you use habitat_baseline config as root then you expected to do:
```
env = NavRLEnv(config=config.TASK_CONFIG)
```

aveen-d, 2020-12-28:
> Also, if you use habitat_baseline config as root then you expected to do:
> 
> ```
> env = NavRLEnv(config=config.TASK_CONFIG)
> ```

Thank you, it worked.

closed 2020-12-28 by aveen-d