facebookresearch/habitat-lab #551: How to set pose fixed after env.reset()?

Jthon-lab, 2020-12-27:
## ❓ Questions and Help
I was wondering how to set pose fixed after reset the environment in habitat.RLEnv, I tried to use super().seed() but it didn't work.

erikwijmans, 2020-12-27:
The position after reset is determined by the episode. Episodes are each completed once before being repeated.  If you want the position to be the same after reset, you will need to make a dataset with 1 episode.  This can be done via https://github.com/facebookresearch/habitat-lab/blob/master/habitat/core/dataset.py#L197

mathfac, 2021-01-14:
Extending Erik's answer you can use next config options to enable single episode loading in cycle mode:
```
ENVIRONMENT:
  ITERATOR_OPTIONS:
    CYCLE = True
    GROUP_BY_SCENE: True
    NUM_EPISODE_SAMPLE: 1
    SHUFFLE: False

```
Please, close the issue if we answered your question.

Jthon-lab, 2021-01-14:
> Extending Erik's answer you can use next config options to enable single episode loading in cycle mode:
> 
> ```
> ENVIRONMENT:
>   ITERATOR_OPTIONS:
>     CYCLE = True
>     GROUP_BY_SCENE: True
>     NUM_EPISODE_SAMPLE: 1
>     SHUFFLE: False
> ```
> 
> Please, close the issue if we answered your question.

Thanks for both solutions, I solved this problem

closed 2021-01-14 by Jthon-lab