facebookresearch/habitat-lab #554: RuntimeError: Incompatible version of Habitat-Sim detected, please upgrade habitat_sim

ethanabrooks, 2020-12-31:
## 🐛 Bug

## Command
```python
python myscript.py
```

## To Reproduce

Steps to reproduce the behavior:

```python
# myscript.py

import habitat

# Load embodied AI task (PointNav) and a pre-specified virtual robot
env = habitat.Env(
    config=habitat.get_config("configs/tasks/pointnav.yaml")
)

observations = env.reset()

# Step through environment with random actions
while not env.episode_over:
    observations = env.step(env.action_space.sample())
```
Run 
```
❯ python myscript.py
Traceback (most recent call last):
  File "myscript.py", line 4, in <module>
    env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))
  File "/Users/ethanbrooks/habitat-lab/habitat/config/default.py", line 363, in get_config
    config.merge_from_file(config_path)
  File "/Users/ethanbrooks/miniconda3/envs/habitat/lib/python3.6/site-packages/yacs/config.py", line 211, in merge_from_file
    with open(cfg_filename, "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: 'configs/tasks/pointnav.yaml'
```

## Expected behavior

Not sure, but no errors...

## Additional context
```
❯ conda list | ag habitat
# packages in environment at /Users/ethanbrooks/miniconda3/envs/habitat:
habitat                   0.1.6                     dev_0    <develop>
habitat-sim               0.1.6                    pypi_0    pypi
```
on commit `38dd4d2dd1ee0a582f2cc11b173693ed272686a4` for habitat lab.

Skylion007, 2021-01-02:
If you are using from the latest master, you need to use the the habitat-sim nightly channel. We should have a new version out soon.

ethanabrooks, 2021-01-05:
This seems to work. Thank you!

closed 2021-01-05 by ethanabrooks