facebookresearch/habitat-lab #556: In habitat, if the next step will collide, how can we not execute the next step?

JeremyLinky, 2021-01-07:
## ❓ Questions and Help
In habitat, if the next step will collide, how can we not execute the next step?

erikwijmans, 2021-01-13:
That is not something we currently support.  You can modify the simulator to do that however.  You can see how the current movement is done here: https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/agent/controls/object_controls.py#L50

The move_filter_fn takes a start/end pair assuming an action was collision free and then provides an updated end point, so you can either write a new one or built on top of the current one

mathfac, 2021-01-14:
Another possible way to do that in Habitat Lab land is to create TaskAction and check if the agent collided, if collided reset agent to the previous position. Some motivation code snippet is [here](https://github.com/facebookresearch/habitat-lab/blob/c19e5666aac3d17ec55f09122ce4acff0873d4b4/habitat/tasks/nav/nav.py#L1050-L1069).

mathfac, 2021-01-14:
Closing the issue as the question was answered. Feel free to reopen if action needed.

closed 2021-01-14 by mathfac