facebookresearch/habitat-lab #558: Size mismatch when importing pretrained vqa model

nilinykh, 2021-01-10:
## ❓ Questions and Help

Hello,

I have encountered a size mismatch problem when loading the state dictionary of the pretrained vqa model.

In particular, when running
`python -u habitat_baselines/run.py --exp-config habitat_baselines/config/eqa/il_vqa.yaml --run-type eval`

I receive the following error
`RuntimeError: Error(s) in loading state_dict for VqaLstmCnnAttentionModel:
	size mismatch for cnn.encoder_seg.weight: copying a param with shape torch.Size([40, 512, 1, 1]) from checkpoint, the shape in current model is torch.Size([41, 512, 1, 1]).
	size mismatch for cnn.encoder_seg.bias: copying a param with shape torch.Size([40]) from checkpoint, the shape in current model is torch.Size([41]).
	size mismatch for cnn.score_pool2_seg.weight: copying a param with shape torch.Size([40, 16, 1, 1]) from checkpoint, the shape in current model is torch.Size([41, 16, 1, 1]).
	size mismatch for cnn.score_pool2_seg.bias: copying a param with shape torch.Size([40]) from checkpoint, the shape in current model is torch.Size([41]).
	size mismatch for cnn.score_pool3_seg.weight: copying a param with shape torch.Size([40, 32, 1, 1]) from checkpoint, the shape in current model is torch.Size([41, 32, 1, 1]).
	size mismatch for cnn.score_pool3_seg.bias: copying a param with shape torch.Size([40]) from checkpoint, the shape in current model is torch.Size([41]).`

Could it be that the problem is in the pretrained vqa model? The size of these cnn-related tensors does not match when compared to the ones specified in il models (41 vs 40)...

mathfac, 2021-01-14:
@mukulkhanna, maybe you can help here.

mukulkhanna, 2021-01-14:
Thanks @nilinykh for pointing this out. It seems the VQA checkpoint provided in the README is an outdated one.
I'll share the latest one here in a few hours, and also open a new PR to update the README.

mukulkhanna, 2021-01-15:
@nilinykh You can use this VQA checkpoint – https://drive.google.com/file/d/1frhIlgF1BpBT_vnRt7J5txlnfvgwXLeq/view?usp=sharing.
Also, I've discovered a bug in this [line](https://github.com/facebookresearch/habitat-lab/blob/master/habitat_baselines/il/metrics.py#L18) of the metrics class that affects the logging – make sure to remove the `sorted()` function there. I'll update it through a PR soon.

rpartsey, 2022-09-04:
Hi @mukulkhanna have you already completed/merged the changes described above?

mukulkhanna, 2022-09-08:
Hi @rpartsey, I just created a PR for the same (#942). Thanks for pointing this out!

We can close this issue as soon as it's merged.

mathfac, 2022-09-08:
Closed with merge of https://github.com/facebookresearch/habitat-lab/pull/942. Thanks.

closed 2022-09-08 by mathfac